AravisProject/aravis #11: set_camera_feature not working in python

maecky, 2016-08-31:
I tried to set the camera features in python for this camera:
Camera vendor: Point Grey Research
Camera model: Blackfly BFLY-PGE-09S2C

```
device = camera.get_device ()

device.set_integer_feature_value ("Width", 1024)
device.set_integer_feature_value ("Height", 1024)

print("Width =  {}".format(device.get_integer_feature_value ("Width")))
print("Height = {}".format(device.get_integer_feature_value ("Height")))
```

The output I get:
Width =  128
Height = 128

I do not get any errors either.
Can I somehow check what is not working?

EmmanuelP, 2016-09-01:
It can happen if the camera is already streaming or used by another process.

You can print the device status using device.get_status ()

Alternatively, you can enable the debug output using the ARV_DEBUG environment variable:

https://aravisproject.github.io/docs/aravis-0.4/aravis-building.html

EmmanuelP, 2016-09-02:
Did you test if you can set Width or Height using arv-tool ? Something like:

arv-tool-0.4 control Width=512 Height=512

maecky, 2016-09-03:
Thank you for your quick answer! I haven't had had the time to test it but as soon as I get a chance, I will report back.

maecky, 2016-09-09:
I got a chance to test a few things you suggested. At first, it worked and I could set the width and height values.
A few hours later (the system kept running in idle) it isn't working anymore...
I tried device.get_status() which results in:
`<enum ARV_DEVICE_STATUS_SUCCESS of type Aravis.DeviceStatus>`

But setting the width and height (or any other values doesn't work) and arv-tool-0.4 control Width=512 Height=512 results in:

```
Point Grey Research-16259895
Width = 512 (min:4;max:1288)
Height = 512 (min:2;max:728)

```

EmmanuelP, 2016-10-25:
> But setting the width and height (or any other values doesn't work) and arv-tool-0.4 control Width=512 Height=512 results in:

Point Grey Research-16259895
Width = 512 (min:4;max:1288)
Height = 512 (min:2;max:728)

I don't see what is wrong. You ask for 512x512, you get 512x512.

EmmanuelP, 2017-05-11:
No comment in months. Closing. Don't hesitate to reopen.

closed 2017-05-11 by EmmanuelP