AravisProject/aravis #15: Instructions on how to contribute with translations

josephgbr, 2016-09-03:
[Damned Lies](https://l10n.gnome.org/module/aravis/) send translators to Aravis wiki to look for instructions on how to submit translations, but there is no instructions in there. It would be nice to have some information specifying if one should open an issue attaching PO file, create a pull request etc.

EmmanuelP, 2016-09-04:
Hi Joseph,

I have created a wiki page which explains how to contribute translations. Please let me know if you think it is not detailed enough.

closed 2016-09-04 by EmmanuelP

josephgbr, 2016-09-04:
Did you published somewhere? I wasn't able to find it.

EmmanuelP, 2016-09-05:
https://github.com/AravisProject/aravis/wiki/Translations

josephgbr, 2016-09-05:
Looks good. However Damned Lies points to [https://wiki.gnome.org/Projects/Aravis](https://wiki.gnome.org/Projects/Aravis). How about (also?) add such information in there?

EmmanuelP, 2016-09-05:
Done.