AravisProject/aravis #17: Restore the previous video output based on autovideosink

EmmanuelP, 2016-09-07:
That will extend the support of GStreamer installations older than 1.6.

The selection of the output video method can be done at runtime. It is also the occasion to add support for Gtksink, in case of missing GL.

EmmanuelP, 2016-09-07:
Fixed by commit 11ca1e0d761fa320f3dc3618ffaed921adeb8d0d

closed 2016-09-07 by EmmanuelP

flixr, 2016-09-07:
Not working for me on Ubuntu 14.04, I get the error
```
arv-viewer:22272): Aravis-ERROR **: The user interface file is missing ('/usr/local/share/aravis-0.6/arv-viewer.ui')
```
but the file is there...

EmmanuelP, 2016-09-07:
You are missing a 'make install'.

flixr, 2016-09-07:
Nope, did that... and the file is really there...

EmmanuelP, 2016-09-07:
May be a file access permission issue ?

flixr, 2016-09-07:
Ok, found the problem, the error message was just a bit misleading: if I print the error from `gtk_builder_add_from_file`:
```required gtk+ version 3.12, current version is 3.10```

So am I out of luck on Ubuntu 14.04 with gtk+ 3.10?
Tried simply setting `version="3.10"` in ar-viewer.ui, but then I just get other errors:
```/usr/local/share/aravis-0.6/arv-viewer.ui:25:33 <object> requires attribute "id"```, etc

flixr, 2016-09-07:
See #18 for a slightly better error message...

EmmanuelP, 2016-09-07:
Indeed, the new UI use GtkPopover, which appeared in Gtk+-3.12.

If really want to use the 0.5.x branch, I will accept a patch that convert the popover back to a GtkExpander placed at the bottom of the video area, like in aravis 0.4.x.

Or, if you don't need USB support, I guess you should use the aravis_0_4 branch, which should work on ubuntu 14.04.

flixr, 2016-09-07:
Ok, thanks... then I'll stick to aravis_0_4 for now.
Do you intend to maintain that branch for longer? What about if we want to introduce new fixes/features there?

EmmanuelP, 2016-09-07:
I will mostly work on master, but I will happily push bug fixes to aravis_0_4. New features may also go to aravis_0_4 as long as ABI is backward compatible (no public struct changes, functions may be added).