AravisProject/aravis #22: Switching pixel format on fly freezes grabbing

ArkadiuszRaj, 2016-10-09:
Assuming grabbing is active, changing of pixel format on fly using following code:
```
arv_camera_abort_acquisition(camera);
g_object_unref(stream);
stream = NULL;
arv_camera_set_pixel_format(camera, newFormat);
stream = arv_camera_create_stream(camera, NULL, NULL);
payload = arv_camera_get_payload (camera);
for (int i = 0; i < 50; i++)
   arv_stream_push_buffer(stream, arv_buffer_new(payload, NULL));
```
makes app unable to get arv_buffer, i.e. _arv_stream_timeout_pop_buffer_ times outs all the time.

EmmanuelP, 2016-10-09:
Thanks for the bug report.

Which version of aravis are you using ?

Could you either rerun your test with ARV_DEBUG environment variable set to 'all' and attach here the console output ?

Also, a simple test program that shows the issue could help me to reproduce the issue.

ArkadiuszRaj, 2016-10-09:
Latest version, which is 0.6.

Frankly speaking trying several times to reproduce the problem I found following thing.
1) ..abort_capturing  is not working with TIS camera.  Have to use ..stop.
2) for even small fps MONO12 gives [GvStream::_check_frame_completion] Incomplete frame ..
3) Sometime switching pixel format gives strange result, had to restart camera (disconnect power) few times
[image]
[image]

Demo app, running for 300 iterations and using PR for OpenCV Aravis support
https://raw.githubusercontent.com/ArkadiuszRaj/opencv/aravis/modules/videoio/src/cap_aravis.cpp

```
#include "opencv2/core.hpp"
#include "opencv2/imgproc.hpp"
#include "opencv2/highgui.hpp"
#include "opencv2/videoio.hpp"
#include <iostream>

using namespace cv;
using namespace std;

int main()
{
    Mat image, src;

    VideoCapture capture(CV_CAP_ARAVIS);

    double fps = 30, gain = fps, exposure = 1./fps;

    capture.set(CV_CAP_PROP_GAIN, gain);
    capture.set(CV_CAP_PROP_EXPOSURE, exposure);
    capture.set(CV_CAP_PROP_FPS, fps);
    capture.set(CV_CAP_PROP_FOURCC, CV_FOURCC('G','R','E','Y'));

    int cc = 0;

    if(capture.isOpened())
    {
        for(int64_t i=0;;i++)
        {
            capture >> image;
            if(image.empty()) {
                cout << "No data " << i << endl;
            } else {
                pyrDown(image, src, image.size()/2);

                char str[200];
                sprintf(str,"%ld - %d",i, cc);
                putText(src, str, Point2f(100,50), FONT_HERSHEY_PLAIN, 2, Scalar::all(255));

                image = src;
                imshow("Sample", image);

                if(waitKey(10) >= 0)
                    break;
            }
            if(i == 300) break;
            if(i % 100); else {
                switch(cc ^= 1) {
                    case 0:
                        capture.set(CV_CAP_PROP_FOURCC, CV_FOURCC('Y','1','2',' '));
                        break;
                    case 1:
                        capture.set(CV_CAP_PROP_FOURCC, CV_FOURCC('G','R','E','Y'));
                        break;
                }
            }
        }
    }
    else
    {
        cout << "No capture" << endl;
        image = Mat::zeros(480, 640, CV_8UC1);
        imshow("Sample", image);
        waitKey(0);
    }
    return 0;
}

```
Network stats for 10 & 30 fps versions
[image]
[image]

And for the end some logs:

[debug-logs.zip](https://github.com/AravisProject/aravis/files/518488/debug-logs.zip)

EmmanuelP, 2016-10-10:
Could you please acquire a log with ARV_DEBUG=stream-thread:3,stream:3,device:3 . For some reason, buffer are not complete.

ArkadiuszRaj, 2016-10-10:
Here you go:

[log_thread.zip](https://github.com/AravisProject/aravis/files/519785/log_thread.zip)

Again 300 frames run but with following configurationMONO8 and 30fps. No pixel format change.

ArkadiuszRaj, 2016-10-10:
Is the camera ... choking?
```
42970	2.408561851	192.168.0.25	192.168.0.24	GVSP	1498	PAYLOAD [Block ID: 27 Packet ID: 1369] 
42971	2.474832314	192.168.0.25	192.168.0.24	GVSP	1498	PAYLOAD [Block ID: 29 Packet ID: 1354] 
```

EmmanuelP, 2016-10-11:
> Is the camera ... choking?
> 42970   2.408561851 192.168.0.25    192.168.0.24    GVSP    1498    PAYLOAD [Block ID: 27 Packet ID: 1369] 
> 42971   2.474832314 192.168.0.25    192.168.0.24    GVSP    1498    PAYLOAD [Block ID: 29 Packet ID: 1354] 

It looks like some packets are missing, and as packet resend is not implemented on this camera, buffers with missing packets are completely lost.

Loss of packets may have different origin. I doubt the camera is not sending all of them. Is the camera directly connected to your machine ?

Did you try to enable the real time thread, or the packet socket, or jumbo frames ?

ArkadiuszRaj, 2016-10-11:
Heard that this is known behavior of TIS 23 camera series, suggestion is
that it may be one of the reason TIS quickly switched to 33 series.

The camera is connected via TP-LINK TP-SG1008P (supports 16 KB jumbo frames)

arv-tool-0.6 control GevSCPSPacketSize
The Imaging Source Europe GmbH-29510170
GevSCPSPacketSize = 1484 (min:576;max:2984)

Changed to 2984 and apparently started to work

Pozdrawiam,

Arkadiusz Raj

ArkadiuszRaj, 2016-10-11:
```
arv-tool-0.6 control GevSCPSPacketSize
The Imaging Source Europe GmbH-29510170
GevSCPSPacketSize = 1484 (min:576;max:2984)
```
Changed to 2984 and apparently started to work. Thanks!

EmmanuelP, 2016-10-11:
Ok, so there is not much we can do at Aravis level.

closed 2016-10-11 by EmmanuelP