AravisProject/aravis #26: Running ldconfig after make install

ojura, 2016-10-14:
Hi, could you add running ldconfig to the install target? On Ubuntu 16.04, aravis executables cannot be run after doing make install (e.g. arv-tool-0.6: error while loading shared libraries: libaravis-0.6.so.0: cannot open shared object file: No such file or directory). This is fixed by running ldconfig (as root).

EmmanuelP, 2016-10-14:
I'm not sure what you need. make install works just fine on my system. Do you have an example of what is needed ?

ojura, 2016-10-14:
Yes, just running sudo ldconfig after doing make install fixes the problem. I was wondering if you could add that to the install target in the makefile.

EmmanuelP, 2016-10-14:
I did not find an example of a software that does something like that. And it seems pretty weird and unsafe to invoke sudo as part of 'make install'. So for now, I will leave 'make install' as is.

Cheers.

closed 2016-10-14 by EmmanuelP

ojura, 2016-10-14:
Of course you should not invoke it with sudo. User invokes make install with sudo, which should run ldconfig at the end as I have suggested.

I've found another example of this issue, which occurs when libraries are installed to /usr/local/lib, like you are doing. Take a look at https://github.com/esnet/iperf/issues/153

They seem to have given up on fixing the issue because automatically invoking ldconfig at the end of make install as I have suggested would cause errors on Mac OS. Maybe you could add a message which would suggest to Ubuntu users to run ldconfig, or consider installing your libraries to /usr/lib. Your software is the only one on my machine which installs libraries there and I am absolutely sure that it's broken on Ubuntu 16.04 (tested on multiple machines) after doing a make install; doing one of the above things would help prevent your users from wasting time on resolving this error.

EmmanuelP, 2016-10-14:
> Your software is the only one on my machine which installs libraries there and I am absolutely sure that it's broken on Ubuntu 16.04 (tested on multiple machines) after doing a make install;

Every library with a build system based on autotools is installed by default in /usr/local. That is the default install prefix. Install path can be changed using configure --prefix option.

Install procedure of aravis is pretty standard.

ojura, 2016-10-14:
YMMV, but after installing a software package, receiving errors that it cannot find its own dynamically linked libraries is really not standard :) It happens on Ubuntu 16.04 (I've just tried it out on a vanilla Ubuntu live CD). You can of course treat the issue as irrevelant as you're the maintainer, but consider at least putting a message somewhere.

fortminors, 2022-09-22:
@EmmanuelP Hello! This is still relevant on Ubuntu 22.04.

```
myuser@myuser:~$ arv-viewer-0.8 
arv-viewer-0.8: error while loading shared libraries: libaravis-0.8.so.0: cannot open shared object file: No such file or directory
myuser@myuser:~$ sudo ldconfig
[sudo] password for myuser: 
myuser@myuser:~$ arv-viewer-0.8
**works**
```

I followed exactly these [instructions](https://aravisproject.github.io/aravis/building.html) and the library seems to be installed in `/usr/local/lib/x86_64-linux-gnu` and not in `/usr/local`.

EmmanuelP, 2022-09-23:
meson added a ldconfig call on install for a while, and reverted this change because a lack of reliability:

https://github.com/mesonbuild/meson/issues/4737
https://github.com/mesonbuild/meson/pull/8833
https://mesonbuild.com/Release-notes-for-0-62-0.html#ldconfig-is-no-longer-run-on-install

Aravis documentation already says it  may be necessary to call ldconfig after install.

I have just added gettext to ubuntu and fedora dependency list in the documentation.