AravisProject/aravis #27: Underrrun destroys frames?

ArkadiuszRaj, 2016-10-16:
This is definitely not a camera problem (checked with wireshark).

When app is not able to grab frames and return buffer eventually a underrun condition arise.
According to [source](https://github.com/AravisProject/aravis/blob/af9e024c917461d6de9fabe1fd7bec6fed7a1bbf/src/arvgvstream.c#L623) files (version 0.6) this shall start ignoring frames. 

The thing is that during this unusual situation, received frames are corrupted, i.e. horizontal white or black bars starts to appear. Below is screenshot from my simple motion detection application.
[image]
Top left is feed from camera, top right a difference between 2 successive frames. Bottom not important.
Orange label, second row shows that for some time app was able to grab frames with speed of 12fps, while camera has produced frames with 20fps. 

After some time situation stabilizes (i.e app feeds aravis with buffers), but there is constant delay - no more real time video footage. This continues until the app/aravis is restarted.

As mentioned I am using aravis-0.6. Code used for grabbing frames:

```
bool CvCaptureCAM_Aravis::grabFrame()
{
    // remove content of previous frame
    framebuffer = NULL;

    if(stream) {
        ArvBuffer *arv_buffer = NULL;
        int max_tries = 10;
        int tries = 0;
        for(; tries < max_tries; tries ++) {
            arv_buffer = arv_stream_timeout_pop_buffer (stream, 200000);
            if (arv_buffer != NULL && arv_buffer_get_status (arv_buffer) != ARV_BUFFER_STATUS_SUCCESS) {
                arv_stream_push_buffer (stream, arv_buffer);
            } else break;
        }
        if (tries < max_tries) {
            size_t buffer_size;
            framebuffer = (void*)arv_buffer_get_data (arv_buffer, &buffer_size);

            arv_buffer_get_image_region (arv_buffer, NULL, NULL, &width, &height);

            arv_stream_push_buffer(stream, arv_buffer);
            return true;
        }
    }
    return false;
}

```

EmmanuelP, 2016-10-17:
In grabFrame, in normal condition, only one buffer is read. How often is this function called ?
If for some reason this function is called with a rate lower than the camera framerate, buffers will stay in the stream output queue.

ArkadiuszRaj, 2016-10-17:
The architecture is grab-parse-grab-parse and so on. If parsing tooks longer time I presume more frames shall be in output buffer, thus subsequent grabs shall be faster than programmed fps.  In worst case situation underrun shall happen, loosing some frames and recover. 

But I have not noticed that. Instead the delay I have mentioned above.

EmmanuelP, 2016-10-17:
Could you add some debugging output in grabFrame, using arv_stream_get_n_buffers and display the numbers ?

ArkadiuszRaj, 2016-10-17:
Here comes the screen  recording [link](https://drive.google.com/open?id=0B5rxQmNJcvrBaFBrV0FPblpXM1U) of Y800 feed from camera, processed by my application.
Short description of second orange line:
- FPS programmed,
- FPS real (counted as time between two successive returns from grab() function),
- processing time (counted as time between receiving frame and showing the debug picture with four quadrants),
- status  of output queue,
- gain (value on screen updated every second),
- exposure (value on screen updated every second),
- rotating thing (4 steps, 1 rotation per second).
Autoexposure algorithm is executed for every second frame.

As you can see despite the processing time is far below 50ms, the number of available buffers ripples.
Previously noted reuse of grab() as cadence generator can not be used to calc time, it is totally unstable. Sometimes it is going up to 50, and grab() no longer receives frames (timeouts). Sometimes goes to 20, than rapidly goes down to 5. And reception fps is far from programmed.

Run under Debian 16.04, Intel i5 4 cores, 2Ghz with 8GB of mem, running only this app, recording software, console and firefox with few static pages. By the way, playing some youtube move does change nothing. CPU usage approx 40% during recording.

Is there way to extract frames ID sent by camera? I mean the frame the buffer I am getting by calling arv_buffer_get_data() corresponds to? This will allow me to properly sync the autoexposure algos, as I will now how long to wait for camera reaction on changes parameters.

EmmanuelP, 2016-10-18:
Please don't edit your comments (apart for simple fixes). It makes them hard to follow. Just fill a new one if you have more informations.

One thing that can retain frame in the receiving threads is when they are incomplete. In this case, they will stay owned by the receiving threads until the frame retention timeout expires.

> Yet bars appear from time to time. example

When it happens, is the status of the buffer ARV_BUFFER_STATUS_SUCCESS ?

> Is this frame number from camera?

Yes.

> By the way, when camera gets disconnected / looses power, how to properly handle such situation?

There is a control-lost signal that gets emitted by the device when control is lost. In this case, the camera object must be destroyed and you have to start with a fresh one, as there is no API yet to recover the control of the camera.

ArkadiuszRaj, 2016-10-18:
My bad, sorry. Just after initial comment posted I thought to make some experiments to give you more detailed view on the "bars" issue.

> When it happens, is the status of the buffer ARV_BUFFER_STATUS_SUCCESS ?

From the **grab** function source above one can see that only if the buffer status is ARV_BUFFER_STATUS_SUCCESS  buffer is passed for further processing (next function constructs opencv mat object and clones the buffer content). Thus only those frames are visible on the screen. 

> There is a control-lost signal that gets emitted by the device when control is lost. In this case, the camera object must be destroyed and you have to start with a fresh one, as there is no API yet to recover the control of the camera.

Now I can see that is the source. Thanks! 
Does this mean that in grab() function I need first to check device status with **arv_device_get_status** ? And in case of failure, release camera object? Something like this?

```
bool CvCaptureCAM_Aravis::grabFrame()
{
    if(arv_device_get_status(arv_camera_get_device(camera)) == ARV_DEVICE_STATUS_NOT_CONNECTED) {
        close();
        return false;
    }

    // remove content of previous frame
    framebuffer = NULL;

    if(stream) {
        ArvBuffer *arv_buffer = NULL;
        int max_tries = 10;
        int tries = 0;
        for(; tries < max_tries; tries ++) {
            arv_buffer = arv_stream_timeout_pop_buffer (stream, 200000);
            if (arv_buffer != NULL && arv_buffer_get_status (arv_buffer) != ARV_BUFFER_STATUS_SUCCESS) {
                arv_stream_push_buffer (stream, arv_buffer);
            } else break;
        }
        if (tries < max_tries) {
            size_t buffer_size;
            framebuffer = (void*)arv_buffer_get_data (arv_buffer, &buffer_size);

            arv_buffer_get_image_region (arv_buffer, NULL, NULL, &width, &height);

            arv_stream_push_buffer(stream, arv_buffer);
            return true;
        }
    }
    return false;
}
```

EmmanuelP, 2016-10-18:
> Does this mean that in grab() function I need first to check device status with arv_device_get_status ?

No. Control loss is different than loss of connection. Status is only updated after a read/write try.

You should connect control-lost to a handler, and update a variable in this handler.

ArkadiuszRaj, 2016-10-21:
Let's leave this for a while, what about those horizontal bars?

EmmanuelP, 2017-03-01:
> Let's leave this for a while, what about those horizontal bars?

If the buffer status is SUCCESS, all the frame data were received completely. Which means if there is an horizontal bar, either a data corruption happens somewhere, or the actual data from the sensor contain these bars.

The data corruption may be due to Aravis, but it will hard for me to debug this issue without the device at hand.

closed 2017-05-09 by ArkadiuszRaj

PauloMendes33, 2021-05-05:
Hello

For High frequency (fps) cameras, the packet resend should be canceled.
Here --- camnode.cpp line 150: 
             ArvGvStream *CreateStream(void){
	                                       gboolean 		bAutoBuffer = FALSE;
                                               // PM	
                                               //gboolean 	bPacketResend = TRUE;
                                               gboolean          bPacketResend = FALSE;
             ...

For a GigE Camera, 10Gb ethernet connetion ~ 226 FPS, with a 1Gb ethernet connection ~ 34.3 FPS, it is lost 3/4 frames ( Frame error: ARV_BUFFER_STATUS_MISSING_PACKETS).
Canceling the packet resend it is achieved an efficiency of 31/34.3 frames, in my case. Possibly, in a 60 FPS camera the number of "missing" packets is low too.