AravisProject/aravis #28: Set of errors printed while starting app

ArkadiuszRaj, 2016-10-22:
Using Aravis from HEAD and camera TIS DMK 23GX236, here is a list of the failed assertions that are printed on terminal every time I start my app.

> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed
> 
> (Sample:16576): Aravis-CRITICAL **: arv_gc_feature_node_get_value_type: assertion 'ARV_IS_GC_FEATURE_NODE (node)' failed

ArkadiuszRaj, 2016-10-22:
While compiling with aravis-0.4 console is clean

EmmanuelP, 2016-10-22:
Restart your app with debug output enabled using:

export ARV_DEBUG=all:3

ArkadiuszRaj, 2016-10-24:
Between my last comment and yours I have several times recompiled/reinstalled versions 0.4 & 0.6.
This was as I have verified OpenCV compliance  with both versions.
 
Then after your suggestion I started app with debug environmental variable set. To my surprise using 0.6 no asserts were printed. Just as it was with 0.4.

closed 2016-10-24 by ArkadiuszRaj