AravisProject/aravis #38: Possible memory leak in arv_gv_interface_discover()

mdavidsaver, 2016-12-27:
This and #37 are the only "definite" leaks flagged for my test program.  I don't see any problem in arv_gv_interface_discover() or handling of ArvGvInterfaceDeviceInfos, but the manual ref. counting and my not being familiar with the semantics of GHashTable, keep me from dismissing this as a false positive.

```
 33 bytes in 1 blocks are definitely lost in loss record 338 of 706
    at 0x4C28C20: malloc (vg_replace_malloc.c:296)
    by 0x53347C9: g_malloc (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4200.1)
    by 0x534D127: g_strndup (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4200.1)
    by 0x4E66E11: arv_gv_interface_discover (in /usr/lib/x86_64-linux-gnu/libaravis-0.6.so.0.0.0)
    by 0x4E67217: arv_gv_interface_open_device (in /usr/lib/x86_64-linux-gnu/libaravis-0.6.so.0.0.0)
    by 0x4E4AE7D: arv_open_device (in /usr/lib/x86_64-linux-gnu/libaravis-0.6.so.0.0.0)
    by 0x4E54235: arv_camera_new (in /usr/lib/x86_64-linux-gnu/libaravis-0.6.so.0.0.0)
    by 0x402587: main (inspectcam.cpp:105)
```

EmmanuelP, 2017-01-03:
Hi Michael.

Do you call arv_shutdown in inspectcam.cpp ? This call is optional, but should be used before exit when you want to debug memory leaks.

mdavidsaver, 2017-01-05:
I was not calling arv_shutdown, but am now.  This record no longer appears.

closed 2017-01-05 by mdavidsaver