AravisProject/aravis #39: arv_uv_device_get_type undefined.

ethanrublee, 2016-12-27:
On HEAD of master, 53a6a545b, starting at 33805d1. I get the following error:

```
make[2]: Entering directory '/home/rublee/code/aravis/src'
  CCLD     arv-fake-gv-camera-0.6
./.libs/libaravis-0.6.so: undefined reference to `arv_uv_device_get_type'
collect2: error: ld returned 1 exit status
```

EmmanuelP, 2017-01-03:
Doesn't happen here...

Did you try to clean your working directory (git clean -xfd) or in a fresh clone ?

ArkadiuszRaj, 2017-01-04:
Same here. My jenkins' head sync job started to fail Dec 26th. After inclusion of:
```
    readme: OSX build instructions (commit: 3c87c37) (detail / githubweb)
    chunk_parser: fix structure name (commit: b997890) (detail / githubweb)
    gvdevice: OSX compilation fix (commit: bd8ae1b) (detail / githubweb)
    data: add PointGrey and Ximea USB descriptor data (commit: 1737eeb) (detail / githubweb)
    fakecamera: fix genicam filename (commit: 77309f9) (detail / githubweb)
    fake_camera: add a cmd line option for custom genicam data (commit: f327004) (detail / githubweb)
    uv_stream: assign buffer pixel format field (commit: 09dff01) (detail / githubweb)
    uv_stream: change maximum transfer size to 1MiB (commit: bdb84a0) (detail / githubweb)
    uv_stream: detect incomplete frame transfers (commit: 3d06b17) (detail / githubweb)
    uv_device: add support for raw genicam data (commit: 614756b) (detail / githubweb)
    camera: Ximea support (commit: aa025c6) (detail / githubweb)
    camera: USB3Vision bandwith control support (commit: 33805d1) (detail / githubweb)
    uv_device: device descriptor-agnostic support (commit: 53a6a54) (detail / githubweb)
```
I have also tried to clone fresh copy from github. No success either.
Here is full log from first failing run: [link](http://jenkins.hijax.pl/job/libravis-head-sync/27/console)

closed 2017-01-04 by EmmanuelP