AravisProject/aravis #43: Can't detect Basler USB Camera on Nvidia Jetson TX1

YuryAlkevich, 2017-02-03:
I've compiled aravis on ubuntu 16.04, x86_64 desktop PC and tested it with two cameras:
Basler acA1920-150um (USB camera)  and Basler avA2300-25gm (Ethernet camera). 
Both cameras works well on that machine. But my primary interest is to use aravis on Jetson TX1 (instead of Balser Pylon SDK which currently doesn't support x64 ARM arch). Therefore I've compiled aravis on Jetson TX1 with Ubuntu 16.04, aarch64 ARM, but aravis can't detect USB camera, though Ethernet camera works fine.

EmmanuelP, 2017-02-03:
Do you see the camera with lsusb ?
Did you connect it to the USB3 port ?
What is the output of 'arv-tool-0.6 -d all:3' ?

EmmanuelP, 2017-02-03:
It can also be an access right issue. You may want to try using the root account. If it works, check the access right of the /dev/usbmon* files. On fedora, user must be part of the usbmon group.

YuryAlkevich, 2017-02-03:
Ok, here is what I've got:

[output.txt](https://github.com/AravisProject/aravis/files/750218/output.txt)
[sudo arv-viewer -d all-3.txt](https://github.com/AravisProject/aravis/files/750219/sudo.arv-viewer.-d.all-3.txt)

You were right, with sudo camera can be detected. But as you can see in output.txt, arv-viewer throws error and terminates. I've also attached the second file where I launched arv-viewer in debug mode, and in that case it just hangs on forever.

YuryAlkevich, 2017-02-03:
Also if I unplug the camera form USB when avr-viewer is hanging in debug mode, I get the following output:
(arv-viewer:19213): GLib-ERROR **: /build/glib2.0-7Ure2w/glib2.0-2.48.2/./glib/gmem.c:130: failed to allocate 8028075781016781059 bytes
Trace/breakpoint trap
And the program terminates

EmmanuelP, 2017-02-03:
Could be an issue with USB endpoints. What is the output of:

lsusb -v -d 2676:ba02

What version of aravis do you use ?
Could you try the master branch. There is some improvement that could fix an endpoint issue.

YuryAlkevich, 2017-02-04:
I've pulled latest version of master. The issue still exists. Here is the output of lsusb
[lsusb-output.txt](https://github.com/AravisProject/aravis/files/752127/lsusb-output.txt)
[master-output.txt](https://github.com/AravisProject/aravis/files/752128/master-output.txt)

YuryAlkevich, 2017-02-07:
up

EmmanuelP, 2017-02-07:
It is not and endpoint issue.

Without the hardware at hand, it is hard to understand what is wrong. For some reason, aravis doesn't get any answer from the device. You could try to debug the calls to libusb in arv_uv_device_bulk_transfer by printing the error values.

mcguire-steve, 2017-02-08:
Another option is to enable debugging in LibUSB by exporting LIBUSB_DEBUG=4  and then running something from the suite that tries to access the camera.

YuryAlkevich, 2017-02-08:
sudo LIBUSB_DEBUG=4 arv-viewer -d all:3
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000012] [00001679] libusb: debug [libusb_init] created default context
[ 0.000322] [00001679] libusb: debug [libusb_init] libusb v1.0.20.11004
[ 0.000410] [00001679] libusb: debug [find_usbfs_path] found usbfs at /dev/bus/usb
[ 0.000468] [00001679] libusb: debug [op_init] bulk continuation flag supported
[ 0.000495] [00001679] libusb: debug [op_init] zero length packet flag supported
[ 0.000535] [00001679] libusb: debug [op_init] sysfs can relate devices
[ 0.000561] [00001679] libusb: debug [op_init] sysfs has complete descriptors
[ 0.000887] [00001683] libusb: debug [linux_udev_event_thread_main] udev event thread entering.
[ 0.005954] [00001679] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.005992] [00001679] libusb: debug [linux_get_device_address] scan usb1
[ 0.006099] [00001679] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.006129] [00001679] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.006151] [00001679] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.006688] [00001679] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.006718] [00001679] libusb: debug [linux_get_device_address] scan usb2
[ 0.006792] [00001679] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.006814] [00001679] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.006833] [00001679] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.007108] [00001679] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.007132] [00001679] libusb: debug [linux_get_device_address] scan 2-1
[ 0.007202] [00001679] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.007223] [00001679] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.007241] [00001679] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.007323] [00001679] libusb: debug [linux_get_parent_info] Dev 0x83f440 (2-1) has parent 0x89f550 (usb2) port 1
[ 0.007561] [00001679] libusb: debug [linux_get_device_address] getting address for device: 2-2 detached: 0
[ 0.007584] [00001679] libusb: debug [linux_get_device_address] scan 2-2
[ 0.007655] [00001679] libusb: debug [linux_get_device_address] bus=2 dev=3
[ 0.007676] [00001679] libusb: debug [linux_enumerate_device] busnum 2 devaddr 3 session_id 515
[ 0.007695] [00001679] libusb: debug [linux_enumerate_device] allocating new device for 2/3 (session 515)
[ 0.007763] [00001679] libusb: debug [linux_get_parent_info] Dev 0x899600 (2-2) has parent 0x89f550 (usb2) port 2
[ 0.008174] [00001679] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.008198] [00001679] libusb: debug [linux_get_device_address] scan usb3
[ 0.008269] [00001679] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.008290] [00001679] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.008308] [00001679] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.008560] [00001679] libusb: debug [linux_get_device_address] getting address for device: 3-1 detached: 0
[ 0.008584] [00001679] libusb: debug [linux_get_device_address] scan 3-1
[ 0.008652] [00001679] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.008673] [00001679] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.008691] [00001679] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.008758] [00001679] libusb: debug [linux_get_parent_info] Dev 0x89c960 (3-1) has parent 0x885d30 (usb3) port 1
[ 0.008898] [00001679] libusb: debug [linux_get_device_address] getting address for device: 3-1.1 detached: 0
[ 0.008920] [00001679] libusb: debug [linux_get_device_address] scan 3-1.1
[ 0.008991] [00001679] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.009013] [00001679] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.009031] [00001679] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.009098] [00001679] libusb: debug [linux_get_parent_info] Dev 0x88b1a0 (3-1.1) has parent 0x89c960 (3-1) port 1
[ 0.009429] [00001679] libusb: debug [linux_get_device_address] getting address for device: 3-1.3 detached: 0
[ 0.009452] [00001679] libusb: debug [linux_get_device_address] scan 3-1.3
[ 0.009522] [00001679] libusb: debug [linux_get_device_address] bus=3 dev=4
[ 0.009544] [00001679] libusb: debug [linux_enumerate_device] busnum 3 devaddr 4 session_id 772
[ 0.009562] [00001679] libusb: debug [linux_enumerate_device] allocating new device for 3/4 (session 772)
[ 0.009628] [00001679] libusb: debug [linux_get_parent_info] Dev 0x88b250 (3-1.3) has parent 0x89c960 (3-1) port 3
[ 0.009884] [00001679] libusb: debug [usbi_add_pollfd] add fd 13 events 1
[ 0.009926] [00001679] libusb: debug [usbi_io_init] using timerfd for timeouts
[ 0.009948] [00001679] libusb: debug [usbi_add_pollfd] add fd 15 events 1
[ 0.009979] [00001679] libusb: debug [libusb_get_device_list] 
[ 0.010023] [00001679] libusb: debug [libusb_get_device_descriptor] 
[... cut ...]
[ 0.135059] [00001679] libusb: debug [handle_control_completion] handling completion status 0
[ 0.135208] [00001679] libusb: debug [disarm_timerfd] 
[ 0.135359] [00001679] libusb: debug [usbi_handle_transfer_completion] transfer 0x89cab8 has callback 0x7f82836708
[ 0.136003] [00001679] libusb: debug [sync_transfer_cb] actual_length=18
[ 0.136128] [00001679] libusb: debug [libusb_free_transfer] transfer 0x89cab8
[ 0.136454] [00001679] libusb: debug [libusb_close] 
[ 0.136593] [00001679] libusb: debug [usbi_remove_pollfd] remove fd 16
[ 0.136750] [00001679] libusb: de...

EmmanuelP, 2017-02-08:
Thanks, but this log is missing ARAVIS debug output which would help to understand when these libusb calls happen.

I have pushed a patch to master that improves the debug output a bit. Could you give it a try, using `./viewer/arv-viewer -d all:3` ?

By the way, the access right issue can be solved by adding an udev rules file in /etc/udev/rules.d. An example is available in aravis.rules. Just add a line with the right vendor id and put the files in /etc/udev/rules.d/.

YuryAlkevich, 2017-02-09:
Sorry I haven't shown full debug log. Here is the full log of viewer with last updates from master.
[090217_debug_aravis.txt](https://github.com/AravisProject/aravis/files/762955/090217_debug_aravis.txt)
At 135.903441 I turned off the camera.
Also thank you for the advice about udev rules

EmmanuelP, 2017-02-13:
For comparison, here is the log of a successful connection:

[successful_connection.txt](https://github.com/AravisProject/aravis/files/770605/successful_connection.txt)

Pragneshvirani, 2017-05-07:
Hi,
You found a solution, Even I am facing the same problem.Is it because of insufficient power for the camera? Because camera Power LED goes off after 1-2  second of connection.

Thanks.

YuryAlkevich, 2017-05-07:
Hello,
We are using Basler cameras and software developers of Basler have released new SDK for arm x64 arch recently. So now we don't need aravis libraries and we still haven't found the solution for the issue.

Pragneshvirani, 2017-05-07:
Thanks for the update. 
Even we are trying to interface Basler camera. From their(Basler) web I do not see arm x64 SDK.Can see only 32bit Arch.Can you please point out from where can get this?
Thank You!

EmmanuelP, 2017-05-07:
For the record, a user reported a successful use of aravis 0.5.7 with 4 Basler USB cameras, from a jetson TX1 board using ubuntu. Here is the camera list:

[image]

Thanks to him and his company, I will soon receive a TX2 board and be able to test aravis using this platform.

@Pragneshvirani If using a proprietary library is not an issue for you, you may also want to try Pleora's eBus software. I don't know if they propose 64 bit ARM support though.

Pragneshvirani, 2017-05-08:
Thanks, @EmmanuelP for the update. If Basler has x64 support then defiantly we look for, because later we need to have a camera I/O control also for LED strobe.which don't I think we can get from Aravis or Pleora? 

@YuryAlkevich Did you already tried with Basler x64 SDK? 

Thanks.

Regards,
Pragnesh

EmmanuelP, 2017-05-08:
> because later we need to have a camera I/O control also for LED strobe.which don't I think we can get from Aravis or Pleora?

If these features are exposed through Genicam, and I guess they are, they should be supported by aravis and pleora.

Pragneshvirani, 2017-05-09:
@EmmanuelP Noted, I will try that. Thanks a lot.

YuryAlkevich, 2017-05-10:
@Pragneshvirani
I asked a support team to send me a beta version of the SDK. Official release will be in June. 
Here is the link to beta, which they send me
https://1drv.ms/u/s!Ag-yO1E4-6Eahp4Qf0i6zFOUJyjcmQ

Pragneshvirani, 2017-05-13:
Just to update, I tried Basler x64 SDK and it's working fine.
Thank you so much @YuryAlkevich & @EmmanuelP

EmmanuelP, 2017-07-06:
Since I have a report of working devices on Jetson TX1 platform, and the discussion is now about Basler SDK, I'm closing this issue.

closed 2017-07-06 by EmmanuelP