AravisProject/aravis #45: install instructions for ubuntu 16.04

waspinator, 2017-02-13:
I'm not very familiar with compiling software with the GNU build system, and I'm having troubling following the instructions in the readme.

I naively tried to download the zip of the repository, extract it, and run the `./configure` in the root extracted directory. Of course there is no such file, so it doesn't work. Should I be doing something before that, or am I running it in the wrong place?

Thanks

EmmanuelP, 2017-02-13:
Hi,

You should not use the download button on the main page, because it doesn't contain a release of Aravis, but a snapshot of the git repository.

If you want to install Aravis, you should use the archives linked in the downloads section of the readme file:

https://github.com/AravisProject/aravis#downloads

  Cheers,

    Emmanuel.

closed 2017-02-13 by EmmanuelP

waspinator, 2017-02-13:
Thanks.

For reference these are the command I ran to install it on ubuntu 16.04

    sudo apt-get install intltool libgstreamer-plugins-base1.0-dev libnotify-dev libgtk-3-dev
    ./configure --enable-viewer
    make
    make install