AravisProject/aravis #46: How to install Aravis on Ubuntu 14.04

matheusns, 2017-02-20:
I'm trying to connect the Teledyne Dalsa Genie Nano C2590 and still now I don't get it. Could something help me, please?

EmmanuelP, 2017-02-22:
What version of aravis are you using ?

What gives 'arv-tool-0.6 -d all:3' ? (or arv-tool-0.4 if you are using stable aravis).

matheusns, 2017-02-22:
@EmmanuelP  I'm using the stable version 0.4.1 and a GigE camera and I'm trying to run it and nothing.

matheusns, 2017-02-22:
When a run 'arv-tool-0.6 -d all:3' I get the follow answer:

``` bash
Found 0 USB3Vision device (among 8 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 169.254.7.55
[GvDiscoverSocket::new] Add interface 172.28.16.182
packet_type  = ack
command      = discovery-ack
size         = 248
id           = 65535
manufacturer = Teledyne DALSA
name         = S1112934
model        = Nano-C2590
address      = 169.254.5.61
```

EmmanuelP, 2017-03-01:
So your camera is listed by Aravis. What do you want to do exactly ?

matheusns, 2017-03-02:
Unfortunately, Teledyne Dalsa Genie Nano isn't listed by Aravis, but I've seen in another "issues" people that got use this camera with Aravis. I want to connect it with my computer and get snapshots and video stream. Give me this hand, please!

EmmanuelP, 2017-03-02:
Is the camera listed in arv-viewer ?

matheusns, 2017-03-02:
I can get arv-viewer because when I run ```bash``` ./autogen.sh --enable-viewer ``` I get the following message:
``` bash```
configure: error: Requested 'gtk+-3.0 >= 3.12' but version of GTK+ is 3.10.8 - Can not build viewer
```
How can I get the GTK 3.12 in the safety way?

EmmanuelP, 2017-03-02:
Installation of another GTK library can be tricky.

Meanwhile, you may want to try to use the gstreamer plugin. Make sure you have the gstreamer plugin compiled, go to the gst directory in the sources, then try:

./gst-aravis-launch aravissrc  camera-name="Teledyne\ DALSA-S1112934" ! videoconvert ! xvimagesink sync="false"

matheusns, 2017-03-02:
So, by now I've got a questions, What release of GTK do you advice me to install? and When I tried to run ``` bash ./gst-aravis-launch aravissrc camera-name="Teledyne\ DALSA-S1112934" ! videoconvert ! xvimagesink sync="false" ``` I got the following messages:
``` bash
atlantis@light:~/Dropbox/LIGHTSIDE/Teledyne/aravis/gst$ ./gst-aravis-launch aravissrc camera-name="Teledyne\ DALSA-S1112934" ! videoconvert ! xvimagesink sync="false"
Setting pipeline to PAUSED ...

(gst-launch-1.0:31421): GStreamer-CRITICAL **: range start is not smaller than end for `GstIntRange'
Pipeline is live and does not need PREROLL ...
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data flow error.
Additional debug info:
gstbasesrc.c(2865): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming task paused, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

So What can we do for that? I'm feeling we are almost there, thanks!

EmmanuelP, 2017-03-03:
Please rerun gst-aravis-launch with the environment variable ARV_DEBUG set to all:3 (done using 'export ARV_DEBUG=all:3' if you are using bash).

EmmanuelP, 2017-03-03:
> What release of GTK do you advice me to install?

The latest stable one, which is currently 3.22.9.

http://ftp.gnome.org/pub/gnome/sources/gtk+/3.22/gtk+-3.22.8.tar.xz

matheusns, 2017-03-23:
@EmmanuelP I've already done everything you told me, but until now I can't visualize anything. I installed the GTK 3.22.9 and solved the first error about the arv-viewer, but now when I run it nothing appear on the screen. What can we do for fix that? 
Thanks for all!

matheusns, 2017-03-23:
When a I had run 
´´´ bash sudo ARV_DEBUG=all:3 ./gst-aravis-launch aravissrc camera-name="Teledyne\ DALSA-S1112934" ! videoconvert ! xvimagesink sync="false ´´´
I got the following answer:

[gst_ARV_DEBUG=all:3.txt](https://github.com/AravisProject/aravis/files/865702/gst_ARV_DEBUG.all.3.txt)

EmmanuelP, 2017-04-10:
Pleas attach the output of:

`./arv-camera-test -d all:3`

EmmanuelP, 2017-11-22:
No update, old report. I'm closing this issue. Please reopen if you have the requested informations.

Thanks.

closed 2017-11-22 by EmmanuelP