AravisProject/aravis #48: Memleak in update_device_list

danrauch, 2017-02-28:
Hi, 

I just want to let you know, that valgrind reports a memleak in my application after calling update_device_list.
I could also verify with the arv-viewer. Note that I use ARAVIS_0_5_7 on Ubuntu 16.04.

Here the relevant valgrind ouput:

arv-viewer:

==24843== 33 bytes in 1 blocks are definitely lost in loss record 6,687 of 14,378
==24843==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==24843==    by 0x6D3D718: g_malloc (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==24843==    by 0x6D56567: g_strndup (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==24843==    by 0x70306E0: arv_gv_interface_device_infos_new (arvgvinterface.c:224)
==24843==    by 0x70306E0: arv_gv_interface_discover (arvgvinterface.c:330)
==24843==    by 0x7030E48: arv_gv_interface_update_device_list (arvgvinterface.c:384)
==24843==    by 0x702C312: arv_interface_update_device_list (arvinterface.c:86)
==24843==    by 0x4068BE: update_device_list_cb (arvviewer.c:685)
==24843==    by 0x408201: activate (arvviewer.c:1221)
==24843==    by 0x6AAB1D3: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==24843==    by 0x6AC59A5: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==24843==    by 0x6AC608E: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==24843==    by 0x67C0AF5: ??? (in /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0.4800.2)

==24843== 48 bytes in 1 blocks are definitely lost in loss record 8,323 of 14,378
==24843==    at 0x4C2FB55: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==24843==    by 0x6D3D770: g_malloc0 (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==24843==    by 0x7030ED1: arv_gv_interface_update_device_list (arvgvinterface.c:396)
==24843==    by 0x702C312: arv_interface_update_device_list (arvinterface.c:86)
==24843==    by 0x4068BE: update_device_list_cb (arvviewer.c:685)
==24843==    by 0x408201: activate (arvviewer.c:1221)
==24843==    by 0x6AAB1D3: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==24843==    by 0x6AC59A5: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==24843==    by 0x6AC608E: g_signal_emit (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==24843==    by 0x67C0AF5: ??? (in /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0.4800.2)
==24843==    by 0x67C0CE1: g_application_run (in /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0.4800.2)
==24843==    by 0x40514C: main (main.c:107)

==24843== 99 bytes in 3 blocks are definitely lost in loss record 12,116 of 14,378
==24843==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==24843==    by 0x6D3D718: g_malloc (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==24843==    by 0x6D56567: g_strndup (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==24843==    by 0x70306E0: arv_gv_interface_device_infos_new (arvgvinterface.c:224)
==24843==    by 0x70306E0: arv_gv_interface_discover (arvgvinterface.c:330)
==24843==    by 0x7030E48: arv_gv_interface_update_device_list (arvgvinterface.c:384)
==24843==    by 0x702C312: arv_interface_update_device_list (arvinterface.c:86)
==24843==    by 0x4068BE: update_device_list_cb (arvviewer.c:685)
==24843==    by 0x6AAB1D3: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==24843==    by 0x6AC59A5: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==24843==    by 0x6AC6554: g_signal_emit_by_name (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==24843==    by 0x6AAB1D3: ??? (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==24843==    by 0x6AC59A5: g_signal_emit_valist (in /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)

my app:

==24991== 33 bytes in 1 blocks are definitely lost in loss record 624 of 1,233
==24991==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==24991==    by 0x68E9718: g_malloc (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==24991==    by 0x6902567: g_strndup (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==24991==    by 0x4E6B6E0: arv_gv_interface_device_infos_new (arvgvinterface.c:224)
==24991==    by 0x4E6B6E0: arv_gv_interface_discover (arvgvinterface.c:330)
==24991==    by 0x4E6BE48: arv_gv_interface_update_device_list (arvgvinterface.c:384)
==24991==    by 0x4E67312: arv_interface_update_device_list (arvinterface.c:86)
==24991==    by 0x50938C1: GetDeviceInterfaces (in /usr/local/lib/libmescan.so.1.0)
==24991==    by 0x40179E: main (in /home/v2d3d/development/examples/GetProfilesCallback/GetProfilesCallback)
==24991== 
==24991== 48 bytes in 1 blocks are definitely lost in loss record 670 of 1,233
==24991==    at 0x4C2FB55: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==24991==    by 0x68E9770: g_malloc0 (in /lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==24991==    by 0x4E6BED1: arv_gv_interface_update_device_list (arvgvinterface.c:396)
==24991==    by 0x4E67312: arv_interface_update_device_list (arvinterface.c:86)
==24991==    by 0x50938C1: GetDeviceInterfaces (in /usr/local/lib/libmescan.so.1.0)
==24991==    by 0x40179E: main (in /home/v2d3d/development/examples/GetProfilesCallback/GetProfilesCallback)

Greetings and thank you!

EmmanuelP, 2017-03-01:
When you want to test an application using aravis with valgrind, you should add a call arv_shutdown() that frees the interface objects. That should fix the memory leak you are seeing.

closed 2017-03-01 by EmmanuelP

danrauch, 2017-03-01:
Okay, thank you very much for the quick response.  I just saw you mentioned this in #38; sorry for the duplicate.
You might want to include this into the documentation/in an example program, to avoid this questions.

w578583202, 2017-05-11:
Will the development package libmescan support armv4 architecture of the processor? I use arm9 processor, compiler suggested illegal instruction, because libmescan sake?

w578583202, 2017-05-11:
thank you from china

w578583202, 2017-05-11:
Development package libmescan support armv4 architecture of the processor? I use arm9 processor, the procedure is running suggested that illegal instructions, because libmescan's sake? Thank you