AravisProject/aravis #50: Race in stream new-buffer signal leads to self destruction

mdavidsaver, 2017-03-08:
It seems that ```g_signal_emit()``` manipulates the ref. count of ```ArvStream``` instances.  This means that it is possible for the stream thread to have the last reference, and to attempt to destroy the stream.  This triggers an assert() failure from ```g_thread_join()``` when the stream thread attempts to join itself.

This is fairly difficult to trigger.  I've encountered it twice in ~2 weeks of destruct testing involving triggering cleanup while still streaming frames.

The symptom is

```
(process:23937): GLib-ERROR **: file /build/glib2.0-y6934K/glib2.0-2.42.1/./glib/gthread-posix.c: line 1218 (g_system_thread_wait): error 'Resource deadlock avoided' during 'pthread_join (pt->system_thread, NULL)'
```

A stack trace of the faulting thread is.

```
#0  0x00007fe5374f2d30 in g_logv () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007fe5374f2f6f in g_log () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007fe53753039e in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007fe537512c95 in g_thread_join ()
   from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007fe537a39fee in arv_gv_stream_finalize (object=0x7fe4ec0c62d0)
    at arvgvstream.c:1188
#5  0x00007fe5377c626a in g_object_unref ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#6  0x00007fe5377db494 in g_signal_emit_valist ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#7  0x00007fe5377db9df in g_signal_emit ()
   from /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#8  0x00007fe537a387df in _close_frame (
    thread_data=thread_data@entry=0x7fe4ec081960, 
    frame=frame@entry=0x7fe51001dc00) at arvgvstream.c:499
#9  0x00007fe537a38fb2 in _check_frame_completion (
    current_frame=<optimized out>, time_us=1488981195843477, 
    thread_data=0x7fe4ec081960) at arvgvstream.c:540
#10 _loop (thread_data=0x7fe4ec081960) at arvgvstream.c:715
#11 arv_gv_stream_thread (data=0x7fe4ec081960) at arvgvstream.c:933
#12 0x00007fe537512845 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007fe53728d0a4 in start_thread (arg=0x7fe4a7bf8700)
```

mdavidsaver, 2017-03-08:
The best idea I have would be to add a mutex to ```_ArvStreamPrivate``` which is locked in ```arv_stream_push_output_buffer()``` around the test of ```emit_signals``` and the call to ```g_signal_emit()```.  Also in ```arv_stream_set_emit_signals()``` when ```emit_signals``` is updated.

The stream shutdown sequence would then be to set emit_signals=0, which by locking the mutex would ensure that ```g_signal_emit()``` isn't running after ```arv_stream_set_emit_signals()``` returns.

Does this seem like the right solution?

danrauch, 2017-03-09:
I've seen this a couple of times now, but it is very hard to trigger indeed. Using a low-performance ARM board or simulating a high load on the test machine seems to maximises the chances of triggering it. Maybe this helps for testing purposes.

Note that under this circumstances the DEADLK also occures in arvexample etc, which stops the acquistion before unrefing the stream object. If a sleep (couple of ms) is introduced between stopping the acquistion and unrefing, I couldn't reproduce the assertion. This makes me belive, that (without sleep) I enter the same state David is describing (cleanup while still streaming frames) with no intention to do so.

EmmanuelP, 2017-03-09:
@mdavidsaver good catch. I have pushed to master a patch that adds a new test to fakegv test that triggers this issue.

Your solutions seems right. It should be easy to test with 'fakegv -p /fakegv/stream'.

mdavidsaver, 2017-03-10:
```c
ArvGvFakeCamera * arv_gv_fake_camera_new(const char *interface_name);
```

neat :+1:

closed 2017-03-18 by EmmanuelP