AravisProject/aravis #52: Swissknife does not handle expressions correctly

TIS-Edgar, 2017-03-10:
Hi,
When reading values from a TIS 33G P1300e some values are incorrectly evaluated and are thus returned as 0.

When executing:
```arv-tool-0.6 control FPS -n "The Imaging Source-47619971"```

The result is:
```
The Imaging Source-47619971 (192.168.0.180)
FPS = 5 (min:5;max:0)
```

You can find the xml description of the camera [here](https://gist.github.com/TIS-Edgar/e9b2d2e1c7fd838c72b6515e876057bb).

This has been tested with current master.

If you require any additional information, please let me know.

EmmanuelP, 2017-03-10:
I suspect a bug in expression nodes. Could you attach the output of `arv-tool-0.6 control FPS -n "The Imaging Source-47619971" -d evaluator:3` ?

EmmanuelP, 2017-03-10:
>  I suspect a bug in expression nodes

I guess I should have read the issue title first...

TIS-Edgar, 2017-03-13:
Here is the output:

[47619971_control_fps.txt](https://github.com/AravisProject/aravis/files/837330/47619971_control_fps.txt)

EmmanuelP, 2017-03-13:
Please try the wip/emmanuel/convert-enum branch.

TIS-Edgar, 2017-03-16:
The branch seems to fix all issues concerning this. We could not find any problems concerning expressions.

EmmanuelP, 2017-03-16:
Ok. I have merged the branch in master.

Thanks.

closed 2017-03-16 by EmmanuelP