AravisProject/aravis #53: Cannot install on macOS Sierra

jschwalbe, 2017-03-14:


EmmanuelP, 2017-03-14:
Aravis was patched a few month ago to add support for MacOS. Compilation on this platform is supposed to have  __APPLE__ defined. No idea how it is supposed to happen though, as I don't have an Apple machine at hand.

EmmanuelP, 2017-03-17:
Any news ?

You may ant to try to ask on aravis maiiing list. Some subscribers are may be using MacOS.

https://www.freelists.org/list/aravis

jschwalbe, 2017-03-18:
@EmmanuelP Thanks for the reply. I replied to your other question on the mailing list. I believe I was able to get it compiled, but now I'm not so sure that I actually have a GigE camera, so I'm not sure if 1) I compiled wrong 2) I'm using it wrong or 3) I have the wrong camera type!

In terms of compiling, it was ugly. Had to copy some libs from other apps. One version wouldn't allow me to go thru the configure process but then I was able to copy files from the one that did, etc. But eventually it built.

Thank you.

EmmanuelP, 2017-03-18:
I'm closing this issue. Feel free to reopen it if you really want to do a clean install and fail to correctly compile aravis on MacOS.

closed 2017-03-18 by EmmanuelP