AravisProject/aravis #54: EnumEntry is not listed.

TIS-Edgar, 2017-03-16:
Hi,

When using a TIS 33GX149e the enumeration "DeviceTemperatureSelector" is not correctly handled.
arv_device_get_available_enumeration_feature_values_as_strings only lists:
    Mainboard
The enum has the entries "Mainboard" and "Sensor".

When querying the current value ```arv-tool-0.6 control DeviceTemperatureSelector -n "The Imaging Source-42619966"``` returns ```DeviceTemperatureSelector = Sensor```.

You can find the xml description [here](https://gist.github.com/TIS-Edgar/0d7f1f502ea8f5582611c445334cfc62). This has been tested on the wip/emmanuel/convert-enum branch.

EmmanuelP, 2017-03-16:
Please attach the output of `arv-tool-0.6 control DeviceTemperatureSelector -n "The Imaging Source-42619966" -d genicam:3,evaluator:3`

Also, what returns `arv-tool-0.6 control R[0xF0000040]` ?

TIS-Edgar, 2017-03-16:
```
The Imaging Source-42619966 (192.168.0.143)
[Gc::register_feature_node] Register node 'Root' [Category]
[Gc::register_feature_node] Register node 'DigitalIO' [Category]
[Gc::register_feature_node] Register node 'AutoFunctionsROI' [Category]
[Gc::register_feature_node] Register node 'ColorTransformationControl' [Category]
[Gc::register_feature_node] Register node 'AnalogControls' [Category]
[Gc::register_feature_node] Register node 'ImageFormat' [Category]
[Gc::register_feature_node] Register node 'ChunkData' [Category]
[Gc::register_feature_node] Register node 'ChunkDataControl' [Category]
[Gc::register_feature_node] Register node 'ActionControl' [Category]
[Gc::register_feature_node] Register node 'UserSetControl' [Category]
[Gc::register_feature_node] Register node 'DeviceControl' [Category]
[Gc::register_feature_node] Register node 'AcquisitionControl' [Category]
[Gc::register_feature_node] Register node 'Device' [Port]
[Gc::register_feature_node] Register node 'ChunkImagePort' [Port]
[Gc::register_feature_node] Register node 'TLParamsLocked' [Integer]
[Gc::register_feature_node] Register node 'StrobeModeRegisterAddress' [Integer]
[Gc::register_feature_node] Register node 'StrobeDelay' [Integer]
[Gc::register_feature_node] Register node 'StrobeDuration' [Integer]
[Gc::register_feature_node] Register node 'StrobeMode' [Integer]
[Gc::register_feature_node] Register node 'GPOut' [Integer]
[Gc::register_feature_node] Register node 'GPIn' [Integer]
[Gc::register_feature_node] Register node 'GPIORegisterAddress' [Integer]
[Gc::register_feature_node] Register node 'GPIO' [Integer]
[Gc::register_feature_node] Register node 'TriggerDelayRaw' [Integer]
[Gc::register_feature_node] Register node 'AcquisitionBurstFrameCount' [Integer]
[Gc::register_feature_node] Register node 'TriggerBurstCountMax' [Integer]
[Gc::register_feature_node] Register node 'GammaRaw' [Integer]
[Gc::register_feature_node] Register node 'AutoFunctionsROIHeight' [Integer]
[Gc::register_feature_node] Register node 'AutoFunctionsROIWidth' [Integer]
[Gc::register_feature_node] Register node 'AutoFunctionsROITop' [Integer]
[Gc::register_feature_node] Register node 'AutoFunctionsROILeft' [Integer]
[Gc::register_feature_node] Register node 'AutoFunctionsROISizeMin' [Integer]
[Gc::register_feature_node] Register node 'PixelFixData' [Integer]
[Gc::register_feature_node] Register node 'PixelFixDataIndex' [Integer]
[Gc::register_feature_node] Register node 'PixelFixMaxPixelCount' [Integer]
[Gc::register_feature_node] Register node 'PixelFixDataBaseAddress' [Integer]
[Gc::register_feature_node] Register node 'PixelFixPixelCountAddress' [Integer]
[Gc::register_feature_node] Register node 'PixelFixControlAddress' [Integer]
[Gc::register_feature_node] Register node 'ExposureAutoMaxRaw' [Integer]
[Gc::register_feature_node] Register node 'ExposureTimeMaxRaw' [Integer]
[Gc::register_feature_node] Register node 'ExposureTimeMinRaw' [Integer]
[Gc::register_feature_node] Register node 'ExposureTimeRaw' [Integer]
[Gc::register_feature_node] Register node 'ExposureAutoReference' [Integer]
[Gc::register_feature_node] Register node 'GainRawMax' [Integer]
[Gc::register_feature_node] Register node 'GainRawMin' [Integer]
[Gc::register_feature_node] Register node 'GainRawHidden' [Integer]
[Gc::register_feature_node] Register node 'Brightness' [Integer]
[Gc::register_feature_node] Register node 'BrightnessMax' [Integer]
[Gc::register_feature_node] Register node 'BrightnessMin' [Integer]
[Gc::register_feature_node] Register node 'HueRaw' [Integer]
[Gc::register_feature_node] Register node 'SaturationRaw' [Integer]
[Gc::register_feature_node] Register node 'VCDProperty_WhiteBalanceBlue' [Integer]
[Gc::register_feature_node] Register node 'VCDProperty_WhiteBalanceGreen' [Integer]
[Gc::register_feature_node] Register node 'VCDProperty_WhiteBalanceRed' [Integer]
[Gc::register_feature_node] Register node 'BalanceWhiteTemperature' [Integer]
[Gc::register_feature_node] Register node 'WhiteBalanceRegisterAddressBlue' [Integer]
[Gc::register_feature_node] Register node 'WhiteBalanceRegisterAddressGreen' [Integer]
[Gc::register_feature_node] Register node 'WhiteBalanceRegisterAddressRed' [Integer]
[Gc::register_feature_node] Register node 'WhiteBalanceMaxRaw' [Integer]
[Gc::register_feature_node] Register node 'WhiteBalanceMinRaw' [Integer]
[Gc::register_feature_node] Register node 'WhiteBalanceNeutralRaw' [Integer]
[Gc::register_feature_node] Register node 'BalanceRatioRaw' [Integer]
[Gc::register_feature_node] Register node 'BalanceRatioSelectorBackEnd' [Integer]
[Gc::register_feature_node] Register node 'SensorNeedsGreenPatternFix' [Integer]
[Gc::register_feature_node] Register node 'PixelFormatValue' [Integer]
[Gc::register_feature_node] Register node 'TestPixelFormat' [Integer]
[Gc::register_feature_node] Register node 'TestHeight' [Integer]
[Gc::register_feature_node] Register node 'TestWidth' [Integer]
[... 557 lines cut ...]
[Evaluator::set_double_variable] E = 2.71828
[Gc::register_feature_node] Register node 'GevSCPAddrCalc' [IntSwissKnife]
[Gc::register_feature_node] Register node 'GevStreamChannelSelector' [Integer]
[Evaluator::set_double_variable] PI = 3.14159
[Evaluator::set_double_variable] E = 2.71828
[Gc::register_feature_node] Register node 'GevStreamChannelSelectorMax' [IntSwissKnife]
[GcEnumeration::get_string_value] value = 0 - string = Sensor
DeviceTemperatureSelector = Sensor
[... cut ...]
[GcRegisterNode::_get_integer_value] address = 0xf0000040, value = 0x1
[GcRegisterNode::_get_integer_value] reglsb = 30, regmsb, 30, lsb = 1, msb = 1
[GcRegisterNode::_get_integer_value] value = 0x00000001
[GcRegisterNode::_get_integer_value] mask  = 0x00000002
[GcRegisterNode::_get_integer_value] address = 0xf0000040, value = 0x0
    Mainboard

```

```arv-tool-0.6 control R[0xF0000040] -n "The Imaging Source-42619966" ```
Returns:
```
The Imaging Source-42619966 (192.168.0.143)
R[0xf0000040] = 0x00000001```

EmmanuelP, 2017-03-16:
`R[0xf0000040] = 0x00000001`

The Sensor entry has an isImplemented node which depends on bit 30 of register 0xf0000040. As the bit is not set, the entry is reported as not implemented, and hence not listed.

The problem is arv-tool-0.6 control DeviceTemperatureSelector returns Sensor, as the default value for the enumeration is 0. May be there is something in the specification that says what to do in this case.

TIS-Edgar, 2017-03-16:
Ok, I just talked to our firmware developer.

The default value is statically defined in the genicam description, thus can not be changed easily.
Newer cameras only have the sensor on the mainboard, thus the genicam description will be changed to have 'mainboard' as the default to fit the rest of the description.

Basically, blame the standard for not allowing good genicam descriptions.

As for aravis, would it make sense to give the user a warning when an enumentry is not actually present in the enum?

EmmanuelP, 2017-03-16:
In this case, I think the best solution in Aravis is to check the node type pointed by pValue, and if it is an <Integer>, force the value to stay in the allowed ones.

EmmanuelP, 2017-03-16:
I have added some checks in arvgcenumeration to ensure the value set is actually available. But it will not fix your issue, as the user has first to set the enumeration to a valid value before reading the temperature. There is nothing realistic I can do in aravis to prevent a read on an unavailable sensor.

It is in wip/emmanuel/enumeration-checks branch.

I think it would be better if your firmware use a default enumeration value that does not correspond to an unimplemented feature.

TIS-Edgar, 2017-03-17:
> I think it would be better if your firmware use a default enumeration value that does not correspond to an unimplemented feature.

Which what has been done. Thank you for dealing with this.

closed 2017-03-17 by EmmanuelP