AravisProject/aravis #55: Aravis support for GenAPI version 3.01

YarochkinMichael, 2017-03-17:
arv_buffer_get_payload_type () is always returning -1 (ARV_BUFFER_PAYLOAD_TYPE_UNKNOWN)

camera is:
MatrixVision mvBlueCOUGAR-x102eC
part no 08787
rev 1.04
firmware version is 2.18.651.0
GenAPI version 3.01 standard.

Our application is working well  with previous version of the same camera model, but different firmware 1.6.1.

[versionInfo.txt](https://github.com/AravisProject/aravis/files/850749/versionInfo.txt)

EmmanuelP, 2017-03-17:
Please attach the output of: tests/arv-camera-test -d all:2

YarochkinMichael, 2017-03-20:
Hello

Here it is.

[test.txt](https://github.com/AravisProject/aravis/files/854277/test.txt)

EmmanuelP, 2017-07-06:
I just had a quick look on the log files.

Payload size is strange in camera_test_GX006810_aravis_0.6.txt and camera_test_GX016564_aravis_0.4.txt. With 1280*1024, we expect a payload of 1310720 or  2621440. 3932160 is strange. It is not a formula issue, as the payload is a direct register read.

camera_test_GX016564_aravis_0.6.txt, there is:

```
GvDevice::create_stream] Can't create stream without control access
Can't create stream thread (check if the device is not already used)
```

Which seems to indicate another process has already a control access lock on the device.

YarochkinMichael, 2017-09-21:
And why it is strange?
Is`t it size of RGB image for 1280*1024?

And who can be a "another process has already a control access lock on the device"? It is embedded device. No access from outside.

EmmanuelP, 2017-09-21:
> And why it is strange?
> Is`t it size of RGB image for 1280*1024?

You are right. I don't know what I was thinking.

I'm afraid I will not be able to help. It is hard to debug without the actual device at hand.

EmmanuelP, 2017-09-21:
You may want to check the log with ARV_DEBUG=thread:3,stream-thread:3

YarochkinMichael, 2017-12-06:
Not really.
Had no time to back to this project.
Sorry.
I will try to do it this year.

EmmanuelP, 2017-12-06:
Ok. No problem.

I'm closing this issue for now. Please reopen when you'll have more informations.

Thanks.

YarochkinMichael, 2018-04-24:
I have an update.
Sorry for delay.

This log is done with ARV_DEBUG=thread:3,stream-thread:3.
If my first line is correct

In my app on call back i am receiving only once ARV_STREAM_CALLBACB_TYPE_START_BUFFER

[NewCamera.txt](https://github.com/AravisProject/aravis/files/1942716/NewCamera.txt)

EmmanuelP, 2018-04-24:
```
[GvStream::finalize] n_received_packets     = 1849409
[GvStream::finalize] n_ignored_packets      = 1849409
[GvStream::finalize] n_duplicated_packets   = 1849408
```
For some reason aravis is not able to understand incoming packets. It would be nice if you could attach a network capture of the gigevision stream, using wireshark.

YarochkinMichael, 2018-04-24:
It is easy to say.  Can you guide me, please? Taking in account what my Linux level 3 of 10.
I am using Debian 9

YarochkinMichael, 2018-04-24:
I managed it! (with help)

[short export.zip](https://github.com/AravisProject/aravis/files/1942818/short.export.zip)

EmmanuelP, 2018-04-24:
That is almost what I need. Your trace only contains the GVCP packets (the control ones). I am more interested in the GVSP packets (the streaming ones). I don't think it is possible to filter them using wireshark. You'll probably have to record all the packets, then select a small sample of packets coming to aravis streaming port (`[GvStream::stream_new] Destination stream port` in the debug output).

YarochkinMichael, 2018-04-24:
[export 400-500.zip](https://github.com/AravisProject/aravis/files/1942907/export.400-500.zip)

EmmanuelP, 2018-04-24:
Thanks a lot for the packet captures. Unfortunately, it looks like the support of this camera requires the implementation of what is called in wireshark leader, payload and trailer image packets with extended ids. So no quick fix :(

I don't know if I will have to work on this support in the near future.

If by chance you want to work on this and contribute to aravis, please keep in mind you should not use GigEVision specification documents. But using the wireshark, or the wireshark GVSP dissector source code is fine.

YarochkinMichael, 2018-04-24:
Thank you for effort.

Maybe i will try.
Can you give me direction of where to look for this?
How "no quick" is this fix? Can you give raw estimation of required effort?

We have a communication with camera manufacturer about this issue.
They are interested to solve this incompatibility.
Can you give as much as possible details of what is the problem. Maybe they can fix this on them side.

EmmanuelP, 2018-04-24:
I guess for someone not familiar with aravis source code, but with C and Glib based development knowledge, it should not take more than a few days of work.
Most of the work should happen in arvgvsp.[ch] and arvgvstream.[ch].
The problem is the bitfields in the beginning of the leader, payload and trailer image packets emited by your MatrixVision camera are different than what aravis understand. The packet type aravis expects is listed in the ArvGvspContentType enum (0x01, 0x02, 0x03) in arvgvsp.h. The MatrixVision content types are 0x81, 0x82, 0x83.
The bitfield structure of the packet headers should be documented in the wireshark dissector sources.
In order to add support for these new packet types, the code must be adapted to read the block ids (called frame id in aravis) and packet ids correctly with respect to the content type id. The block ids can be tricky, because currently it is a 16 bit value with 0 as a forbidden value. In the MatrixVision packets, it is a 64 bit value, and I don't know if 0 is a valid one or not.
May be the code should also be adapted regarding the content of the leader and trailer packets.

StefanBattmer, 2018-04-25:
Hi guys,

I am a developer at MATRIX VISION. I guess I can provide some help here:

The block IDs EmmanuelP is referring to have been change with the release of the GigE Vision standard 2.0. Previous versions did use a 16 bit block ID to tag blocks of data but as these tend to roll over quite fast newer version of the standard use 64 bit now. Because of that the size and layout of certain network packets did change and a host software needs to support this if it wants to operate GigE Vision 2.0 and greater devices. However a GigE Vision device is only supposed to actually use the new 64-bit format is a certain bit in it's bootstrap registers is set. The factory default of our cameras for this behaviour is:

Firmware < 2.18: Always 16-bit block ID (GigE Vision 1.2)
Firmware >= 2.18 - 2.24 : Start with 64-bit block ID (GigE Vision 2.0), allow to set back to 16-bit
Firmware >= 2.25 : Start with 16-bit block ID (GigE Vision 2.1), allow to set to 64-bit

So assuming you have a device with a firmware version 2.18 - 2.24 and a host software that cannot deal with 64-bit block IDs you may have problems. One fix then would be to set the 'GevGVSPExtendedIDMode' feature of the device to 'Off' before starting to stream data. Another option would be to up/downgrade the firmware of the device to a compatible version using our Firmware update tool: https://www.matrix-vision.com/manuals/mvBlueCOUGAR-X/GUI_page_0.html#GUI_section_mvDeviceConfigure

Hope that helps!

YarochkinMichael, 2018-04-25:
Hi, Stefan

Thank you for reply.

Another solution is to prepare Aravis to recognize and handle correctly such packets.

StefanBattmer, 2018-04-25:
Hi Michael,

you will find all the relevant information inside the latest version of GigE Vision specification. Unfortunately to be able to deal with GigE Vision 2.0 devices properly it is more than just a bit somewhere. All the packet headers will use a slightly different layout.

You can get the GigE Vision specification here: https://www.visiononline.org/vision-standards-details.cfm?id=168&type=5

You might also look at the current master branch of wireshark to see what is needed:

https://github.com/wireshark/wireshark/blob/master/epan/dissectors/packet-gvsp.c
https://github.com/wireshark/wireshark/blob/master/epan/dissectors/packet-gvcp.c

EmmanuelP, 2018-04-25:
Hi Stefan,

Thanks for the comments. I have deleted the link to the AIA specification documents in your comment as, as far as I know (and the last time I have asked to the AIA), it is not possible to use them to contribute to an open source project like Aravis.

EmmanuelP, 2018-04-25:
@YarochkinMichael ,

Until the support for extended ids is added to aravis, I guess we can disable this mode for now as suggested by @StefanBattmer . I have pushed a branch that does that: https://github.com/AravisProject/aravis/tree/wip/disable-extended-ids

Could you test it please, adn tell me if it makes your camera work ?

YarochkinMichael, 2018-04-30:
Hello, Emmanuel

Thank you for effort.
But i prefer to keep main brunch and configure it in start up:

//disable 64 bit headers
//https://github.com/AravisProject/aravis/issues/55
    ArvGcNode *gevGVSPExtendedIDMode = arv_device_get_feature(this->device, "GevGVSPExtendedIDMode");
    if(ARV_IS_GC_FEATURE_NODE(gevGVSPExtendedIDMode) && arv_gc_feature_node_is_implemented(ARV_GC_FEATURE_NODE(gevGVSPExtendedIDMode), NULL)) {
      if(ARV_IS_GC_ENUMERATION(gevGVSPExtendedIDMode)) {
        arv_gc_enumeration_set_string_value(ARV_GC_ENUMERATION(gevGVSPExtendedIDMode), "Off", NULL);
      }
    }

EmmanuelP, 2018-04-30:
> But i prefer to keep main brunch 

I don't understand.

YarochkinMichael, 2018-05-02:
It is typo: brunch -> branch. (yes my English not perfect. sorry)
I wan to continue on origin/master and add setting to camera init function, like i have with other presetted parameters.

EmmanuelP, 2018-05-02:
Ok, I get it now.
I put the fix in a separate branch just for testing. My goal was to merge it to master after you have tested it. I will do a release in the next coming days.

YarochkinMichael, 2018-05-02:
Ok.
Sorry for asking this question and here. Probably i have to remove it from this chain later, but i have another problem with this camera:
When i change ExposureTime acquisition is stopping, but not immediately. after about 1 sec of streaming.
-can i ask something like this from you?
-should i make new issue? i do not think it is Aravis problem

EmmanuelP, 2018-05-02:
It is most probably not an aravis issue, but one never knows.
Please open a new issue, and attach the console output when your application is running with `ARV_DEBUG` set to `stream:3,device:3`.

EmmanuelP, 2018-05-02:
I'm pushing the fix to master.

closed 2018-05-02 by EmmanuelP