AravisProject/aravis #57: Gstreamer plugin 'bayer2rgb' is missing. (on Ubuntu 14.04)

drewm1980, 2017-03-20:
I compiled the latest version on an Ubuntu 14.04 machine. 

> commit 64bbbf8b265fc8f961fa0427723f139222188f2f
> Author: Emmanuel Pacaud <emmanuel@gnome.org>
> Date:   Sun Mar 19 11:45:55 2017 +0100
>     tests: put longer tests at the end of the list

When I try to run the viewer it is unable to find the bayer2rgb plugin, although gst-inspect reports that it is installed:

> awagner@hal:/home/awagner/src/aravis/viewer
> $ ./arv-viewer 
> 
> ** (lt-arv-viewer:12663): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-H6XDrK3475: Connection refused
> Gstreamer plugin 'bayer2rgb' is missing.
> Check your gstreamer installation.
> 
> awagner@hal:/home/awagner/src/aravis/viewer
> $ gst-inspect | grep bayer
> bayer:  bayer2rgb: Bayer to RGB decoder for cameras
> bayer:  rgb2bayer: RGB to Bayer converter

Among the other dependencies, I did install the gstreamer "bad" plugins set, which contains the bayer conversion plugins:

> $ aptitude search gstreamer-plugins-bad
> i   libgstreamer-plugins-bad0.10-0                               - GStreamer shared libraries from the "bad" set                          
> p   libgstreamer-plugins-bad0.10-0:i386                          - GStreamer shared libraries from the "bad" set                          
> i   libgstreamer-plugins-bad0.10-dev                             - GStreamer development files for libraries from the "bad" set           
> p   libgstreamer-plugins-bad0.10-dev:i386                        - GStreamer development files for libraries from the "bad" set           
> i   libgstreamer-plugins-bad1.0-0                                - GStreamer development files for libraries from the "bad" set           
> p   libgstreamer-plugins-bad1.0-0:i386                           - GStreamer development files for libraries from the "bad" set           
> i   libgstreamer-plugins-bad1.0-dev                              - GStreamer development files for libraries from the "bad" set           
> p   libgstreamer-plugins-bad1.0-dev:i386                         - GStreamer development files for libraries from the "bad" set           

I'm new to gstreamer, but it seems I can launch a gstreamer pipeline containing bayer2rgb:

> $ gst-launch "bayer2rgb"
> Setting pipeline to PAUSED ...
> Pipeline is PREROLLED ...
> Setting pipeline to PLAYING ...
> New clock: GstSystemClock

I suspect it is unrelated, but to get through the configure script I had to lower the required version of gtk+-3.0 to >= 3.10 in configure.ac.

Thanks!

EmmanuelP, 2017-03-20:
> gst-inspect | grep bayer

Please try  with gst-inspect-1.0

EmmanuelP, 2017-03-22:
@drewm1980 Any news ?

EmmanuelP, 2017-04-10:
I'm closing this issue. Please reopen if you have the requested informations.

closed 2017-04-10 by EmmanuelP

EFraim, 2018-01-15:
Both gst-inspect and gst-inspect-0.10 don't have any mention of bayer on Ubuntu 16.04

EFraim, 2018-01-15:
Sorry for the noise, the gst-inspect indeed displays nothing but the application works when gstreamer1.0-plugins-bad is installed