AravisProject/aravis #58: How to init Aravis on Ubuntu 16.04

TDaoust, 2017-03-28:
Hello,

I have installed Avaris 0.4 and 0.6 and arv-viewer from 0.6. When I open arv-viewer I see my camera: Teledyne Dalsa Nano-M1940.

When I try to view the image stream I see nothing.

I have run the viewer in debug mode: `$ arv-viewer -d all`

I notice a few interesting things:
Failed to connect make realtime: org.freedesktop.DBus.Error.AccessDenied
[GvStream::check_frame_completion] Timeout for frame 1 at dt = 133258

Here is the full output: [arv-viewer_debug.txt](https://github.com/AravisProject/aravis/files/876352/arv-viewer_debug.txt)

Note: I am trying to get the camera to work in ROS using [camera_avaris](https://github.com/takiaine/camera_aravis), which uses Avaris 0.4. But at this point I can't get anything to work.

Thanks for the help.

matheusns, 2017-03-29:
Hello @TDaoust  I'm having the same problem. When I run  $ arv-viewer -d all both stream on screen and message error appear. I'm using Teledyne Dalsa Genie Nano C2590.

The message error is something like yours:

[GvStream::stream_new] Packet size = 1500 byte(s)
[GvStream::stream_new] Destination stream port = 35284
[GvStream::stream_new] Source stream port = 38787
[GvStream::stream_thread] Packet timeout = 20 ms
[GvStream::stream_thread] Frame retention = 100 ms
[GvStream::loop] Standard socket method
[GvStream::check_frame_completion] Timeout for frame 1 at dt = 151121
[GvStream::check_frame_completion] Timeout for frame 2 at dt = 150996
[GvStream::check_frame_completion] Timeout for frame 3 at dt = 150782
[GvStream::check_frame_completion] Timeout for frame 4 at dt = 150950
[GvStream::check_frame_completion] Timeout for frame 5 at dt = 150928
[GvStream::check_frame_completion] Timeout for frame 6 at dt = 150852
[GvStream::check_frame_completion] Timeout for frame 7 at dt = 150790
[GvStream::check_frame_completion] Timeout for frame 8 at dt = 150916
[GvStream::check_frame_completion] Timeout for frame 9 at dt = 150894

Here's the full output: 
[arv-viewer-Nano_debug.txt](https://github.com/AravisProject/aravis/files/879065/arv-viewer-Nano_debug.txt)

Thanks!

EmmanuelP, 2017-03-29:
It looks like there is one or several missing expected packets. May be the payload size is miscomputed.

Could you attach the output of:

./tests/arv-camera-test -d sp:3

TDaoust, 2017-03-29:
Thanks for the quick reply. I have run the command you suggested using both avaris_0.4.1 and 0.5.7 as I have installed both. I also notice that the gain is set to an absurdly large number.

I have avaris 0.4.1 installed because that is the latest version I found the ROS node works with, however it is also runs into what seems to be the same issue, there is an error that comes up at every frame grab,   Frame error: ARV_BUFFER_STATUS_TIMEOUT.

Note I only installed avaris-viewer when I installed 0.5.7. And the avaris-viewer command is not version controlled what will happen if I try to reinstall avaris 0.4.1 with the viewer?

Thanks for your help.

The outputs are below:

rtuser@rtuser-ThinkPad-P70:~/Downloads/aravis-0.4.1/tests$ ./arv-camera-test -d sp:3
Looking for the first available camera
vendor name           = Teledyne DALSA
model name            = Nano-M1940
device id             = S1111692
image width           = 1936
image height          = 1216
horizontal binning    = 0
vertical binning      = 0
payload               = 0 bytes
exposure              = 15000 µs
gain                  = -2147483648 dB
gv n_stream channels  = 1
gv current channel    = 0
gv packet delay       = 0 ns
gv packet size        = 1500 bytes
Frame rate = 50 Hz
Frame rate = 30 Hz
Frame rate = 30 Hz
Frame rate = 30 Hz
Frame rate = 30 Hz
Frame rate = 30 Hz
Frame rate = 30 Hz
Frame rate = 30 Hz
Frame rate = 30 Hz
Frame rate = 30 Hz
Frame rate = 30 Hz
Frame rate = 30 Hz
Frame rate = 30 Hz
^CFrame rate = 30 Hz
Completed buffers = 440
Failures          = 0
Underruns         = 0
rtuser@rtuser-ThinkPad-P70:~/Downloads/aravis-0.4.1/tests$ cd ../../aravis-0.5.7/tests/
rtuser@rtuser-ThinkPad-P70:~/Downloads/aravis-0.5.7/tests$ ./arv-camera-test -d sp:3
Looking for the first available camera
vendor name           = Teledyne DALSA
model name            = Nano-M1940
device id             = S1111692
image width           = 1936
image height          = 1216
horizontal binning    = 0
vertical binning      = 0
payload               = 2355640 bytes
exposure              = 15000 µs
gain                  = -2147483648 dB
gv n_stream channels  = 1
gv current channel    = 0
gv packet delay       = 0 ns
gv packet size        = 1500 bytes
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
^CFrame rate = 0 Hz
Completed buffers = 0
Failures          = 252
Underruns         = 0

EmmanuelP, 2017-03-30:
image width           = 1936
image height          = 1216

payload               = 2355640 bytes

but 1936*1216 = 2354176

So there is something in the calculation of the payload.

What is the output of ./Sources/aravis/src/arv-tool-0.6 control PayloadSize -d all:3 ?

TDaoust, 2017-03-30:
Hi @EmmanuelP,

I ran the arv-tool from both version 0.4 and 0.6, again I will ultimately be trying to get v0.4 to work, but hopefully it is the same issue with both.

Please see the attached outputs from the two commands:  [arv_tool_0.4_payload.txt](https://github.com/AravisProject/aravis/files/882748/arv_tool_0.4_payload.txt) [arv_tool_0.6_payload.txt](https://github.com/AravisProject/aravis/files/882749/arv_tool_0.6_payload.txt)

Thanks again for the quick reply.

EmmanuelP, 2017-04-10:
I had a look at the logs you have attached.

The problem is the formula used for the computation of the payload size:

(( DATA_SIZE / DATA_PACKET_SIZE) + ((DATA_SIZE % DATA_PACKET_SIZE)?1:0)) * DATA_PACKET_SIZE

makes the assumption (DATA_SIZE / DATA_PACKET_SIZE) is an integer division.

Unfortunately, the formulas in the Genicam standard are under specified. It would be safer if the expression used the FLOOR function.

Please contact the manufacturer of your camera.

sputnick1124, 2017-05-24:
@EmmanuelP, I'm dealing with the same issues here with a Dalsa C4020. Are you saying that the xml camera description needs to be changed to use a different formula, or is there some issue with the firmware onboard?Forgive me if that's a stupid question, but I am a bit new to the Genicam world. If it helps, here is the relevant section of the camera description file:
```xml
      <IntSwissKnife Name="PayloadSizeVal">
         <pVariable Name="PACKET_SIZE">pGevSCPSPacketSizeReg</pVariable>
         <pVariable Name="PIXEL_SIZE">PixelSizeVal</pVariable>
         <pVariable Name="WIDTH">Width</pVariable>
         <pVariable Name="HEIGHT">Height</pVariable>
         <pVariable Name="MAX_CHUNK_SIZE">max_chunk_size</pVariable>
         <pVariable Name="CHUNK_MODE">chunkModeActive_reg</pVariable>
         <Expression Name="DATA_PACKET_SIZE">PACKET_SIZE-36</Expression>
         <Expression Name="DATA_SIZE">(WIDTH * HEIGHT * PIXEL_SIZE / 8) + ((CHUNK_MODE)?MAX_CHUNK_SIZE:0)</Expression>
         <Formula>(( DATA_SIZE / DATA_PACKET_SIZE) + ((DATA_SIZE % DATA_PACKET_SIZE)?1:0)) * DATA_PACKET_SIZE</Formula>
         <Representation>Linear</Representation>
      </IntSwissKnife>
```
Thanks for any help.

EmmanuelP, 2017-05-29:
@sputnick1124 yes I think the Genicam data for these camera should be fixed to not use a behaviour that is not explicitly defined by the Genicam specification. Genicam data are stored in the firmware.

But may be I'm wrong and there is somewhere in the documentation on explanation of the intended behaviour.

dnik2000, 2017-06-23:
Hello!
I have a troubles with the same camera
```
vendor name           = Teledyne DALSA
model name            = Nano-C1920
device id             = S1122838
image width           = 0
image height          = 0
horizontal binning    = 33554688
vertical binning      = 33554688
payload               = 0 bytes
exposure              = 15017 µs
gain                  = 2147483647 dB
gv n_stream channels  = 1
gv current channel    = 0
gv packet delay       = 2000 ns
gv packet size        = 1508 bytes
packet_type  = ok (0000)
content_type = block (0003)
.....
Frame rate = 21 Hz
Completed buffers = 86
Failures          = 33
Underruns         = 0
```

The same formula on last firmware v1.06beta. After writing values to Width, Height, Gain, I have corrupted data in these parameters.

sputnick1124, 2017-07-01:
@dnik2000 it looks like the latest v1.06 firmware (not beta) was released on June 5. Is this the firmware you tried? I have not yet uploaded it test and won't be able to for a few days. I will report back when I have.

dnik2000, 2017-07-04:
I tried to use Baumer SDK to work with the Teledyne DALSA Nano-C1920. And got the same errors with PayloadSize and with other parameters. In this camera very complicated formulas.
JAI Control Tool works well, but has problems with setting some parameters.

hap1961, 2017-07-04:
Make sure the buffers are allocated to PayloadSize bytes. PayloadSize is the mandatory GigeVision feature indicating the maximum number of bytes that are coming in the payload. For an image payload, the first width*height*pixel_size bytes will be image data (but PayloadSize bytes are coming). 
For this particular camera the PayloadSize value is correct.
The GVSP receiver either needs to dump/crop the extra bytes or ensure that the buffer is sized properly.

dnik2000, 2017-07-04:
[image]
I can't allocate so much memory for this PayloadSize

hap1961, 2017-07-05:
What is the value of the PayloadSize feature you are seeing before you write to the feature tree and corrupt it. (You mentioned above that as soon as you write to it you get bad values for gain, binning, width, etc....)

sputnick1124, 2017-07-05:
I've been in contact with the Teledyne Dalsa with regards to this issue and this is their reply:
"GVSP receivers (and GenTL providers) should recognize that the amount of data being transferred is the published in the mandatory GigeVision feature called PayloadSize.

Buffers to received data payloads need to be allocated to the size of PayloadSize.

The image contained with an image payload will have the height, width, pixel format, offsetx, and offset delivered with each frame.

For the case of variable length imaging, the trailer packet is used to provide the actually transmitted y size.

Even  though the “height*width*pixel_size” gives a smaller size than the PayloadSize feature, it is PayloadSize bytes that are being transferred."

hap1961, 2017-07-06:
It's an IntSwissKnife with all variables referenced being IInteger  - formulas using IInteger are always treated as integer (at least in GenICam Std v2.0 doc - page 13 (of 53)) with the conversion to integer done when the variable are accessed.

(If it was a SwissKnife, then it is unclear when a conversion to float would be done but I would assume it would be immediately on accessing the variables and before the formula is applied - like the integer case).

EmmanuelP, 2017-07-06:
Well, I think the standard is not clear about intSwissKnife. I understand it uses integer values of the variables, and produce an integer, but I didn't find a word saying the intermediate results should be handled as integer. But I guess it is what the reference implementation does.

sputnick1124, 2017-08-02:
Any progress on this issue? Am able to test if there is code to try out.

EmmanuelP, 2017-08-02:
I'm planning to work on this issue tomorrow. I'll let you know when there is something to test.

sputnick1124, 2017-08-02:
Sounds great. Thanks!

EmmanuelP, 2017-08-03:
Please try https://github.com/AravisProject/aravis/tree/wip/intswissknife

sputnick1124, 2017-08-03:
Seems to work for me! Thanks for your efforts.
```bash
$ ./tests/arv-camera-test -d sp:3
Looking for the first available camera
vendor name           = Teledyne DALSA
model name            = Nano-C4020
device id             = S1128527
image width           = 4112
image height          = 3008
horizontal binning    = 33554688
vertical binning      = 33554688
payload               = 12369336 bytes
exposure              = 15014 µs
gain                  = -2147483648 dB
gv n_stream channels  = 1
gv current channel    = 0
gv packet delay       = 2000 ns
gv packet size        = 1500 bytes
Frame rate = 4 Hz
Frame rate = 4 Hz
Frame rate = 4 Hz
^CFrame rate = 4 Hz
Completed buffers = 16
Failures          = 0
Underruns         = 0
```

[PayloadSize_Debug.txt](https://github.com/AravisProject/aravis/files/1197838/PayloadSize_Debug.txt)

Camera appears now with `arv-viewer` and I can save snapshots as well.
[image]

EmmanuelP, 2017-08-03:
Great.

Gain value is obviously wrong though. Could you send me privately the genicam data using this command: `arv-tool-0.6 genicam` please.

sputnick1124, 2017-08-03:
[Genia_Nano_c4020_Genicam.txt](https://github.com/AravisProject/aravis/files/1198114/Genia_Nano_c4020_Genicam.txt)

EmmanuelP, 2017-08-03:
Did not spot nayhting wrong in the genicam data. Could you attach the output of `arv-tool-0.6 control Gain --debug=evaluator:3` ?

sputnick1124, 2017-08-03:
[Gain_Debug.txt](https://github.com/AravisProject/aravis/files/1198195/Gain_Debug.txt)

EmmanuelP, 2017-08-03:
This features requires support for pValueIndexed. I'm opening a new issue about this one.

EmmanuelP, 2017-08-04:
Fixed by 39ce531155d8fdcbe71edfa76ddce769e72ea574

closed 2017-08-04 by EmmanuelP

sputnick1124, 2017-08-04:
Sorry to resurrect this issue, but why is it the case that  `payload  = 12369336 bytes` but
`(image height) * (image width) = 12368896`?

EmmanuelP, 2017-08-04:
I don't know. Sometimes payload is not exactly what is required to store the image. In this case, payload seems correct, as aravis doesn't complain it receives less data than expected (Failures=0).