AravisProject/aravis #60: aravis viewer shows blank picture

IamSierraCharlie, 2017-04-06:
Great project so far!
I am using the following camera:
vendor name           = ContrasTech
model name            = Mars 1300-210uc

For the most part, this camera seems to work fine, but on occasion, the camera shows a blank screen in the viewer.  I find that disconnecting and reconnecting the camera (sometimes a few times) gets it to work again.

Could anyone offer up some useful troubleshooting steps?
The only thing I've been able to find so far was when running 'arv-device-test'
I got the following output:
Looking for the first available camera
vendor        = ContrasTech
model         = Mars 1300-210uc

(lt-arv-device-test:948): Aravis-CRITICAL **: arv_gc_string_get_value: assertion 'ARV_IS_GC_STRING (gc_string)' failed
device id     = (null)
sensor width  = 1280
sensor height = 1024
image width   = 1280 (max:1280)
image height  = 1024 (max:1024)

(lt-arv-device-test:948): Aravis-CRITICAL **: arv_gc_integer_get_value: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed

(lt-arv-device-test:948): Aravis-CRITICAL **: arv_gc_integer_get_max: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed

(lt-arv-device-test:948): Aravis-CRITICAL **: arv_gc_integer_get_min: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed
horizontal binning  = 0 (min:0 - max:0)

(lt-arv-device-test:948): Aravis-CRITICAL **: arv_gc_integer_get_value: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed

(lt-arv-device-test:948): Aravis-CRITICAL **: arv_gc_integer_get_max: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed

(lt-arv-device-test:948): Aravis-CRITICAL **: arv_gc_integer_get_min: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed
vertical binning    = 0 (min:0 - max:0)
exposure            = 1000 (min:-1.79769e+308 - max:1.79769e+308)
exposure auto mode  = Off

I also note that the light on the back of the camera flashes as it its still grabbing frames.  
If i disconnect and reconnect the camera, the light returns to a non flashing state.
To be honest, I'm not sure if these 2 things are related.

Feedback appreciated
Steve

IamSierraCharlie, 2017-04-06:
Just to follow up - other errors when running the same test
gain                = 0 (min:0 - max:0)
gain auto mode      = Off
trigger selector    = FrameStart
reverse x          = FALSE
payload size  = 1310720 (0x140000)

(lt-arv-device-test:1103): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_TIMEOUT

(lt-arv-device-test:1103): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_TIMEOUT

(lt-arv-device-test:1103): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_TIMEOUT

(lt-arv-device-test:1103): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_TIMEOUT

(lt-arv-device-test:1103): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_TIMEOUT

(lt-arv-device-test:1103): GLib-GObject-WARNING **: invalid cast from 'ArvUvStream' to 'ArvGvStream'

IamSierraCharlie, 2017-04-06:
and one more - here its like it just stops giving frames
sudo ./arv-camera-test -w 1280 -h 1024 
Looking for the first available camera
vendor name           = ContrasTech
model name            = Mars 1300-210uc
device id             = (null)
image width           = 1280
image height          = 1024
horizontal binning    = 0
vertical binning      = 0
payload               = 1310720 bytes
exposure              = 1000 µs
gain                  = 0 dB
Frame rate = 282 Hz
Frame rate = 209 Hz
Frame rate = 84 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
^CFrame rate = 0 Hz
Completed buffers = 575
Failures          = 0
Underruns         = 0

EmmanuelP, 2017-04-10:
> (lt-arv-device-test:1103): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_TIMEOUT

This should be fixed by 92839b8e2e17b711285f039c6b8245d02720534c

> (lt-arv-device-test:948): Aravis-CRITICAL **: arv_gc_integer_get_value: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed

> (lt-arv-device-test:1103): GLib-GObject-WARNING **: invalid cast from 'ArvUvStream' to 'ArvGvStream'

And these ones by commit 95ceccbd86a060ae362c15fa07cd3d480fbf8f7d

> sudo ./arv-camera-test -w 1280 -h 1024

Please have a look at the USB permissions chapter in README. You should be able to use aravis without using sudo.

> Frame rate = 282 Hz
> Frame rate = 209 Hz
> Frame rate = 84 Hz
> Frame rate = 0 Hz

Could you attach the output of `./arv-camera-test -w 1280 -h 1024 -d all` ?

IamSierraCharlie, 2017-04-10:
Thanks for your  response Emmanuel,
Yes you are right - no need to run with sudo.
I installed the new code today.
Please find the log attached:
[aravis_log.txt](https://github.com/AravisProject/aravis/files/909394/aravis_log.txt)
Not sure how long I should be letting this run for?  If it needs more time to run, I'll do it again.

Let me know if there is anything else you need from me.  Happy to help where I can.

I also note:
I saw this when I ran the new code - I think it might have been in the older version too.
I ran arv-viewer -a all and the snippet blow was noted.
[ArvDevice::get_integer_feature_value] Node 'BinningHorizontal' is not an integer
[ArvDevice::get_integer_feature_value] Node 'BinningVertical' is not an integer
[ArvDevice::get_integer_feature_bounds] Node 'BinningHorizontal' is not an integer
[ArvDevice::get_integer_feature_bounds] Node 'BinningVertical' is not an integer

The "Binning" section on the aravis viewer is not changeable. 
I can raise a new issue for this if you need - once again forgive my ignorance here as I don't want to bombard you with a new issue if its not necessary.
Thanks 
Steve

EmmanuelP, 2017-04-10:
> Please find the log attached:
> aravis_log.txt
> Not sure how long I should be letting this run for? If it needs more time to run, I'll do it again.

Please run it until frame rate goes to 0.

> I saw this when I ran the new code - I think it might have been in the older version too.
> I ran arv-viewer -a all and the snippet blow was noted.
> [ArvDevice::get_integer_feature_value] Node [...]

That is fine, unless Binning is actually supported by your camera. Please check the camera documentation.

IamSierraCharlie, 2017-04-10:
Hi Emmanuel,
I'm in the process of running the requested log.
Presently, I have found that the camera appears to run fine when running the requested command...  i.e. the frame rate is maintained at 210Hz as it should.  I do note that if I run the arv-viewer first and then attempt the command ./arv-camera-test -w 1280 -h 1024 -d all,
After doing this, on most occasions, I see 0hz immediately as per this log attached - emphasis on "most" as the camera shows frames on occasion.  I've see it work without issue on 2 occasions after running the arv-viewer
[aravis_log1.txt](https://github.com/AravisProject/aravis/files/909600/aravis_log1.txt)
I'll keep collecting logs and let you know the outcome
I'll also check the camera documentation and come back to you about that too
Thanks again for your help Emmanuel!
Steve

EmmanuelP, 2017-04-10:
It would be interesting to get a log with '-d all,sp:3'.

IamSierraCharlie, 2017-04-10:
Sure!  Here you go.
When I started my machine and ran the command for the first time,  I got 0fps straight away, but the log was too big and I was unable to collect it from the screen....
No issues on this log though
[aravis_log2.txt](https://github.com/AravisProject/aravis/files/910033/aravis_log2.txt)
Here is a second log - once again, I ran arv-viewer first and then ran the command:
./arv-camera-test -w 1280 -h 1024 -d all,sp:3 > aravis_log3.txt
[aravis_log3.txt](https://github.com/AravisProject/aravis/files/910043/aravis_log3.txt)

The aravis_log4.txt, I ran with the command arv-viewer -d all,sp:3 > aravis_log4.txt
[aravis_log4.txt](https://github.com/AravisProject/aravis/files/910070/aravis_log4.txt)

Finally, I ran the original command again:
./arv-camera-test -w 1280 -h 1024 -d all,sp:3 > aravis_log5.txt
logs attached again
[aravis_log5.txt](https://github.com/AravisProject/aravis/files/910078/aravis_log5.txt)

Is there by chance a command line way to reset the camera?  Without needing to disconnect and reconnect it via the cable?

Thanks for looking at this Emmanuel
Let me know if you need anything else

Steve

EmmanuelP, 2017-04-10:
What platform are you using ?

It could be the same issue than the one fixed by https://github.com/AravisProject/aravis/pull/42

IamSierraCharlie, 2017-04-10:
Im on Ubuntu 16.04

IamSierraCharlie, 2017-04-10:
My version of lib usb:
libusb-dev:
Installed: 2:0.1.12-28
Candidate: 2:0.1.12-28
Version table:

libusb-1.0-0:
Installed: 2:1.0.20-1
Candidate: 2:1.0.20-1
Version table:
*** 2:1.0.20-1 500

IamSierraCharlie, 2017-04-10:
I've investigated some more on this issue this morning.
When running the viewer, running the camera seems to work okay as long as I don't click the 'x' to simply terminate the viewer.  Under normal operating conditions, there is a blue light on the back of the camera that remains solid when the camera is ready to operate.  When calling frames from the camera, that light flashed blue.  If I disconnect from the viewer mid capture, that light continues to flash after the software shutdown.  When I reload the viewer, it sees the camera, but the camera itself never recovers to provide any more frames.  
If I use the back arrow (return to camera list) in the viewer, it appears to gracefully terminate the image grabbing.  I then shut the program down and restarting the view doesn't appear to trigger this blank screen issue.  
I think this is one way to trigger the issue, but doesn't explain why it may occur the other times.
I was running the debug on the back of the viewer and managed to get this info:
[aravis_log6.txt](https://github.com/AravisProject/aravis/files/911704/aravis_log6.txt)

closed 2018-03-23 by IamSierraCharlie