AravisProject/aravis #61: Question - How to open raw pictures samed by arv-viewer

IamSierraCharlie, 2017-04-10:
The file extension is .raw and I can't seem to find anything in Ubuntu to open it.  I've tried shotwell and gimp, both do not show the pictures.
Any suggestions?

[Aravis - pics.tar.gz](https://github.com/AravisProject/aravis/files/911733/Aravis.-.pics.tar.gz)

EmmanuelP, 2017-04-11:
I don't know a generic software that can open all the pixel formats a camera can generate. It is more a debugging tool for now.

Mono8 and Mon16 can be opened using ImageJ.

It could be interresting to be able to save images to jpg or tiff files, but it is not implemented yet.

closed 2017-04-11 by EmmanuelP

IamSierraCharlie, 2017-04-11:
Thanks for your response Emmanuel,
Would it then be fair to say attempting to use python wrappers for the aravis project to grab an image and show it via OpenCV2 or Matplotlib would be a pointless exercise?  The reason I ask is because this is what I'm looking to do with my camera.  Or are you just saying that the "save" part of the arv-viewer is purely for debugging only.
Steve

pontusnordin, 2017-10-23:
@IamSierraCharlie were you able to open the raw files generated by arv-viewer?