AravisProject/aravis #62: Colour channels split

IamSierraCharlie, 2017-04-12:
Hi Emmanuel,
Wondering if you could offer your expert opinion on this one - not sure if it relates to your project or not, but I am using the aravis module in python to grab some images.

Here is my code on python 3.4
import sys
import aravis
import cv2
import numpy as np

cam = aravis.Camera()
cam.set_feature('Width', 1280)
cam.set_feature('Height', 720)
cam.set_feature('ExposureTime', 18000)
cam.set_feature('AcquisitionFrameCount', 1)
cam.set_feature('AcquisitionFrameRate', 60)
cam.set_feature('AcquisitionFrameRateEnable', True)
cam.set_feature('PixelFormat', 'BayerRG8')
cam.set_feature('PixelSize', 'Bpp8')
cam.set_feature('PixelColorFilter', 'BayerRG')

cam.start_acquisition()
frame = cam.pop_frame()
npimg = np.asarray(frame)
print(npimg.shape)
b = 'testpic.png'
print(b)
cv2.imwrite(b, npimg)
cam.stop_acquisition()

Here is the picture (see attachment)
[image]

I want to grab images to manipulate in opencv.
In this case, it looks like the colour channels are split and I'm not sure where to start looking.  The colour of the ball is supposed to be yellow (for reference)

Also note that the picture grabbed should be 1280 x 720 - the frame is the right size, but the image has shrunk

Appreciate any suggestion - just want to know if the issue might be related to the module or not and how I might go about proving it one way or another?

Thanks
Steve

EmmanuelP, 2017-04-12:
Hi Steve.

Please use the mailing list for this sort of questions.

You can not use directly the image buffer and feed OpenCV with it and expect it will work magically. I don't know what offers OpenCV, but at some point there is a format conversion needed from what the camera sends to what OpenCV expects, especially in the case of bayer pixel formats. The documentation of your camera probably explains the different pixel formats.

closed 2017-04-12 by EmmanuelP

EmmanuelP, 2017-04-12:
It looks like OpenCv can do the conversion:

http://docs.opencv.org/trunk/db/d8c/group__cudaimgproc__color.html