AravisProject/aravis #63: Error building aravis under RHEL6

hhslepicka, 2017-04-14:
Hi,

I'm trying to build the aravis library under RHEL6 and I am facing the following error:

```
Making all in src
make[1]: Entering directory `/afs/slac.stanford.edu/g/lcls/vol9/package/aravis/0.5.8/build/rhel6-x86_64/src'
make  all-am
make[2]: Entering directory `/afs/slac.stanford.edu/g/lcls/vol9/package/aravis/0.5.8/build/rhel6-x86_64/src'
  CC     arvgvstream.lo
../../../src/aravis-ARAVIS_0_5_8/src/arvgvstream.c:793: error: field 'h1' has incomplete type
../../../src/aravis-ARAVIS_0_5_8/src/arvgvstream.c: In function '_ring_buffer_loop':
../../../src/aravis-ARAVIS_0_5_8/src/arvgvstream.c:801: error: storage size of 'req' isn't known
../../../src/aravis-ARAVIS_0_5_8/src/arvgvstream.c:818: error: 'TPACKET_V3' undeclared (first use in this function)
../../../src/aravis-ARAVIS_0_5_8/src/arvgvstream.c:818: error: (Each undeclared identifier is reported only once
../../../src/aravis-ARAVIS_0_5_8/src/arvgvstream.c:818: error: for each function it appears in.)
../../../src/aravis-ARAVIS_0_5_8/src/arvgvstream.c:830: error: 'TP_FT_REQ_FILL_RXHASH' undeclared (first use in this function)
../../../src/aravis-ARAVIS_0_5_8/src/arvgvstream.c:890: error: dereferencing pointer to incomplete type
../../../src/aravis-ARAVIS_0_5_8/src/arvgvstream.c:898: error: dereferencing pointer to incomplete type
../../../src/aravis-ARAVIS_0_5_8/src/arvgvstream.c:801: warning: unused variable 'req'
make[2]: *** [arvgvstream.lo] Error 1
make[2]: Leaving directory `/afs/slac.stanford.edu/g/lcls/vol9/package/aravis/0.5.8/build/rhel6-x86_64/src'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/afs/slac.stanford.edu/g/lcls/vol9/package/aravis/0.5.8/build/rhel6-x86_64/src'
make: *** [all-recursive] Error 1
```

**Information about the system:**
**Kernel:** 2.6.32-573.18.1.el6.x86_64
**Glib version:** 2.52.0
**Config Line:** PKG_CONFIG_PATH=/afs/slac/g/lcls/package/glib/2.52.0/rhel6-x86_64/lib/pkgconfig/ ../../src/aravis-ARAVIS_0_5_8/autogen.sh --prefix=/afs/slac/g/lcls/package/aravis/0.5.8/rhel6-x86_64 --disable-gst-plugin --disable-viewer

The output of the configure and also full output of make can be found in here: https://gist.github.com/hhslepicka/ef73fc8e276ffd23f93dc1046028c0de

Thank you!

EmmanuelP, 2017-04-14:
That is probably because TPACKET_V3 is not supported by your kernel. Try --disable-packet-socket.

hhslepicka, 2017-04-14:
Thank you @EmmanuelP! That solved my issue! 👍

closed 2017-04-14 by hhslepicka