AravisProject/aravis #67: Connection issue with acA1920-155um (U3V) on Manjaro 16.10

vaxxol, 2017-05-11:
I am running on Manjaro with Kernel 4.4.19 or 4.9.?? and libusb 1.0.21 and I am getting thje following messages when connecting with the camera:

(lt-arv-viewer:24459): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_TIMEOUT 
(lt-arv-viewer:24459): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_TIMEOUT
(lt-arv-viewer:24459): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_TIMEOUT 
(lt-arv-viewer:24459): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_TIMEOUT
(lt-arv-viewer:24459): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_TIMEOUT 
(lt-arv-viewer:24459): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_TIMEOUT
(lt-arv-viewer:24459): Aravis-CRITICAL **: arv_device_read_memory: assertion 'buffer != NULL' failed 
(lt-arv-viewer:24459): Aravis-CRITICAL **: arv_dom_document_new_from_memory: assertion 'buffer != NULL' failed
(lt-arv-viewer:24459): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed 
….

(lt-arv-viewer:24459): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_IO
(lt-arv-viewer:24459): Aravis-WARNING **: [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_IO 
(lt-arv-viewer:24459): XML-CRITICAL **: Document is empty
(lt-arv-viewer:24459): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed

Other PtGrey-U3V cameras work w/o a problem - Good job !
The acA1920-155um works fine with Basler's pylon software.
Attached are logfiles with libusb debug messages turned on.
[aca1920-155-logs.tar.gz](https://github.com/AravisProject/aravis/files/994488/aca1920-155-logs.tar.gz)

Any ideas ?

EmmanuelP, 2017-05-12:
Some additional informations: 
* This camera works under fedora 25 (linux 4.10.13 - libusb 1.0.21)
* Even with Pylon, there is a lot of error 71 (protocol error). But Pylon seems to be able to recover from this errors

vaxxol, 2017-05-16:
Tried with Manjaro with Linux kernel 4.10.12 and libusb 1.0.21 - problems as described above.

eudoxos, 2017-10-13:
(Perhaps unrelated) acA1920-150um (not 155um) works fine here: libusb 1.0.21, inux 4.13.0).

EmmanuelP, 2018-08-10:
Did you finally succeed to make this device work using Manjaro ?

vaxxol, 2018-09-07:
Yes, I could make it work with a new USB3 cable as recommended by the Basler support. 
I can't say what effect the upgrade to  Kernel 4.4.136  and the latest aravis version had, but I stlll get the startup errors when I use the old non-Basler USB3Vision cable.

EmmanuelP, 2018-09-07:
Ok. I'm closing this issue then.

Thanks for the report.

Cheers.

closed 2018-09-07 by EmmanuelP