AravisProject/aravis #68: ArvBuffer has no member named 'data', "frame_id", 'size', 'timestamp_ns'

haseeb-uob, 2017-05-12:
Hello,

I am new with aravis and I got following errors. what can be the possible reason?
Using arv_get_device_id, I can see the camera name which means probably library is properly linked.

error: 'ArvBuffer' has no member named 'data'
error: 'ArvBuffer' has no member named 'frame_id'
error: 'ArvBuffer' has no member named 'size'
error: 'ArvBuffer' has no member named 'timestamp_ns'

Thanks in advance.

EmmanuelP, 2017-05-13:
Which version of aravis ? Which language ? Python ?

haseeb-uob, 2017-05-15:
Thanks for your response. aravis version 0.4, glib version 2.0, C++.

EmmanuelP, 2017-05-15:
Since 0.4, ArvBuffer members are private. They can be accessed using accessors like arv_buffer_get_data, arv_buffer_get_tiemstamp etc...

closed 2017-05-15 by EmmanuelP

haseeb-uob, 2017-05-15:
thanks. it's working now using following accessors.
guint64	arv_buffer_get_timestamp	(ArvBuffer *buffer);
guint32 	arv_buffer_get_frame_id 	(ArvBuffer *buffer);
const void *	arv_buffer_get_data	(ArvBuffer *buffer, size_t *size);

Is there accessor to get the buffer size? How can I change these lines for version 0.4

            ArvBuffer *pBuffer;
            std::vector<uint8_t> this_data(pBuffer->size);
            memcpy(&this_data[0], pBuffer->data, pBuffer->size);

EmmanuelP, 2017-05-15:
> Is there accessor to get the buffer size? 

It is the size parameter of `arv_buffer_get_data`

haseeb-uob, 2017-05-15:
Thank you so much. I successfully compiled the code but in run-time, i am getting an error and can't see any image.
ARV_BUFFER_STATUS_MISSING_PACKETS
what can be the reason of this?

haseeb-uob, 2017-05-15:
I think the reason is image size is very big (2592x1944) and it's missing some frames. 
I am getting only black screen probably the conversion of buffer_data (char *) to this_data (std::vector<uint8_t>) is not correct.
I am converting the buffer_data into ROS supported image data type as shown below: 
   
    ArvBuffer *pBuffer;
    char *buffer_data;
    size_t buffer_size=0;
            buffer_data = (char *) arv_buffer_get_data (pBuffer, &buffer_size);
            std::vector<uint8_t> this_data(buffer_size);
            memcpy(&this_data[0], buffer_data, buffer_size);

Please give your suggestion....

EmmanuelP, 2017-05-15:
Try to run arv-camera-test -d all and attach the output here.

haseeb-uob, 2017-05-15:
$ ./arv-camera-test -d all
Looking for the first available camera
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 134.102.189.204
[GvInterface::discovery] Device 'The Imaging Source Europe GmbH-17710119' found (interface 134.102.189.204) user_name 'cam_01' - MAC_name '00:07:48:4a:76:f7'
[GvDevice::new] Interface address = 134.102.189.204
[GvDevice::new] Device address = 134.102.189.243
[GvDevice::load_genicam] xml url = 'Local:genicam_AEE5A.zip;00010000;165B' at 0x200
[GvDevice::load_genicam] Xml address = 0x10000 - size = 0x165b - genicam_AEE5A.zip
[GvDevice::load_genicam] Zipped xml data
[GcFeature::set_attribute] Unknown attribute 'VendorName'
[GcFeature::set_attribute] Unknown attribute 'ToolTip'
[GcFeature::set_attribute] Unknown attribute 'MajorVersion'
[GcFeature::set_attribute] Unknown attribute 'MinorVersion'
[GcFeature::set_attribute] Unknown attribute 'SubMinorVersion'
[GcFeature::set_attribute] Unknown attribute 'ProductGuid'
[GcFeature::set_attribute] Unknown attribute 'VersionGuid'
[GcFeature::set_attribute] Unknown attribute 'StandardNameSpace'
[GcFeature::set_attribute] Unknown attribute 'xmlns'
[GcFeature::set_attribute] Unknown attribute 'xmlns:xsi'
[GcFeature::set_attribute] Unknown attribute 'xsi:schemaLocation'
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Representation)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Representation)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Representation)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Representation)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Slope)
[... 129 lines cut ...]
Frame rate = 13 Hz
Frame rate = 14 Hz
[GvStream::_check_frame_completion] Incomplete frame 419
Frame rate = 13 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 13 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
Frame rate = 14 Hz
^C^C^C^C^C^CFrame rate = 14 Hz
Completed buffers = 621
Failures          = 58
Underruns         = 0

haseeb-uob, 2017-05-15:
[output.txt](https://github.com/AravisProject/aravis/files/1001280/output.txt)

EmmanuelP, 2017-05-15:
How many buffer do you read ? Is this a continuous acquisition ?

I don't know ROS, so I can't help you in this area.

haseeb-uob, 2017-05-15:
yes, this is a continuous acquisition. 
using arv-camera-test,
Completed buffers = 621
Failures = 58

EmmanuelP, 2017-05-15:
So you should be able to at least display 9 images over 10.

If you want to increase the reliability of the connection, you may want to change de packet size, or try to make the receiving thread realtime.

haseeb-uob, 2017-05-15:
Yes, you are right I should able to see images. I think the problem is camera parameter settings that's why I am getting a black image which I have to modify.

haseeb-uob, 2017-05-15:
I have solved the ARV_BUFFER_STATUS_MISSING_PACKETS problem. Now the only problem is black images.

haseeb-uob, 2017-05-16:
Thanks for your support I am successfully receiving the images from the camera. But now I have two problems. 
1. Why am I getting gray scale image using my code or aravis-viewer? I don't know how can I get a color image?
2. The camera (DFK Z12GP031) has an internal autofocus feature. How can I activate this feature?