AravisProject/aravis #71: Multiple GigE cameras (not working all together)

haseeb-uob, 2017-05-18:
I am using three GigE cameras from imaging source and they are connected via GigE switch. I can get the video stream from all cameras individually but when I try all cameras together I received nothing. What can be the reason behind this?

EmmanuelP, 2017-05-19:
Either the required bandwidth for the simultaneous streams is too high for a gigabit link, in this case you should try to reduce the framerate and/or the size of the images, or the switch packet buffer is filled and overflows, which can be mitigated by tweaking the inter packet delay (see arv_camera_gv_set_packet_delay).

You can also try to increase the socket buffer size using socket-buffer and socket-buffer-size properties of ArvStream. Have a look at tests/arvcameratest.c.

closed 2017-05-19 by EmmanuelP

th0rgil, 2017-07-04:
Does you camera names (the strings) contain whitespaces?
I've experienced similar issues but with cameras on separate ports but I traced it down to one of my cameras having a spaces in its name.

/Tobias