AravisProject/aravis #73: Hardware trigger - "Could not create image stream"

haseeb-uob, 2017-05-24:
I have successfully configured the three cameras and can get the stream. Now I am trying to trigger all cameras simultaneously with hardware interrupt and I am getting error:  "Could not create image stream"
I used following lines to activate the hardware trigger mode. 
        arv_device_set_string_feature_value(Device, "TriggerMode", "On"); 
        arv_device_set_string_feature_value(Device, "TriggerSource", "Line1");
        arv_device_set_string_feature_value(Device, "TriggerActivation", "RisingEdge"); 
        arv_device_set_string_feature_value(Device, "AcquisitionMode", "Continuous"); 

For triggering I am generating PWM using arduino.

EmmanuelP, 2017-05-29:
It is hard to tell what goes wrong without knowing what is in your code. Could you share it ? Or at least a small sample code that shows the issue.

haseeb-uob, 2017-05-29:
Here are the specific lines of code.
[sample_code.txt](https://github.com/AravisProject/aravis/files/1035408/sample_code.txt)

EmmanuelP, 2017-05-29:
Please try to run your code with environment variable ARV_DEBUG set to `all:3` and report the output.

haseeb-uob, 2017-05-29:
here is the output with set all:3 and set all.
[debug_all.txt](https://github.com/AravisProject/aravis/files/1035546/debug_all.txt)
[debug_all:3.txt](https://github.com/AravisProject/aravis/files/1035547/debug_all.3.txt)

EmmanuelP, 2017-05-29:
Debug_all:3.txt is not long enough, there is no "Can't create image stream" message.

haseeb-uob, 2017-05-29:
Yes because pStream is true and looks like its able to create image stream. But still no image.

haseeb-uob, 2017-05-29:
Here is the code which might be helpful to understand. 

[sample_code.txt](https://github.com/AravisProject/aravis/files/1035834/sample_code.txt)

closed 2017-05-29 by haseeb-uob