AravisProject/aravis #75: Blank screen on viewer with FLIR BFS PGE 13Y3C

ashnarayan13, 2017-06-08:
The Blackfly S cameras are detected and the images are sometimes visualized else I get a blank viewer. 
I am using the Aravis from the git repo
Any fix for it?

EmmanuelP, 2017-06-15:
What platform ?

What is the console output of `arv-viewer -d all:3` ?

EmmanuelP, 2017-06-22:
Which protocol ? GigeVision or USB3Vision ?

fabio-d, 2017-06-28:
Hello,
I am dealing with the same problem using a TX2, so ARM64 platform.
The camera is USB3 and attached you can find the console output.
Using x86_64 it works as expected, if you need the console output let me know.
Thanks
[image]
[13Y3C.txt](https://github.com/AravisProject/aravis/files/1109427/13Y3C.txt)

EmmanuelP, 2019-09-04:
There was a number of fixes in the USB3Vision support since this report. I'm closing it. Please reopen if the issue still persist.

Thanks. Cheers.

closed 2019-09-04 by EmmanuelP