AravisProject/aravis #76: Aravis with Basler daA2500-14uc (U3V)

lkinion, 2017-06-22:
Hi,

I'm having trouble using my Basler daA2500-14uc USB3.0 camera with Aravis. With both arv-viewer and the aravissrc plugin on Gstreamer, it has performance issues at higher resolutions. I can open the camera stream at 30fps for resolutions up to about 1280x720, without issues, but once I start trying 1920x1080 or any higher resolutions, the video is very choppy, and only achieves a few frames per second. I am using YCbCr_422 (YUY2) colorspace.

The Basler camera works just fine in Pylon5 in the exact same mode; it will display the camera stream at 1920x1080, 30fps with no problem. I've also got other cameras working through arv-viewer and aravissrc just fine. For example, with my Pixelink camera I can operate 1920x1080 in a similar colorspace at 30fps or more without any choppiness. 

Here's some additional platform info
Platform: Intel Joule 570x
kernel 4.4.0-1000-joule on Ubuntu SMP PREEMPT x86_64 (this is the default distro from Intel)
aravis version 0.5.9
libusb-1.0-0 version 1.0.20-1
libxml2 version 2.9.3+dfsg1-1ubuntu0.2
libglib2.0-0 version 2.48.2-0ubuntu1

Is there any advice you can give me?

Thanks
Lucas Kinion

EmmanuelP, 2017-06-22:
It would be interresting to look at thread cpu use using top or htop, and compare arv-viewer with the Basler camera and the Pixelink one.

EmmanuelP, 2017-09-21:
@lkinion did you have a chance to look at the thread CPU use ?

EmmanuelP, 2017-11-22:
No update, old report. I'm closing this issue. Please reopen if you have the requested informations.

Thanks.

closed 2017-11-22 by EmmanuelP