AravisProject/aravis #78: OSX build fail

micuat, 2017-06-27:
based on the readme I tried compiling on OSX 10.10.5:

```
brew install gettext intltool gtk-doc automake
brew link --force gettext
aclocal
autoconf
autoheader
glibtoolize --copy
brew install glibtoolize
brew install libtool
glibtoolize --copy
automake --add-missing
intltoolize
export CPPFLAGS="-I/usr/local/Cellar/libxml2/2.9.4_3/include/libxml2/"
./configure
make
```

but it fails with

```
  CCLD     arv-camera-test
  CCLD     arv-chunk-parser-test
  CCLD     arv-heartbeat-test
  CCLD     arv-acquisition-test
  CCLD     arv-auto-packet-size-test
  CCLD     realtime-test
  CCLD     load-http-test
  CCLD     arv-device-test
  CCLD     arv-example
Undefined symbols for architecture x86_64:
  "_sched_getparam", referenced from:
      _print_status in realtimetest.o
  "_sched_getscheduler", referenced from:
      _print_status in realtimetest.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [realtime-test] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [all-recursive] Error 1
```

EmmanuelP, 2017-07-06:
I am a bit puzzled arv-tool compiles fine but not arv-example. I am of little help, as I don't build aravis on OSX.

Could you try to ask on aravis mailing list ? There is probably some OSX users there.

https://www.freelists.org/list/aravis

snickell, 2017-09-14:
I'm having this problem too

snickell, 2017-09-14:
The problem seems to be that aravis is using an API that is not supported on OSX, namely sched_getscheduler

EmmanuelP, 2017-09-15:
fb6596b6737f8a4f78cb3d761809308c1add3a63 should fix this issue.

Thanks.

closed 2017-09-15 by EmmanuelP

zackwyatt, 2017-09-19:
OSX builds working for me now.  thanks!  (note: I just build with "./configure --enable-usb --enable-gst-plugin=no --enable-viewer=no")

EmmanuelP, 2017-09-19:
Thanks @zackwyatt 

I have added OSX in travis and should be able to detect build breakage now.