AravisProject/aravis #79: MSVC 2015 Compiler Errors

mualloc, 2017-06-28:
Hi,
I want to compile Aravis SDK 0.4.1 (not GUI stuff) on Windows platform with MSVC2015 using QtCreator. I downloaded GLib from http://ftp.gnome.org/pub/GNOME/binaries/win64/glib/2.26/ .
I put related include paths and .lib paths of both GLib and Aravis but I got tons of compiler errors like:
C:\Users\user\Documents\Aravis\arvgvsp.h(99): error C2054: expected '(' to follow '__packed__'
C:\Users\user\Documents\Aravis\arvgvsp.h(119): error C2059: syntax error: '<parameter-list>'
C:\Users\user\Documents\Aravis\arvgvsp.h(138): error C2059: syntax error: '<parameter-list>'
C:\Users\user\Documents\Aravis\arvgvsp.h(152): error C2059: syntax error: '<parameter-list>'
C:\Users\user\Documents\Aravis\arvgvsp.h(159): error C2143: syntax error: missing '{' before '*'
                                                                              .
                                                                              .
                                                                              .

I know these errors are directly related to Aravis SDK and I could successfully build the same code on Ubuntu. What might be the problem and how can I solve it?

EmmanuelP, 2017-06-28:
Hi,

Well, that is because packed attribute is a gcc feature. But I was may be over cautious when I have declared gvcp structs with this attribute, as default struct padding should lead to the same member alignment. Try to remove them.

I suspect you will encounter other compilation issue on Windows. I have never heard of someone using aravis on this platform. Please let me know of any trouble you face.

Thanks.

EmmanuelP, 2017-06-28:
Should be fixed by 93a1414a12f0b2b48c3ba3572e7bc0b4c736c0ed.

closed 2017-06-28 by EmmanuelP