AravisProject/aravis #80: SwissKnife evaluation error

TIS-Edgar, 2017-06-29:
Hi Emmanuel,

I am getting wrong results when querying the framerate for different resolutions.
The camera in question is a TIS DMK 33GP1300.
As you might know TIS gige cameras allow you to set TestWidth and TestHeight to change the FPS range without actually changing the used format.

When setting the test resolution to 1280x1024 (the maximum resolution) I still get 5 - 15151.5152 fps, which is what I would expect for the minimum resolution 

I have attached the output of `arv-tool-0.6 control FPS -n "The Imaging Source-23519996" -d evaluator:3`.
From what I can see the SwissKnife 'ResultingFrameClocks' is not correctly evaluated.

You can find the genicam xml [here](https://gist.github.com/TIS-Edgar/60e3ea422f15d4509fa10b24adb2b916).

[23519996_control_FPS_debug_evaluator_3.txt](https://github.com/AravisProject/aravis/files/1111062/23519996_control_FPS_debug_evaluator_3.txt)

EmmanuelP, 2017-06-29:
Hi Edgar,

I think there is an issue in the genicam data. In Genicam standards, it i said:

« The  SwissKnife syntax has  some  extensions:  You  can  use  named  constants using  the Constant entry and  named  sub  expressions using  the  Expression entry as  shown  in  the following example. The sub expressions may not refer to other sub expressions. »

In your Genicam xml, you have:

```
<Expression Name="FULL_FRAME_KERNELS">SENSOR_WIDTH / 8</Expression>
<Expression Name="BLACK_CYCLES">BLACK_LINES * (ROT_BL + FULL_FRAME_KERNELS * 2 + 6)</Expression>
```

Which is not valid.

TIS-Edgar, 2017-06-30:
Hi Emmanuel,

You are right, that is an error in the genicam description.
We fixed it in our firmware.
Apparently aravis is the only one that adheres to that part of the standard.
All other implementations seem to allow subexpression, which is why were caught of guard.
As to whether that should be considered a bug or not I don't know.

EmmanuelP, 2017-06-30:
> As to whether that should be considered a bug or not I don't know.

Most of the time, I prefer to have a code that accepts deviation from the specification. But in this particular case, implementation wise, it was easier to forbid expression references in expressions. I don't remember why exactly, but at least it protects against circular reference issues.

If I get other bug reports about faulty firmwares, I may try to make evaluator accepts expression references.

closed 2017-06-30 by EmmanuelP