AravisProject/aravis #82: Bug in aravismisc.c / gstreamer/ GRAY16_LE

th0rgil, 2017-07-04:
Posted previously on mailing list 

My camera can deliver 10,12,14 or16 bit Grayscale.

when using gstreamer 1.0
gst-launch-1.0 aravissrc ! video/x-raw, format=GRAY16_LE ! filesink location=fname.raw
only 10bit data is saved 
If I comment out row 609-622 (blocks about 10bit and 12bit)
and recompile, gstreamer saves 16bit data properly

I guess its a loockup problem and my pipe caps match the earliest GRAY16_LE entry (Mono 10)
the solution seems to be to use 16 bit pixel format as default and use 12,10 bits if specified by gstreamer 'depth' keyword (also available in gstreamer-1.0)

/Tobias

arwmisc.c row 609-629
{
ARV_PIXEL_FORMAT_MONO_10,
"video/x-raw, format=(string)GRAY16_LE",
"video/x-raw", "GRAY16_LE",
"video/x-raw-gray, bpp=(int)16, depth=(int)10",
"video/x-raw-gray", 16, 10, 0
},
{
ARV_PIXEL_FORMAT_MONO_12,
"video/x-raw, format=(string)GRAY16_LE",
"video/x-raw", "GRAY16_LE",
"video/x-raw-gray, bpp=(int)16, depth=(int)12",
"video/x-raw-gray", 16, 12, 0
},
{
ARV_PIXEL_FORMAT_MONO_16,
"video/x-raw, format=(string)GRAY16_LE",
"video/x-raw", "GRAY16_LE",
"video/x-raw-gray, bpp=(int)16, depth=(int)16",
"video/x-raw-gray", 16, 16, 0
},

EmmanuelP, 2017-07-06:
I have pushed a fix to master.

Thanks.

closed 2017-07-06 by EmmanuelP

th0rgil, 2017-07-07:
Super... tnx!