AravisProject/aravis #83: Incorrect check for introspection support (--enable-introspection=yes) on OS X

petrsmid, 2017-07-05:
I need Python support so I wanted to build with `--enable-introspection=yes`.
I have installed gobject-introspection with homebrew (`brew install gobject-introspection`)
I run the configure script on OS X with enabled introspection (`--enable-introspection=yes`)
`./configure  --prefix=/Users/petr/devel/aravis/dist --enable-introspection=yes  --enable-viewer=false --enable-gst-plugin=false`
but it fails with message: `checking for gobject-introspection... configure: error: gobject-introspection-1.0 is not installed`

I removed the checks for introspection in the configuration script. The configuration and the build then worked fine and the Aravis-0.6.typelib file was correctly created and works fine.

I believe the check for the gobject-introspection is just incorrectly implemented in the configuration script.

EmmanuelP, 2017-07-06:
Do you have a gobject-introspection.pc file installed by homebrew ?

petrsmid, 2017-07-06:
Well, the homebrew installed file `gobject-introspection-1.0.pc` in path `/usr/local/Cellar/gobject-introspection/1.52.1/lib/pkgconfig/gobject-introspection-1.0.pc`

EmmanuelP, 2017-07-06:
If the pc file is not in pkg-config path (PKG_CONFIG_PATH), it is an issue in either homebrew or the gobject-introspection package.

There is nothing I can do i aravis.

Thanks for the report.

closed 2017-07-06 by EmmanuelP

petrsmid, 2017-07-06:
The symlink `/usr/local/lib/pkgconfig/gobject-introspection-1.0.pc` does exist too.
However I have run `pkg-config --exists --print-errors 'gobject-introspection-1.0 >= 0.10.0'` to find out what is happening and it turned out the problem is somewhere else:
```
Package libffi was not found in the pkg-config search path.
Perhaps you should add the directory containing `libffi.pc'
to the PKG_CONFIG_PATH environment variable
Package 'libffi', required by 'gobject-introspection-1.0', not found
```
But the package libffi has been installed before. But what was missing is symlink of the libffi (possibly a problem of the homebrew libffi package?)- so finally this fixed the problem: `ln -s /usr/local/Cellar/libffi/3.2.1/lib/pkgconfig/libffi.pc /usr/local/lib/pkgconfig/libffi.pc`
Now the `pkg-config --exists --print-errors 'gobject-introspection-1.0 >= 0.10.0'` works fine.