AravisProject/aravis #84: Stream destination port being set to control port

TIS-Tim, 2017-07-14:
Hi Emmanuel,

I am having trouble with a setup that is really stress-testing aravis.

Basically, what we do is start 9 instances of a program that starts 9 cameras at the same time. The instances exit once they got an image. When all programs exit, the 9 instances are started again.

It sometimes occurs that the stream destination port is being set to the same port number that is being used for controlling the camera, which results in things not working properly (no images, wireshark unable to dissect, unexpected control messages...).

Here is a wireshark capture of it happening:

[aravis-port-problem.zip](https://github.com/AravisProject/aravis/files/1148551/aravis-port-problem.zip)

The problem is in packet 2664, where the control application at port 0x8d16 sets the stream destination port to 0x8d16.

I am pretty sure this is not supposed to happen, or am I missing something?

TIS-Arne, 2017-07-18:
Might the issue be related to the usage of "g_socket_bind"? Aravis binds to port 0 while also setting the "allow_reuse" flag for both the streaming and the control sockets. The port value of 0 tells the kernel to assign a free port and the "allow_reuse" tells the kernel it is OK to give both sockets the same port number. This would cause the conflict at random.

EmmanuelP, 2017-07-18:
You are probably right. Could you patch your aravis version to change every allow_reuse to FALSE.

EmmanuelP, 2017-07-19:
@TIS-Arne @TIS-Tim I have pushed the fix to master 25d8b942a56cd12d7f561849a3e8edf504a15f19 .

Please reopen if it doesn't fix your issue.

Thanks.

closed 2017-07-19 by EmmanuelP

TIS-Tim, 2017-07-19:
Thanks. Our testing so far shows that doing exactly this change makes the problem no longer appear.