AravisProject/aravis #87: Support for FLIR Blackfly S BFS-U3-51S5M

MartynGunson, 2017-07-19:
Almost all functionality is supported, however some of the camera parameters require the use a ROUND() evaluator, which is not currently implemented

EmmanuelP, 2017-07-19:
ROUND is implemented by commit 84a9803a16bfa370f1c4d92edf887dc15b3b38f3 

For the BlackLevel feature, could you attach the output of `arv-tool-0.6 control BlackLevel -d all:3`

Thanks.

MartynGunson, 2017-07-19:
Please find attached the log file as requested.

[blacklevel.txt](https://github.com/AravisProject/aravis/files/1159561/blacklevel.txt)

EmmanuelP, 2017-07-20:
Could you send me privately the genicam data of this camera:

`arv-tool-0.6 genicam`

Thanks.

MartynGunson, 2017-07-20:
Hello,

Please find attached the requested output

Best Regards
Martyn

EmmanuelP, 2017-09-21:
@MartynGunson I'm still interested in the genicam data.

MartynGunson, 2017-09-21:
Hello Emmanual,

I had sent an email to what I thought was your private email address. It must not have arrived.

Can you give me the email address that you would like the data to be sent to?

Regards
Martyn

EmmanuelP, 2017-09-21:
Oh, sorry. I have checked my email archive and found your anwser.

Thanks.

EmmanuelP, 2017-09-21:
Here is the relevent part:

```
  <Integer Name="BlackLevelRaw" NameSpace="Custom">
    <ToolTip>Controls the offset of the video signal in camera specific units.</ToolTip>
    <Description>Controls the offset of the video signal in camera specific units.</Description>
    <DisplayName>Black Level Raw</DisplayName>
    <Visibility>Expert</Visibility>
    <pIsImplemented>BlackLevelRaw_Imp</pIsImplemented>
    <pIsAvailable>BlackLevelRaw_Avl</pIsAvailable>
    <pIsLocked>BlackLevelRaw_Lck</pIsLocked>
    <pValue>BlackLevelRaw_Val</pValue>
    <pMin>BlackLevelRaw_Min</pMin>
    <pMax>BlackLevelRaw_Max</pMax>
    <pInc>BlackLevelRaw_Inc</pInc>
    <Representation>Linear</Representation>
  </Integer>
  <Float Name="BlackLevel" NameSpace="Standard">
    <ToolTip>Controls the offset of the video signal in percent.</ToolTip>
    <Description>Controls the offset of the video signal in percent.</Description>
    <DisplayName>Black Level</DisplayName>
    <Visibility>Beginner</Visibility>
    <pIsImplemented>BlackLevel_Imp</pIsImplemented>
    <pIsAvailable>BlackLevel_Avl</pIsAvailable>
    <pIsLocked>BlackLevel_Lck</pIsLocked>
    <pAlias>BlackLevel_Int</pAlias>
    <pValue>BlackLevelPercent_Val</pValue>
    <pMin>BlackLevelPercent_Min</pMin>
    <pMax>BlackLevelPercent_Max</pMax>
    <Unit>%</Unit>
    <Representation>Linear</Representation>
  </Float>
  <Converter Name="BlackLevelPercent_Val">
    <pVariable Name="DECPT">BlackLevel_Posn</pVariable>
    <FormulaTo>FROM*(2**DECPT)/100.0</FormulaTo>
    <FormulaFrom>TO/(2**DECPT)*100.0</FormulaFrom>
    <pValue>BlackLevel_Val</pValue>
    <Slope>Increasing</Slope>
  </Converter>
  <SwissKnife Name="BlackLevelPercent_Min">
    <pVariable Name="DECPT">BlackLevel_Posn</pVariable>
    <pVariable Name="RAW">BlackLevel_Min</pVariable>
    <Formula>RAW/(2**DECPT)*100</Formula>
  </SwissKnife>
  <SwissKnife Name="BlackLevelPercent_Max">
    <pVariable Name="DECPT">BlackLevel_Posn</pVariable>
    <pVariable Name="RAW">BlackLevel_Max</pVariable>
    <Formula>RAW/(2**DECPT)*100</Formula>
  </SwissKnife>
  <Boolean Name="BlackLevelClampingEnable" NameSpace="Custom">
    <ToolTip>Enable the black level auto clamping feature which performing dark current compensation.</ToolTip>
    <Description>Enable the black level auto clamping feature which performing dark current compensation.</Description>
    <DisplayName>Black Level Clamping Enable</DisplayName>
    <Visibility>Guru</Visibility>
    <pIsImplemented>BlackLevelClampingEnable_Imp</pIsImplemented>
    <pIsLocked>TLParamsLocked</pIsLocked>
    <ImposedAccessMode>RW</ImposedAccessMode>
    <pValue>BlackLevelClampingEnable_Val</pValue>
    <OnValue>1</OnValue>
    <OffValue>0</OffValue>
  </Boolean>
  <IntSwissKnife Name="BlackLevelSelectorValueToIndex">
    <Visibility>Invisible</Visibility>
    <pVariable Name="SWITCH_TARGET">BlackLevelSelector</pVariable>
    <Formula>( SWITCH_TARGET = 0 ) ? 0 : ( ( SWITCH_TARGET = 1 ) ? 1 : ( ( SWITCH_TARGET = 2 ) ? 2 : 0 )  )</Formula>
  </IntSwissKnife>
  <IntSwissKnife Name="BlackLevelRawAddr">
    <pVariable Name="SELECTOR_INDEX_0">BlackLevelSelectorValueToIndex</pVariable>
    <Formula>0x00106400 + 0x20 * (  ( SELECTOR_INDEX_0 * 1  )  )</Formula>
  </IntSwissKnife>
[... 359 lines cut ...]
    <pInvalidator>HbinISP_Val</pInvalidator>
    <pInvalidator>HbinMode_Val</pInvalidator>
    <pInvalidator>VbinMode_Val</pInvalidator>
    <Sign>Signed</Sign>
    <Endianess>LittleEndian</Endianess>
  </IntReg>
  <MaskedIntReg Name="BlackLevelClampingEnable_Imp">
    <Address>0x00107F80</Address>
    <Length>4</Length>
    <AccessMode>RO</AccessMode>
    <pPort>Device</pPort>
    <Bit>31</Bit>
    <Sign>Unsigned</Sign>
    <Endianess>LittleEndian</Endianess>
  </MaskedIntReg>
  <IntReg Name="BlackLevelClampingEnable_Val">
    <Address>0x00107F84</Address>
    <Length>4</Length>
    <AccessMode>RW</AccessMode>
    <pPort>Device</pPort>
    <Cachable>NoCache</Cachable>
    <Sign>Unsigned</Sign>
    <Endianess>LittleEndian</Endianess>
  </IntReg>
```

EmmanuelP, 2019-09-04:
There was a number of fixes lately in Aravis master. I'm closing this issue. Please reopen if the issue persists.

Cheers.

closed 2019-09-04 by EmmanuelP