AravisProject/aravis #88: Cannot take pictures using Python on a Raspberry Pi

Mutated-Neutrino, 2017-07-20:
I'm trying to use the avaris library to take pictures on a GenICam camera, using a Raspberry Pi. I can a live video feed from the camera using the arv-viewer tool in the viewer directory. When I previously tried to take pictures the resulting images came out ether completely black or with a small line of pixels at the top of the image which were white or grey, we think this is due to a buffer overflow issue as these images still appeared when the lenses cover was on. After trying to solve this issue I now get the error of: 
pi@raspberrypi:~/aravis_2/tests/python $ python mike-test.py 
Camera vendor : Smartek
Camera model  : GCC2062M
Camera id     : 6CD146020812
ROI           : 520x128 at 0,0
Payload       : 66560
Pixel format  : Mono8
Start acquisition
Acquisition
<Buffer object at 0x763089b8 (ArvBuffer at 0x2029a88)>
Traceback (most recent call last):
  File "mike-test.py", line 48, in <module>
    dataFromBuffer = buffer.get_data()
  File "/usr/lib/python2.7/dist-packages/gi/overrides/GObject.py", line 545, in _unsupported_data_method
    raise RuntimeError('Data access methods are unsupported. '
RuntimeError: Data access methods are unsupported. Use normal Python attributes instead
pi@raspberryp

EmmanuelP, 2017-08-02:
Could you please attach a simple test program that shows the issue ?
Thanks.

lionfish0, 2017-08-08:
Thanks for this,
(off topic: I've got a long way with the c code (in tests instead), and am about to python wrap it myself, but if you could help with this that would be great!)

Code:
```
#!/usr/bin/env python

import sys
import time
from gi.repository import Aravis
Aravis.enable_interface ("Fake")
try:
	if len(sys.argv) > 1:
		camera = Aravis.Camera.new (sys.argv[1])
	else:
		camera = Aravis.Camera.new (None)
except:
	print ("No camera found")
	exit ()

camera.set_region (0,0,128,128)
camera.set_frame_rate (10.0)
camera.set_pixel_format (Aravis.PIXEL_FORMAT_MONO_8)

payload = camera.get_payload ()

[x,y,width,height] = camera.get_region ()

print "Camera vendor : %s" %(camera.get_vendor_name ())
print "Camera model  : %s" %(camera.get_model_name ())
print "Camera id     : %s" %(camera.get_device_id ())
print "ROI           : %dx%d at %d,%d" %(width, height, x, y)
print "Payload       : %d" %(payload)
print "Pixel format  : %s" %(camera.get_pixel_format_as_string ())

stream = camera.create_stream (None, None)

for i in range(0,10):
	stream.push_buffer (Aravis.Buffer.new_allocate (payload))

print "Start acquisition"

camera.start_acquisition ()

print "Acquisition"

for i in range(0,20):
	buffer = stream.pop_buffer ()
	print buffer
	dataFromBuffer = buffer.get_data(buffer)
	#dataFromBuffer = arv_buffer_get_data()
	print dataFromBuffer
	if buffer:
		stream.push_buffer (buffer)
		print "Buffer Pushed"

print "Stop acquisition"

camera.stop_acquisition ()
```

The error message:
```
pi@raspberrypi:~/aravis_2/tests/python $ python mike-test.py 
Camera vendor : Smartek
Camera model  : GCC2062M
Camera id     : 6CD146020812
ROI           : 520x128 at 0,0
Payload       : 66560
Pixel format  : Mono8
Start acquisition
Acquisition
<Buffer object at 0x76250850 (ArvBuffer at 0x1d76a88)>
Traceback (most recent call last):
  File "mike-test.py", line 48, in <module>
    dataFromBuffer = buffer.get_data(buffer)
  File "/usr/lib/python2.7/dist-packages/gi/overrides/GObject.py", line 545, in _unsupported_data_method
    raise RuntimeError('Data access methods are unsupported. '
RuntimeError: Data access methods are unsupported. Use normal Python attributes instead
```

larsrin, 2017-08-10:
It is 
dataFromBuffer = buffer.get_data()
not 
dataFromBuffer = buffer.get_data(buffer)

Hope it works ☺️

lionfish0, 2017-08-10:
Thanks for the help! I realise I added in the extra 'buffer' in to see if I could get it working (just messing about). It doesn't work without the 'buffer' either (same error as above).

```
print "Acquisition"

for i in range(0,20):
        buffer = stream.pop_buffer ()
        print buffer
        dataFromBuffer = buffer.get_data()
        #dataFromBuffer = arv_buffer_get_data()
        print dataFromBuffer
        if buffer:
                stream.push_buffer (buffer)
                print "Buffer Pushed"

print "Stop acquisition"
```

I've found [this page](https://www.freelists.org/post/aravis/Aravis-RaspberryPi-Gige-Cam,4) that describes the same problem.

I just tried reinstalling using instructions [here](https://github.com/sightmachine/SimpleCV/wiki/Aravis-(Basler)-GigE-Camera-Install-Guide).
then adding one line:       
`buffer.get_data() `
to the `arv-camera-test.py` file.

So it reads:
```
for i in range(0,20):
        buffer = stream.pop_buffer ()
        print buffer
        buffer.get_data()
        if buffer:
                stream.push_buffer (buffer)
```
but I get the same error as before.

I wonder if I reinstall the whole of raspbian maybe that'll help? @Mutated-Neutrino I wonder if the changes you made to get it installed the first time, might be causing an issue?

I should say, that I can get images out etc using the c code examples. So things are working - just not the python stuff...

lionfish0, 2017-08-10:
Got it working. I don't know if this is the right way to make it work.

I had a poke around at the innards of the buffer class, and found a data pointer. So I just access that using ctypes. Seems to work!
The key line:

`b = ctypes.cast(buffer.data,ctypes.POINTER(ctypes.c_ubyte))`

Is there any reason this is a bad idea?

Here's the code:
```
import sys
import time
import ctypes
from gi.repository import Aravis

Aravis.enable_interface ("Fake")

try:
        if len(sys.argv) > 1:
                camera = Aravis.Camera.new (sys.argv[1])
        else:
                camera = Aravis.Camera.new (None)
except:
        print ("No camera found")
        exit ()
camera.set_region (0,0,2048,1536)
camera.set_frame_rate (1.0)
camera.set_exposure_time(500000)
camera.set_pixel_format (Aravis.PIXEL_FORMAT_MONO_8)
payload = camera.get_payload ()
stream = camera.create_stream (None, None)
camera.start_acquisition ()
pixels = [' ','.',',',',','`',':',';','i','+','t','l','O','I','X','#','#']
pixels = pixels[::-1] #flip at github's black on white
for i in range(0,20):
        stream.push_buffer (Aravis.Buffer.new_allocate (payload))

        buffer = stream.pop_buffer ()
        b = ctypes.cast(buffer.data,ctypes.POINTER(ctypes.c_ubyte))
        print("------------------------");
        for col in range(0,1530*2048,48*2048):
                c = b[col:(col+2048):24]
                print("".join([pixels[ci/16] for ci in c]))
```

And here's the output (me! :) (proof it's working! + I feel like I'm in the matrix :)
```
::::```````,``````````````````::::;;;;iiiiii++++ttttttllllllOlOOOOOOOOOIIIIIIIIIIOOOIX
:::::````````````````````::::::;;;;;iiiiiii+++++tttttlllllllOOOOOOOOOOOIIIIIIIIIOOIIXX
::::`:::````````````::`::::::::;;;;;iiii+++++++ttttllllllllOOOOOOOOOOIIIIIIIIIOOOIXXX#
`::::::::```:`:`::::::::::::::;;iii+++++++++++ttttlllllllOOOOOOOOOOOIIIIIIIIOOOIXXXXXX
,```````::`:::`::::::::::;i;;;i+lOOOOOOOllltttttttllllOOOOOOOOOIIIIIIIIIIIOOOIXXXXXXXX
,,````````::`:::::::::::i+++tllOIIIXXXX##XOOOllttlllllOOOOOOOIOIIIIIIIIIOOOIXXXXXXXX##
``````,,````:::::;:;:;;itOOIOlt+ttlOIIXX####IOOlllllllOOOOOOOIIIIIIIIIIOOOXXXXXXXXXX##
```:````,,`,,`::;;;;;;itOIO:```:;i++tlOIX####XOOttllOOOOOOOOOIIIIIIIIOOOIXXX#XXXX#####
`::::`,,,,    ,`:;;;;i+lOt.  .,,``::i+llOIXX##OlttlllOOOOOOIIIIIIIIOOOIXXXXXXXXX######
``:::`,,,,. .,`,  .`;itl+,       .,.,;t++lOIXXOlltllOOOOOIOIIIIIIIOOIXXX#XXXXX########
;;;;:`,,,``````.   .:it+;.         `+:,`:;+tOIIOOllOOOOIOIIIIIIIOOOXXXXXXXXXXX#XX#####
i;;;;:::::;`,``:`,`:;iii;,      .`:`,, ..,:+lOIOOOIOOOIIIIIIIIOOOXXXXXXIIIIIIOXIIIIXXX
+iiii;;;iii;::::;;;iiii+i,    . :,`:,,.  .,:+lOIIXIIIOIIIIIIOOOIXXXXIIOIOOOOOOOOIOOOOO
t+i;iiiiiiii;;;i;iiiii++ii.   i,.,,:i:    .,:itOIIXXX##XXXIIIXIXXIOOOlOOlOIOlOOOlOOOOl
l+ii+iiiiiii;;;iiiiiiiiiii; ,;,     :.,... ,,`itOIIXXX#X######IXOOlOlllO+lOlOOOOtllOtO
t+++iiiii;i;;;;iiiiiiiiiii;:,:,         ,:, ,;itlOIIOIXIXX#XXXIItOtltlllilOtOOOtttlllO
t++++++iiiii;iiiiiiiiiiiii++++;.           .,:+lOOOltlIXXX##XItOtO+ttttt+lOtIOItttl+ll
++++i+iiiiii;iiiiiiii+++++++++++i.          .:+lOlt;;;tIXXXXXItl+li+t+t+ittOIXIlltlill
++++++iiiiiiiiii++++tttttttttttttt+.        .:+tt;,,,ilOOIXOOIO+Ol++tit;;++IXXXltltill
t++++iiiiiiii++++llltttttlllllOllllllt;,.  ,;i;,.   .;OOXXItiOOOt+tt;ltii+iIXXIltl+llO
++++++iiiiiiiI;;;#XX;;;;llOlllOOlOOOOOllllOIO;:,      ;ttXIOOO+`;;;lllti:+iOIXIIOOllll
+++++++iii++OXOlIIXOiiii#OIlllllOOOOOOOlllOI+l+i:,. ,ii+iIli`llttilli``tt+lOIXIXXIlltt
+++++iiii++OI++tttt+ttttOlOlOOOOOOOOIOOOOOOl+Oiltli,.;t;:Olllt` ::,tllt++tllIIIXXIlltt
t+++++iiOlOXOOOOlOlOlOlOOOOOOOOOOOOOOOOOOOOIlit+OOli:ti `iO+, , ,:lll;; ,:i:ilI##IOll+
tt++++++tl#IIIIIOOOOOOOOOIIIOOOOIOIIIIIIIIOtti:++;`,iOO,.,+Olli`;i: .tii:itlOIX##XIOO+
ttt++++#tXOIXIIOOOOOOOIOOOOOOOOOOOOOOIIIIIOlI+;l:.,XIIt   tIi,+,,`tllt+iii+tOIXX#XXIOt
tttt++Ol#+OIIOOOllllllOllllllllllllllllOIIO+Xt:,,,lXIit;  :XIl: ++t+;,    . ,OXX##XXOi
ttttttX#OlOIIOOOlllllIlllllllllllllllllOIIOl;;:` .O+Xl;.  ,Xt,+;` ,tlt+`;i+tlOIX###XO+
lttttIXItIOIOOOOllllIlllllllllOOOOlOOOlOIIOI+::,,It`l`tt.  iIl: itlllii`:;+i+tOOIX##Il
llttl#IltIIIOOOOOOOIOlOlOOOOOOOOOOOOOOOOIXOIIIIIIO+;itt.   ,O;i;ii:, ,.    ;tlOIX####I
llll#XllIIIIIIIIIOIXOOOOOOIIIOOOOOOIIIIIIIIOIIIIlt+lt  t,   it   ,tlt+i:`:i+tllIXX###X
lllOXOllIIOIIIIIOOOOOOOIIIIIIIIIIIIXXXXXXXXIIIIIIO`:`tt+.   ::,:itll;: ``:,``::;itI###
```
Thanks for the help with this.

EmmanuelP, 2017-08-16:
Which version of aravis are you using ? It looks like you are using an old version, older than 0.4.

get_data method was added in  0.4 serie.

lionfish0, 2017-08-18:
Ah, maybe I copied the code from the linked website without checking - looks like it uses a fork of aravis. I'll try again this evening, thanks @EmmanuelP!

EmmanuelP, 2017-09-21:
@lionfish0 did you succeed ?

lionfish0, 2017-09-21:
Sorry - I never had time in the end - as I got it working I didn't try fixing it more!
I'll probably try it again in a month or so - I'll add a post to this issue when I've tried it. Thanks for following up! Feel free to close the issue if you like (I'll open it and add any future progress when I try it).

EmmanuelP, 2017-09-21:
Ok, thanks.

closed 2017-09-21 by EmmanuelP

mehtajaghvi, 2018-02-01:
I am consistently getting bad frames after  9-10 images. I might have to do with the buffer. How can I delete buffer after every image acquisition? Also, i am seeing the same buffer address for multiple images (is that OK). I have added the output at the bottom here.  I am doing start acquisition..pop buffer..stop acquisition..push buffer...in a while loop. 
I have initialize my stream when i initialize my camera for the first time.  
Does pop buffer reduce the stream size by 1? Why do we execute push buffer after we pop?

<Aravis.Buffer object at 0x7f980f4a6730 (ArvBuffer at **0x22b269**0)>
2018-01-31 16.10.51_Start acquisition
('Emite signals', False)
('capture time', 186005681462000L)
Printing buffer
<Aravis.Buffer object at 0x7f980f4a6870 (ArvBuffer at 0x22b2490)>
2018-01-31 16.10.58_Start acquisition
('Emite signals', False)
('capture time', 168485194757000L)
Printing buffer
<Aravis.Buffer object at 0x7f980f4a69b0 (ArvBuffer at 0x22b2710)>
2018-01-31 16.12.12_Start acquisition
('Emite signals', False)
('capture time', 186086353588000L)
Printing buffer
<Aravis.Buffer object at 0x7f980f4a67d0 (ArvBuffer at **0x22b2690**)>

EmmanuelP, 2018-02-01:
Hi,

Please open a new issue instead of commenting an older one.

In the new issue, please add the following informations:
- what you mean by bad frame (corrupted images ?)
- language used
- aravis version
- the console output of your code with ARV_DEBUG environment variable set to stream:3

You have to pop/push because aravis stores the frame it gets in a circular buffer. The use of a circular buffer helps to not constantly alloc/free memory.