AravisProject/aravis #89: arv_buffer_get_data undefined reference?

lionfish0, 2017-07-26:
Hi,
I'm trying to modify the `arvcameratest.c` example to let us check the data from the camera makes sense. I've just changed the `new_buffer_cb` function (line 106) to include a call to `arv_buffer_get_data` (see below), however when I try to compile (by calling make at the command line) I get:

```
pi@raspberrypi:~/aravis_2/tests $ make
  CC       arvcameratest.o
  CCLD     arv-camera-test
arvcameratest.o: In function `new_buffer_cb':
/home/pi/aravis_2/tests/arvcameratest.c:116: undefined reference to `arv_buffer_get_data'
```

I'm not sure why that method is undefined, when other methods (also listed on the [library's reference manual](https://aravisproject.github.io/docs/aravis-0.4/ArvBuffer.html) are used in the code).

Thanks for any advice,

Modified code fragment:
```
static void
new_buffer_cb (ArvStream *stream, ApplicationData *data)
{
        ArvBuffer *buffer;
        char *buffer_data;

        buffer = arv_stream_try_pop_buffer (stream);
        if (buffer != NULL) {
                if (buffer->status == ARV_BUFFER_STATUS_SUCCESS)
                        data->buffer_count++;
                size_t size = 0;
                buffer_data = (char *) arv_buffer_get_data (buffer, &size);
                printf("%d\n",buffer_data[0]);
                /* Image processing here */
                arv_stream_push_buffer (stream, buffer);
        } 
}
```

EmmanuelP, 2017-08-02:
Works for me. I have modified arv-camera-test.c to use arv_buffer_get_data, and it compiles fine.

What version of aravis are you using ?
May be you have tried to install different versions of aravis, and have the older includes and pkg-config files (<0.4) installed on your system.

lionfish0, 2017-08-02:
Good point. I'm not sure - I've got something working now - I lost track of versions and reinstalled some stuff.

Thanks for the help.

The next problem I've got is getting the images off the camera quicker. Currently I'm getting less than 1Hz!

closed 2017-08-02 by lionfish0