AravisProject/aravis #90: Can't connect to cameras on a different subnet

mjog, 2017-08-01:
I can't connect to a camera on another subnet by IP address using arv-tool from git master.

The host is on 172.30.3.7 (a /24 subnet) and the camera is on 172.30.4.110 (a separate /24 subnet), and packets are routed between the two by a L3 switch. Hosts on both subnets can ping hosts on the other, and in particular I can ping the camera from the host, and the camera's netmask and default gateway seems to be set correctly.

As requested on the list, here's the output from `arv-tool-0.6 --debug=all:3 --address=172.30.4.110`:

> mjg@payens:~/aravis$ ./src/arv-tool-0.6 --debug=all:3
> Found 0 USB3Vision device (among 5 USB devices)
> [GvDiscoverSocket::new] Add interface 127.0.0.1
> [GvDiscoverSocket::new] Add interface 10.0.3.1
> [GvDiscoverSocket::new] Add interface 25.40.218.103
> [GvDiscoverSocket::new] Add interface 10.248.47.163
> [GvDiscoverSocket::new] Add interface 172.30.3.7
> No device found

When running the command above, using Wireshark on the host, I see a UDP broadcast to 255.255.255.255 on port 3956 which Wireshark lists as a GVCP DISCOVERY_CMD message, but no response. Per some suggestions online, I've tried disabling the kernel's reverse path filtering using `sysctl -w net.ipv4.conf.*.rp_filter=0`, but with no success.

hinxx, 2017-08-01:
FWIW, I usually had to make sure that no iptables rules on the host PC were active, flushed.. Also for any 'subtables' like nat, mangle, ..

EmmanuelP, 2017-08-01:
> When running the command above, using Wireshark on the host, I see a UDP broadcast to 255.255.255.255 on port 3956 which Wireshark lists as a GVCP DISCOVERY_CMD message, but no response.

Did you try to put a computer on the same subtnet as the camera, to check if the broadcasted discovery packet is seen here ?

mjog, 2017-08-02:
On the same subnet with the same command, this is what I get:

> mjg@payens:~/aravis$ ./src/arv-tool-0.6 --debug=all:3 --address=172.30.4.110
> Found 0 USB3Vision device (among 5 USB devices)
> [GvDiscoverSocket::new] Add interface 127.0.0.1
> [GvDiscoverSocket::new] Add interface 10.0.3.1
> [GvDiscoverSocket::new] Add interface 25.40.218.103
> [GvDiscoverSocket::new] Add interface 10.248.47.163
> [GvDiscoverSocket::new] Add interface 172.30.4.199
> packet_type  = ack
> command      = discovery-ack
> size         = 248
> id           = 65535
> manufacturer = Point Grey Research
> name         = 
> model        = Blackfly BFLY-PGE-12A2C
> address      = 172.30.4.109
> 00000000 00 00 00 03 00 f8 ff ff 00 01 00 02 00 00 00 01  ................
> 00000010 00 00 00 b0 9d f6 62 f6 00 00 00 07 00 00 00 05  ......b.........
> 00000020 00 00 00 00 00 00 00 00 00 00 00 00 ac 1e 04 6d  ...............m
> 00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff 00  ................
> 00000040 00 00 00 00 00 00 00 00 00 00 00 00 ac 1e 04 fe  ................
> 00000050 50 6f 69 6e 74 20 47 72 65 79 20 52 65 73 65 61  Point Grey Resea
> 00000060 72 63 68 00 00 00 00 00 00 00 00 00 00 00 00 00  rch.............
> 00000070 42 6c 61 63 6b 66 6c 79 20 42 46 4c 59 2d 50 47  Blackfly BFLY-PG
> 00000080 45 2d 31 32 41 32 43 00 00 00 00 00 00 00 00 00  E-12A2C.........
> 00000090 46 57 3a 76 31 2e 32 37 2e 33 2e 31 30 20 46 50  FW:v1.27.3.10 FP
> 000000a0 47 41 3a 76 32 2e 30 32 00 00 00 00 00 00 00 00  GA:v2.02........
> 000000b0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
> 000000c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
> 000000d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
> 000000e0 31 36 31 34 37 31 39 30 00 00 00 00 00 00 00 00  16147190........
> 000000f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
> 
> [GvInterface::discovery] Device 'PointGrey-16147190' found (interface 172.30.4.199) user_name '' - MAC_name 'Point Grey Research-16147190'
> packet_type  = ack
> command      = discovery-ack
> size         = 248
> id           = 65535
> manufacturer = Point Grey Research
> name         = 
> model        = Blackfly BFLY-PGE-12A2C
> address      = 172.30.4.110
> 00000000 00 00 00 03 00 f8 ff ff 00 01 00 02 00 00 00 01  ................
> 00000010 00 00 00 b0 9d da 73 24 00 00 00 07 00 00 00 05  ......s$........
> 00000020 00 00 00 00 00 00 00 00 00 00 00 00 ac 1e 04 6e  ...............n
> 00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff 00  ................
> 00000040 00 00 00 00 00 00 00 00 00 00 00 00 ac 1e 04 fe  ................
> 00000050 50 6f 69 6e 74 20 47 72 65 79 20 52 65 73 65 61  Point Grey Resea
> 00000060 72 63 68 00 00 00 00 00 00 00 00 00 00 00 00 00  rch.............
> 00000070 42 6c 61 63 6b 66 6c 79 20 42 46 4c 59 2d 50 47  Blackfly BFLY-PG
> 00000080 45 2d 31 32 41 32 43 00 00 00 00 00 00 00 00 00  E-12A2C.........
> 00000090 46 57 3a 76 31 2e 32 37 2e 33 2e 31 30 20 46 50  FW:v1.27.3.10 FP
> 000000a0 47 41 3a 76 32 2e 30 32 00 00 00 00 00 00 00 00  GA:v2.02........
> 000000b0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
> 000000c0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
> 000000d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
> 000000e0 31 34 33 31 36 33 32 34 00 00 00 00 00 00 00 00  14316324........
> 000000f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
> 
> [GvInterface::discovery] Device 'PointGrey-14316324' found (interface 172.30.4.199) user_name '' - MAC_name 'Point Grey Research-14316324'
> PointGrey-14316324 (172.30.4.110)
> 

Wireshark reports the GVCP DISCOVERY_CMD being sent out, and two DISCOVERY_ACK responses sent back from the two cameras.

The problem in the first instance is that I assume that the DISCOVERY_CMD broadcast is not being relayed from the computer's subnet to the camera's subnet. This is the correct behaviour from the network's perspective since limiting the broadcast domain is exactly why you have different subnets.

The question however is why the DISCOVERY_CMD is broadcast in the first place when `--address` is specified? If GigE protocol requires one to be sent, would it be possible to send that DISCOVERY_CMD targeted directly at the specified address rather than as a broadcast? If the user has specified a specific device, there's no point enumerating the others anyway.

EmmanuelP, 2017-08-02:
> The question however is why the DISCOVERY_CMD is broadcast in the first place

That could be avoided, but when the IP address is known, aravis should still be able to open a connection to the device, it does not rely on a discovery acknowledge packet. That is what is done in arvgvinterface.c, in _open_device and arv_gv_device_locate.

The goal of arv_gv_device_locate is to determine which ethernet interface should be used to reach the device. It does so by trying to read a register from all the available interfaces.

Using Wireshark, you should see a bunch of read register command packets.

EmmanuelP, 2017-08-02:
I have just read the code of arv-tool, and now I remember how it works.

The address parameter is used as a filter on the discovered devices. It does not allow to access a device that doesn't answer to a discovery command, just like arv-viewer.

You should be able to see your camera using the arv-camera-test tool present in the 'tests' directory. You can specify the device IP address using the --name option, something like:

./tests/arv-camera-test --name="134.158.98.197"

mjog, 2017-08-05:
No joy using arv-camera-test, but wireshark showed something a bit more interesting. This is what I get from the command:

> mjg@payens:~/aravis$ ping -c 1 172.30.4.110PING 172.30.4.110 (172.30.4.110) 56(84) bytes of data.
> 64 bytes from 172.30.4.110: icmp_seq=1 ttl=62 time=0.529 ms
> --- 172.30.4.110 ping statistics ---
> 1 packets transmitted, 1 received, 0% packet loss, time 0ms
> rtt min/avg/max/mdev = 0.529/0.529/0.529/0.000 ms
> mjg@payens:~/aravis$ ./tests/arv-camera-test --name=172.30.4.110
> Looking for camera '172.30.4.110'
> No camera found

And this is what Wireshark sees:
> 1	0.000000000	172.30.3.7	172.30.4.110	GVCP	> READREG_CMD [Number of Stream Channels]
> 2	0.000014957	10.248.47.163	172.30.4.110	GVCP	> READREG_CMD [Number of Stream Channels]
> 3	0.000021943	25.40.218.103	172.30.4.110	GVCP	> READREG_CMD [Number of Stream Channels]
> 4	0.000028660	172.30.3.7	172.30.4.110	GVCP	> READREG_CMD [Number of Stream Channels]
> 5	1.002008727	172.30.3.7	255.255.255.255	GVCP	> DISCOVERY_CMD 

Packets 1) and 5) from 172.30.3.7 should be making it to the camera, but I'm not sure why there is not response. I'll look into that, but it will probably take a few days to get to it.

EmmanuelP, 2017-11-22:
No update, old report. I'm closing this issue. Please reopen if you have the requested informations.

Thanks.

closed 2017-11-22 by EmmanuelP

mjog, 2017-11-27:
Hey, yeah sorry we had to change the networking topology to get something working, so I can't test this out right now.