AravisProject/aravis #91: arv-viewer Display Problem with Pleora iPort Cl-GigE External Frame Grabber

mualloc, 2017-08-02:
Hello,

I have seen on mail list that there are other people that faced to view stream of Pleora devices via arv-viewer. Since I am not very familiar with neither GigE Vision nor the Aravis, I cannot solve the problem although I have some insight.

To describe the situation: I ran arv-fake-gv-camera-0.6 and there is a Pleora iPort Cl-GigE External Frame Grabber connected to a ethernet hub. arv-viewer can discover and enumurate successfully both. When I click "start video acquisition" button, I can view arv-fake-gv-camera-0.6 stream but when I do the same thing for iPort, only a dark black screen appears. (which should be a test pattern stream alike arv-fake-gv-camera-0.6 stream)

In the output file of the command

./arv-viewer -d all:3 > arv-viewer-only-iport-stream-all-log.txt

I detected some error evocative lines such as

                                                            .

                                                            .

[PixelFormat::to_gst_caps_string] 0x01080001 -> video/x-raw, format=(string)GRAY8
[PixelFormat::to_gst_caps_string] 0x01080002 not found
[PixelFormat::to_gst_caps_string] 0x01100003 -> video/x-raw, format=(string)GRAY16_LE
[PixelFormat::to_gst_caps_string] 0x010c0004 not found

                                                            .

                                                            .

[ArvDevice::get_float_feature_bounds] Node 'ExposureTimeAbs' is not a float
[ArvDevice::get_integer_feature_bounds] Node 'GainRaw' is not an integer
[ArvDevice::get_float_feature_value] Node 'AcquisitionFrameRateAbs' is not a float
[ArvDevice::get_string_feature_value] Node 'GainAuto' is not a string
[ArvDevice::get_string_feature_value] Node 'GainAuto' is not a string
[ArvDevice::get_string_feature_value] Node 'GainAuto' is not a string
[ArvDevice::get_string_feature_value] Node 'ExposureAuto' is not a string
[ArvDevice::get_integer_feature_value] Node 'GainRaw' is not an integer
[ArvDevice::get_float_feature_value] Node 'ExposureTimeAbs' is not a float

                                                            .

                                                            .

Are these really error implying logs, and how I can I solve the issue so that I can view the test pattern of iPort? (After that I will try to get live stream from the camera.) By the way, I can view the test pattern of iPort on Windows eBus Player.

Thanks.

PS: Aravis version is 0.5.9.

EmmanuelP, 2017-08-02:
`[PixelFormat::to_gst_caps_string] 0x01080001 -> video/x-raw, format=(string)GRAY8
[PixelFormat::to_gst_caps_string] 0x01080002 not found
[PixelFormat::to_gst_caps_string] 0x01100003 -> video/x-raw, format=(string)GRAY16_LE
[PixelFormat::to_gst_caps_string] 0x010c0004 not found`

That means Mono8 and Mono16 are available on your device, and are supported by arv-viewer, which is good.

`[ArvDevice::get_float_feature_value] Node 'AcquisitionFrameRateAbs' is not a float`

ArvCamera probably fails to setup continuous acquisition mode needed by arv-viewer. Could you refer to the iPort documentation and see what has to be done in order to configure a conituous acquisition.

EmmanuelP, 2017-09-21:
Could you please attach the output of: `arv-tool-0.6 features` ?

Thanks.

bgrunert, 2017-09-27:
Hello,

I'm not the OP, but facing the exact same issue.

This is the output of `arv-tool-0.6 features`:
[pleora_iport_cl-gige_features.txt](https://github.com/AravisProject/aravis/files/1336298/pleora_iport_cl-gige_features.txt)

Thanks for looking into it!

EmmanuelP, 2017-09-27:
For the frame rate setting, ArvCamera, hence ArvViewer, relies on some features that are not available on this device: TriggerMode, TriggerSelector and more importantly AcquisitionFrameRate or AcquisitionFrameRateAbs.

@bgrunert, Could you have a look at the Pleora documentation and see which features must be used to configure a frame rate limited continuous acquisition ?

bgrunert, 2017-09-27:
It does not seem to be possible to set arbitrary frame rates (which kind of makes sense, as that would be a camera feature rather than a frame grabber feature). The only feature I could find in the docs that can limit frame rate during continuous acquisition is AcquisitionFrameToSkip which obviously restricts us to fractions of one nth the base frequency.

So I suppose some specific workarounds would be necessary if that grabber were to be supported eventually?
@mualloc, did you make any progress in the meantime?

zackwyatt, 2018-07-11:
Hi -- was curious if @mualloc or @bgrunert have any updates?

I also have a Pleora iPort Cl-GigE adapter.   Was wondering if you've been able to read buffer data from your Pleoras with Aravis?  I can read genicam property values with Aravis but can't set them (unless I update register value).  Have also had trouble reading buffers from the ArvStream.

lililla, 2021-02-21:
Hi, has any solution been found, i am having the same problem with aravis 0.8.6. Thanks