AravisProject/aravis #94: problems on ubuntu 16.04

larsrin, 2017-08-09:
I'm having some serious issues on ubuntu 16.04. The problems seem to be related to gstreamer 1.0 but I don't know much about gstreamer so I have become stuck. 

If I run ./arv-viewer it has two problems (maybe single source?)
1) It gives warnings about fonts 
(lt-arv-viewer:26082): Aravis-WARNING **: Failed to make stream thread high priority

(lt-arv-viewer:26082): Pango-WARNING **: Invalid UTF-8 string passed to pango_layout_set_text()
2) it gives an segmentation fault. See attachment

If I run ./launch arv-single-buffer-test.py there are no errors but the image is all zeros.

I have installed aravis by 
[error-arv-viewer.txt](https://github.com/AravisProject/aravis/files/1211623/error-arv-viewer.txt)

sudo apt-get -y remove  libgtk-3-dev libgstreamer1.0 libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-bad gstreamer1.0* 

sudo apt-get -y install libgirepository1.0-dev libnotify-dev libgtk-3-dev libgstreamer1.0 libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-bad gstreamer1.0-plugins-good  gstreamer1.0* 

git clone https://github.com/AravisProject/aravis

cd aravis
./autogen.sh --enable-viewer --enable-introspection=yes --enable-gst-plugin --enable-gst-plugin-0.10
make 
sudo make install
sudo ldconfig
reboot

EmmanuelP, 2017-08-17:
> It gives warnings about fonts

That is probably not related to problem 2.

> it gives an segmentation fault. See attachment

Could you try to get a backtrace using gdb.

gdb /path/to/arv-viewer
run

then after the crash:

bt full

larsrin, 2017-08-21:
Hi Emmanuel

I did as you proposed and the output is attached. 

I fiddled around with the 'fake' input and changed some of the parameters in the GUI. Then it crashes. No cameras were connected.

What do you make of it?

Best, Lars

[aravis_error_seg_fault.txt](https://github.com/AravisProject/aravis/files/1238523/aravis_error_seg_fault.txt)

EmmanuelP, 2017-09-11:
I'm closing this issue as it looks like it is a problem in gdk.

You may want to report it to gtk+ or ubuntu people. It would be better to get the backtrace with gtk+ debug symbols, but I don't know how it is done on ubuntu.

Thanks.

closed 2017-09-11 by EmmanuelP