AravisProject/aravis #95: Teledyne Dalsa Genie Nano C2020 offset error

Fabio81x, 2017-08-10:
Hello
we are starting to use Aravis to acquire an image from a Teledyne DALSA GigE camera on nvidia TX2. 
I dowloaded aravis "intswissknife" because the errors described in other posts (payload int error) and seems to work all but the offsets. 
If I set the offsets with the command arv_camera_set_region() with offsets different than 0, the stream returns that the buffers have TIMOUT error. Could this be an Aravis problem or a camera problem? 

What we do is run the example program (aravis wiki), print the status of the buffer (in the callback). We also check that the feature is set after the closing of the example program with "arv-tool-0.6 control OffsetX" and we see that the parameter is right. We use Width 320, Height 128, OffsetX 640, OffsetY 512.

Thanks for the help

EmmanuelP, 2017-08-17:
> I dowloaded aravis "intswissknife" because the errors described in other posts (payload int error) and seems to work all but the offsets.

The code in this branch has been merged into master.

> If I set the offsets with the command arv_camera_set_region() with offsets different than 0, the stream returns that the buffers have TIMOUT error. Could this be an Aravis problem or a camera [...]

Could you attach the console output when your test is running with ARV_DEBUG environment variable set to 'thread:3,stream-thread:3' ?

IamSierraCharlie, 2017-08-17:
Hi! Just as another point of interest, does the camera come with its own propriety software?  If so, what happens when you apply the same offset / settings in that software?  If the offset works in the cameras propriety software, then it might suggest an aravis issue, otherwise; if not, then something at play on the camera side?  In my experience, I always found it useful to compare capabilities.  Admittedly the camera I use is different to yours.

Fabio81x, 2017-08-28:
Hello to all

1) here there is the log

Command:
`ARV_DEBUG=thread:3,stream-thread:3 ./GigeVision`

LOG:
_[GvStream::stream_thread] Packet timeout = 40 ms
[GvStream::stream_thread] Frame retention = 200 ms
[GvStream::find_frame_data] Start frame 1
[GvStream::process_packet] Error packet at dt = 49, packet id = 1 frame id = 1
[GvStream::find_frame_data] Start frame 2
[GvStream::missing_packet_check] Resend request at dt = 40285, packet id = 54/55
[GvStream::send_packet_request] frame_id = 2 (23 - 54)
[GvStream::missing_packet_check] Resend request at dt = 80621, packet id = 54/55
[GvStream::send_packet_request] frame_id = 2 (23 - 54)
[GvStream::missing_packet_check] Resend request at dt = 120899, packet id = 54/55
[GvStream::send_packet_request] frame_id = 2 (23 - 54)
[GvStream::missing_packet_check] Resend request at dt = 161153, packet id = 54/55
[GvStream::send_packet_request] frame_id = 2 (23 - 54)
[GvStream::check_frame_completion] Timeout for frame 1 at dt = 217999
Frame Error, status: 2
[GvStream::close_frame] Close frame 1
[GvStream::check_frame_completion] Timeout for frame 2 at dt = 201433
Frame Error, status: 2
[GvStream::close_frame] Close frame 2
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz_

2) Yes, with the camera own sw (Sapera) seems that we can change the offsets ant the image comes right, but only in windows form not on linux systems.

Other informations:

- It seems that the first time you try to run the sw, the offsets did not affect the capture and it works. If i run the `arv-tool-0.6 control OffseX` I found a value of 0 instead 640. The other times the camera goes wrong (and sometimes chrashes, I have to power off and on the camera, with the arv-tool-0.6 I can't find it). If I run the test `arv-tool-0.6 control OffseX` after the second time i see the right offsets set.

- I also tried to set with 2 different code lines the dimension of the images (forst) wth offsets to 0 and than to set the offsets (with the same image)

        arv_camera_set_region (camera, 0, 0, 320, 128);
        arv_camera_set_region (camera, 640, 512, 320, 128);

I found that the first time you run the code (after power reset of the camera) the problem occours. 
LOG:
_[GvStream::stream_thread] Packet timeout = 40 ms
[GvStream::stream_thread] Frame retention = 200 ms
[GvStream::find_frame_data] Start frame 1
[GvStream::process_packet] Error packet at dt = 53, packet id = 1 frame id = 1
[GvStream::find_frame_data] Start frame 2
[GvStream::process_packet] Error packet at dt = 33, packet id = 1 frame id = 2
[GvStream::find_frame_data] Start frame 3
[GvStream::process_packet] Error packet at dt = 36, packet id = 1 frame id = 3
[GvStream::find_frame_data] Start frame 4
[GvStream::process_packet] Error packet at dt = 28, packet id = 1 frame id = 4
[GvStream::find_frame_data] Start frame 5
[GvStream::process_packet] Error packet at dt = 25, packet id = 1 frame id = 5
[GvStream::check_frame_completion] Timeout for frame 1 at dt = 240122
Frame Error, status: 2
[GvStream::close_frame] Close frame 1
[GvStream::find_frame_data] Start frame 6
[GvStream::check_frame_completion] Timeout for frame 2 at dt = 200052
Frame Error, status: 2
[GvStream::close_frame] Close frame 2
[GvStream::process_packet] Error packet at dt = 218, packet id = 1 frame id = 6
[GvStream::find_frame_data] Start frame 7
[GvStream::check_frame_completion] Timeout for frame 3 at dt = 200095
Frame Error, status: 2
[GvStream::close_frame] Close frame 3
[... 224 lines cut ...]
            old_sigint_handler = signal (SIGINT, set_cancel);

            /* Run the main loop */
            g_main_loop_run (data.main_loop);

            signal (SIGINT, old_sigint_handler);

            g_main_loop_unref (data.main_loop);

            /* And disable emission of this signal */
            arv_stream_set_emit_signals (stream, FALSE);

            /* Stop the video stream */
            arv_camera_stop_acquisition (camera);

            g_object_unref (stream);
        } else
            printf ("Can't create stream thread (check if the device is not already used)\n");

        g_object_unref (camera);
    } else
        printf ("No camera found\n");

    return 0;
}

EmmanuelP, 2017-09-21:
What gives `arv-tool-0.6 -d evaluator:3,genicam:3,device:3 control Width=320 Height=128 OffsetX=640 OffsetY=512 Width Height OffsetX OffsetY` when streaming fails ?

EmmanuelP, 2017-11-22:
No update, old report. I'm closing this issue. Please reopen if you have the requested informations.

Thanks.

closed 2017-11-22 by EmmanuelP