AravisProject/aravis #96: Unable to claim interface for USB3 cameras

Bidski, 2017-08-25:
I have 2 PointGrey Flea3 USB3 cameras connected an Ubuntu 17.04 server system. I did not compile the viewer or gstreamer aspects of Aravis as I do not require this functionality in my system.

I can successfully use `arv-tool-0.6` to list all of the features of both of my cameras and I can use a simple custom program (using Aravis) to display basic information about the USB devices (id, serial number, vendor). I am also able to compile and run `arvexample.c` and it seems to open a camera and print out a frame rate of 60Hz.

However, when I try to open my cameras from my multi-threaded c++ application I am unable to claim the usb interface for either of my cameras. Here is debugginh the output from Aravis (I added some extra debug statement to help pinpoint the issue).

`[UvDevice::new] Vendor  = Point Grey Research
[UvDevice::new] Product = Flea3 FL3-U3-13E4C
[UvDevice::new] S/N     = 00E81B94
[UvDevice::new] Using control endpoint 2, interface 0
[UvDevice::new] Using data endpoint 1, interface 2
[UvDevice::new] Failed to claim USB control interface to 'Flea3 FL3-U3-13E4C - #00E81B94', LIBUSB_ERROR_BUSY

(process:6822): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
Found 2 USB3Vision devices (among 6 USB devices)
[UvDevice::new] Vendor  = Point Grey Research
[UvDevice::new] Product = Flea3 FL3-U3-13E4C
[UvDevice::new] S/N     = 00E81B94
[UvDevice::new] Using control endpoint 2, interface 0
[UvDevice::new] Using data endpoint 1, interface 2
[UvDevice::new] Failed to claim USB control interface to 'Flea3 FL3-U3-13E4C - #00E81B94', LIBUSB_ERROR_BUSY

(process:6822): GLib-GObject-CRITICAL **: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
`
While the application is multi-threaded, object creation and camera connection is all happening inside a single thread.

All applications that I have described are run from the same user, so I do not think there is an issue with permissions. There also does not seem to be any other services/programs using the cameras either.

Can you provide any help with debugging this issue?

closed 2017-08-25 by Bidski

EmmanuelP, 2017-08-26:
I guess you have found a solution to the issue by yourself. I would be interested by what was going wrong.

Bidski, 2017-08-27:
I have not resolved this yet. I realised that there is a mailing list and that this is issue is probably best discussed there until it can be determined that my issue is actually a bug/issue with Aravis and not with my own work.