AravisProject/aravis #97: Can't set Height/Width to non-zero value with Teledyne Dalsa Nano C4900

zackwyatt, 2017-08-28:
Hi -- 

I'm using a Teledyne DALSA Nano C4900 and am unable to set the Height nor Width to non-0 values (and thus unable to get a non-0 Payload size to capture an image).  I've tried the Aravis 0.5.9 release and the master branch (to include fixes from https://github.com/AravisProject/aravis/issues/58). 

Here's what I see:
```$ arv-tool-0.6 control Width=96 Height=96
Teledyne DALSA-S1138485 (10.10.100.107)
Width = 0
Height = 0
```

I added the `-d all:3` to my arv-tool control command.  Full output is here: [nano_set.txt](https://github.com/AravisProject/aravis/files/1256998/nano_set.txt)  I noticed an error about HEIGHT__INC (and WIDTH__INC, too) not being found:
```
...
[Evaluator::evaluate] Result = (int64) 536871056
[GcRegisterNode::_get_integer_value] address = 0x20000090, value = 0x4
[Evaluator::set_int64_variable] TO = 4
[Evaluator::evaluate_as_double] Expression = '(TO / HEIGHT__INC) * HEIGHT__INC / FACTOR_ALL'
[Evaluator::parse_expression] (TO / HEIGHT__INC) * HEIGHT__INC / FACTOR_ALL
[Evaluator::parse_expression] Found 9 items in expression
[Evaluator::parse_expression] 2 items in garbage list
[Evaluator::parse_expression] 7 items in token list
[Evaluator::evaluate_as_double] Parsing status = 0
(var) TO = 4
(var) HEIGHT__INC = 0 not found
[Evaluator::set_error] Error 'unknown variable'
Height = 0
...
```

I believe this is from the height_convertor section of the genicam config from the camera:

``` <IntConverter Name="height_convertor" NameSpace="Custom">
         <Visibility>Invisible</Visibility>
         <pVariable Name="FACTOR">VERTICAL_FACTOR</pVariable>
         <pVariable Name="REG_STREAMING">writeStreamingState</pVariable>
         <pVariable Name="HEIGHT_INC_">HEIGHT_INC</pVariable>
         <Expression Name="HEIGHT__INC">(REG_STREAMING)? 1:(FACTOR * HEIGHT_INC_)</Expression>
         <Expression Name="FACTOR_ALL">(REG_STREAMING)? 1:FACTOR</Expression>
         <FormulaTo>FROM* FACTOR_ALL</FormulaTo>
         <FormulaFrom>(TO / HEIGHT__INC) * HEIGHT__INC / FACTOR_ALL</FormulaFrom>
         <pValue>height_raw</pValue>
         <Representation>Linear</Representation>
         <Slope>Increasing</Slope>
      </IntConverter>
```

Using Teledyne's Windows control software, I *am* able to set heights/widths and capture images. 

I'm new to this stuff and uncertain if I'm using the camera incorrectly or there is actually some issue upstream.  Sorry in advance if the HEIGHT__INC is normal and/or I've missed something simple!

Thanks very much for your help. 

  - Zack

p.s. I'm evaluating a few cameras and would be happy to pass some along for your testing+verification purposes, too. 

p.s. in case it's helpful, this is output from arv-camera-test:
```
Looking for the first available camera
vendor name           = Teledyne DALSA
model name            = Nano-C4900
device id             = S1138485
image width           = 0
image height          = 0
horizontal binning    = 1
vertical binning      = 1
payload               = 0 bytes
exposure              = 15001 µs
gain                  = 1 dB
gv n_stream channels  = 1
gv current channel    = 0
gv packet delay       = 2000 ns
gv packet size        = 1500 bytes
Frame rate = 5 Hz
Frame rate = 3 Hz
Frame rate = 3 Hz
Frame rate = 3 Hz
^CFrame rate = 3 Hz
Completed buffers = 17
Failures          = 0
Underruns         = 0
```

EmmanuelP, 2017-09-05:
>  Full output is here: nano_set.txt I noticed an error about HEIGHT__INC (and WIDTH__INC, too) not being found

Good catch. Expression is not yet supported by IntConverter and Converter.

EmmanuelP, 2017-09-08:
Should be fixed by 722178861462ab06edb52e9d0a929220a2b38a9a.

closed 2017-09-08 by EmmanuelP

zackwyatt, 2017-09-08:
thanks!  Will give it a try.

zackwyatt, 2017-09-19:
Hi -- just wanted to confirm that the commit did fix the problem.  Dalsa Nano C4900 Width/Height/etc now working as expected.  thanks!