AravisProject/aravis #98: PointGrey Blackfly - problem with setting feature props

ArkadiuszRaj, 2017-09-04:
Using latest Aravis library, under Linux OS, I am not able to completely deactivate auto exposure settings:

```
op@mdcx1:~$ arv-tool-0.6
PointGrey-15144900 (192.168.3.13)
TIS-50610342 (192.168.3.12)
op@mdcx1:~$ arv-tool-0.6 -a 192.168.3.13 control AcquisitionStop
PointGrey-15144900 (192.168.3.13)
AcquisitionStop executed
op@mdcx1:~$ arv-tool-0.6 -a 192.168.3.13 control GainAuto=?
PointGrey-15144900 (192.168.3.13)
GainAuto = Continuous
op@mdcx1:~$ arv-tool-0.6 -a 192.168.3.13 control GainAuto=Off
PointGrey-15144900 (192.168.3.13)
GainAuto = Off
op@mdcx1:~$ arv-tool-0.6 -a 192.168.3.13 control ExposureAuto=?
PointGrey-15144900 (192.168.3.13)
ExposureAuto = Continuous
op@mdcx1:~$ arv-tool-0.6 -a 192.168.3.13 control ExposureAuto=Off
PointGrey-15144900 (192.168.3.13)
ExposureAuto = Continuous
op@mdcx1:~$
```

zackwyatt, 2017-09-05:
FWIW -- I'm running into same issue with a Flea3 USB.  https://www.ptgrey.com/flea3-12-mp-color-usb3-vision-sony-imx172

I attached at run of arv-tool control ExposureAuto=Off with -d all:3 to this ticket.  [debug_output.txt](https://github.com/AravisProject/aravis/files/1279128/debug_output.txt)

Possibly something with?  

``` <IntSwissKnife Name="ExposureAutoOffImp_Int">
    <pVariable Name="EXPMODE">ExposureMode_Int</pVariable>
    <pVariable Name="EXPMANUALINQ">ExposureManualInq_Bit</pVariable>
    <Formula>
(( EXPMODE = 1 ) || ( EXPMANUALINQ = 1 )) ? 1 : 0
</Formula>
  </IntSwissKnife>
```

I also attached the genicam export for the flea I'm using: [flir_flea-xml.txt](https://github.com/AravisProject/aravis/files/1279134/flir_flea-xml.txt)

ArkadiuszRaj, 2017-09-06:
So I attach similar output from in BlackFly camera

[arv-tool-6.0 genicam](https://github.com/AravisProject/aravis/files/1279982/genicam.xml.txt)
[arv-tool-6.0 control -d all:3 ExposureAuto=Off ](https://github.com/AravisProject/aravis/files/1279983/arv-debug.txt)

EmmanuelP, 2017-09-11:
It is an issue with the precedence of the `~` operator in the expression `(~(~TO|0xFEFFFFFF)) ? 2:0` in the expression evaluator.

With TO replaced by its actual value, the debug output is:

```
[Evaluator::evaluate_as_int64] Expression = '(~(~0xC2000221|0xFEFFFFFF)) ? 2:0'
[Evaluator::parse_expression] (~(~0xC2000221|0xFEFFFFFF)) ? 2:0
[Evaluator::parse_expression] Found 13 items in expression
[Evaluator::parse_expression] 4 items in garbage list
[Evaluator::parse_expression] 9 items in token list
[Evaluator::evaluate_as_int64] Parsing status = 0
(int64) 3254780449
(int64) 4278190079
(operator) |
(operator) ~
(operator) ~
(int64) 2
(int64) 0
(operator) :
(operator) ?
[Evaluator::evaluate] Result = (int64) 2
```
The `|` operation is done before `~`, which is wrong.

EmmanuelP, 2017-09-11:
Fixed in 3d26c4ab80187c113b76a6590f571af6c13e5143

Thanks for the report.

closed 2017-09-11 by EmmanuelP

zackwyatt, 2017-09-19:
hi -- just wanted to note that the commit did fix the issue. I'm now able to set ExposureAuto = Off.  Thanks!

EmmanuelP, 2017-09-19:
Great. Thanks for the follow up.