AravisProject/aravis #100: Video greatly deformed for a GigeVision camera

hcaloto, 2017-09-07:
I'm facing an issue when trying to get the camera video stream. I'm using a non-commercial GigeVision camera and I get the video greatly deformed. What I get is:

[image]

If I access the camera stream using the tool eBusPlayer I can see the video properly:

[image]

The point is that after using eBusPlayer I can get the right camera image by using Aravis (Gstreamer aravissrc or arv-viewer). But once I  power-cycle the camera I continue facing the same issue as shown before: wrong video with Aravis.

Any suggestions? It looks like eBusPLayer does some negotiation that Aravis does not.

The Gstreamer output is always the some in both cases, whether the stream is correct and whether is not):
```
LOG                aravissrc gstaravis.c:292:gst_aravis_start:<gige_src_S0_src> Open camera '(null)'
LOG                aravissrc gstaravis.c:86:gst_aravis_get_all_camera_caps:<gige_src_S0_src> Get all camera caps
LOG                aravissrc gstaravis.c:138:gst_aravis_get_caps:<gige_src_S0_src> Available caps = video/x-raw, format=(string)GRAY8, width=(int)[ 4, 16384 ], height=(int)[ 16, 16384 ], framerate=(fraction)[ -2147483647/1, 2147483647/1 ]; video/x-raw, format=(string)GRAY16_LE, width=(int)[ 4, 16384 ], height=(int)[ 16, 16384 ], framerate=(fraction)[ -2147483647/1, 2147483647/1 ]; video/x-raw, format=(string)RGB, width=(int)[ 4, 16384 ], height=(int)[ 16, 16384 ], framerate=(fraction)[ -2147483647/1, 2147483647/1 ]
LOG                aravissrc gstaravis.c:427:gst_aravis_fixate_caps:<gige_src_S0_src> Fixate caps
LOG                aravissrc gstaravis.c:155:gst_aravis_set_caps:<gige_src_S0_src> Requested caps = video/x-raw, format=(string)GRAY16_LE, width=(int)1024, height=(int)768, framerate=(fraction)159/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive
DEBUG              aravissrc gstaravis.c:181:gst_aravis_set_caps:<gige_src_S0_src> Frame rate = 159 Hz
DEBUG              aravissrc gstaravis.c:192:gst_aravis_set_caps:<gige_src_S0_src> Buffer timeout = 2000000 µs
DEBUG              aravissrc gstaravis.c:194:gst_aravis_set_caps:<gige_src_S0_src> Actual frame rate = 158.946 Hz
DEBUG              aravissrc gstaravis.c:201:gst_aravis_set_caps:<gige_src_S0_src> Gain = 0
DEBUG              aravissrc gstaravis.c:205:gst_aravis_set_caps:<gige_src_S0_src> Actual gain = 0
DEBUG              aravissrc gstaravis.c:217:gst_aravis_set_caps:<gige_src_S0_src> Actual exposure = 0 µs
LOG                aravissrc gstaravis.c:256:gst_aravis_set_caps:<gige_src_S0_src> Start acquisition
```

Any ideas on what the issue might be?

Thanks in advance.

KennisL, 2017-09-08:
Thanks for the updated information.

________________________________
kennisl@outlook.com

[bad]<https://user-images.githubusercontent.com/6048965/30166433-a7350ce8-93e3-11e7-956a-f9bc3e1a39f6.png>

[good]<https://user-images.githubusercontent.com/6048965/30166537-eb04f49c-93e3-11e7-8090-17c33bcd33b7.png>

EmmanuelP, 2017-09-11:
Could you attach the output of 'arv-viewer --debug device:3' please ?

hcaloto, 2017-09-11:
Hi @EmmanuelP and thank you for your prompt answer.

This is the output of 'arv-viewer --debug device:3' after I power-cycle the camera:

```
[GvDevice::new] Interface address = 192.168.42.19
[GvDevice::new] Device address = 192.168.42.13
[GvDevice::load_genicam] xml url = 'Local:gvrd-ac701.xml;FEC20000;00013AAB' at 0x200
[GvDevice::load_genicam] Xml address = 0xfec20000 - size = 0x13aab - gvrd-ac701.xml
[GvDevice::new] Packet resend = yes
[GvDevice::new] Write memory = yes
[ArvDevice::set_integer_feature_value] Node 'ChunkModeActive' is not an integer
[ArvDevice::get_integer_feature_value] Node 'BinningHorizontal' is not an integer
[ArvDevice::get_integer_feature_value] Node 'BinningVertical' is not an integer
[ArvDevice::get_integer_feature_bounds] Node 'BinningHorizontal' is not an integer
[ArvDevice::get_integer_feature_bounds] Node 'BinningVertical' is not an integer
[GvDevice::Heartbeat] Ack value = 2
[GvDevice::Heartbeat] Ack value = 2
[GvDevice::create_stream] Number of stream channels = 1
[GvStream::stream_new] Packet size = 576 byte(s)
[ArvDevice::get_float_feature_bounds] Node 'ExposureTimeAbs' is not a float
[ArvDevice::get_integer_feature_bounds] Node 'GainRaw' is not an integer
[ArvDevice::get_string_feature_value] Node 'GainAuto' is not a string
[ArvDevice::get_string_feature_value] Node 'GainAuto' is not a string
[ArvDevice::get_string_feature_value] Node 'GainAuto' is not a string
[ArvDevice::get_string_feature_value] Node 'ExposureAuto' is not a string
[ArvDevice::get_integer_feature_value] Node 'GainRaw' is not an integer
[ArvDevice::get_float_feature_value] Node 'ExposureTimeAbs' is not a float
[GvDevice::Heartbeat] Ack value = 2
[GvDevice::Heartbeat] Ack value = 2
 ```
 
 After using eBusPlayer the difference is:
 
 ```
 [GvStream::stream_new] Packet size = 1444 byte(s)
 ```
  
  instead of:
  
  ```
  [GvStream::stream_new] Packet size = 576 byte(s)
  ```
  
  Is there any way of setting the packet size through the gstreamer element (aravissrc)?
  
  Thanks

EmmanuelP, 2017-09-11:
> Is there any way of setting the packet size through the gstreamer element (aravissrc)?

No.

Could you try to reset the camera, then set the packet size using `arv-tool-0.6 control GevSCPSPacketSize=1444`, then run the gstreamer pipeline, and let me know if it fixes the issue.

hcaloto, 2017-09-11:
Yes, if I set the packet size using `arv-tool-0.6 control GevSCPSPacketSize=1444` I can get the right camera image.

EmmanuelP, 2017-09-11:
It could be a firmware bug, some mis-initialized register that is fixed after the packet size is set. You may want to try to contact the camera manufacturer.

Meanwhile, you could also test the camera using the baumer-gapi-sdk, or any other tool that doesn't set the packet size at initialization. That would help to understand if it is indeed a firmware bug.

I'm closing this issue for now, but don't hesitate to reopen it if you think it is an aravis issue, or add a comment if the firmware bug is confirmed.

Thanks.

closed 2017-09-11 by EmmanuelP