AravisProject/aravis #102: USB IO errors when accessing multiple cameras

nerian-vision, 2017-09-27:
I frequently receive the following error when accessing more than one camera at once:

```
...
[UvDevice::read_memory] Command sending error: LIBUSB_ERROR_IO
[UvDevice::read_memory] Command sending error: LIBUSB_ERROR_IO
[UvDevice::read_memory] Command sending error: LIBUSB_ERROR_IO
[UvDevice::read_memory] Command sending error: LIBUSB_ERROR_IO
[UvDevice::read_memory] Command sending error: LIBUSB_ERROR_IO
[UvDevice::read_memory] Command sending error: LIBUSB_ERROR_IO
[UvDevice::read_memory] Command sending error: LIBUSB_ERROR_IO
../../glib-2.48.2/glib/gmem.c:130: failed to allocate 18446743528248704992 bytes
```

Unfortunately I haven't been able to narrow down the source of error.

EmmanuelP, 2017-10-02:
Which camera/manufacturer ?

A user reported a similar issue solved by a firmware upgrade:

https://www.freelists.org/post/aravis/SOLVED-Re-Re-Unable-to-claim-interface-for-USB3-cameras

nerian-vision, 2017-10-03:
The above test run was with two FLIR Chameleon3 CM3-U3-13S2M-CS. They do have the most recent firmware (which is actually two years old). I think I also observed this error before with two Basler ace acA1300-200um, but I'm not so sure anymore. The error only happens sometimes. Other times the camera work without issues.

nerian-vision, 2017-10-06:
I can confirm that this also occurs with a pair of Balser ace acA1300-200um

eudoxos, 2018-12-13:
Just hitting this issue with Basler acA1920-150um -- it happened when connecting to a USB3 hub which was itself only connected to USB2 port in the computer.  When I connect straight to USB3 port, or to the USB3 hub connected to USB3 port, the camera works just fine. It also happened it was failing over USB3, then I had to reconnect it and it was okay again.... Not sure if this helps somewhat. Cheers!

nerian-vision, 2018-12-14:
I'm not entirely sure what I did to fix that problem for me. I think this was related to another issue I had with the Linux Kernel. The Kernel was flooding the log with millions of messages like this:

xhci_hcd WARN Event TRB for slot 2 ep 8 with no TDs queued

The warning can be ignored, but the performance impact of writing so many logs actually caused USB problems. So check your Kernel logs and possibly switch to a different Kernel.

EmmanuelP, 2018-12-14:
Thanks for the follow up Konstantin.

I'm closing this issue then.

Cheers.

closed 2018-12-14 by EmmanuelP