AravisProject/aravis #103: ARV_BUFFER_STATUS_TIMEOUT with the Teledyne DALSA Genie Nano C2020

clecocel, 2017-09-30:
Hi!

We have a Teledyne DALSA Genie Nano C2020 that we want to integrate with ROS using [`camera_aravis`](https://github.com/shone-io/camera_aravis). It works fine at low framerates (up to ~15 fps).

Above that we get some `Frame error: ARV_BUFFER_STATUS_TIMEOUT` messages. Note that not all frames fail, we get a continuous chunk of video, then it freezes, then starts again.

We've tried playing around with the `"packet-timeout"` and `"frame-retention"` properties with no success.

Any help would be greatly appreciated! Thanks :)

EmmanuelP, 2017-10-02:
Did you check the bandwidth requirement for the stream you try to setup, and the bandwidth capability of your network ?

Is it a wired network ?

clecocel, 2017-10-02:
Hi,
It’s a GigE wired network (camera connected directly to ethernet port of a laptop). 

We want to acquire 30fps, the camera is sending raw images using Bayer RG 8-bit at a resolution of 2048x1536. This adds up to ~755Mb/s of pure image data. Adding network packets overhead it should still be in the realm of possible.

From your answer it sounds like this is most likely a networking issue. Reading about proprietary drivers (Sapera LT SDK for Teledyne or FlyCapture for Point Grey) it looks like they bypass the OS’s network stack to increase effective bandwidth. Is this implemented in Aravis as well?

Thanks for your time!

EmmanuelP, 2017-10-02:
Hi,

Le lun. 2 oct. 2017 à 16:48, clecocel <notifications@github.com> a 
écrit :
> Hi,
> It’s a GigE wired network (camera connected directly to ethernet
> port of a laptop).
>
> We want to acquire 30fps, the camera is sending raw images using
> Bayer RG 8-bit at a resolution of [...]

Aravis doesn't have a kernel driver, but it is possible to use 
AF_PACKET sockets that have good performances. Please refer to the 
README for some informations regarding the system setup.

Also, you could try to increase the socket buffer size in order to 
minimize the lost of packets.

Finally, you may want to try to make the stream thread high-priority.

arv-camera-test in the tests directory will help you to test these 
different settings, and its source code will show you how it is done.

Cheers,

	Emmanuel.

closed 2017-11-22 by EmmanuelP