AravisProject/aravis #108: Feature Request/Help: Update ROS node for TIS GigE camera

Kennyal3, 2017-10-04:
Hello, EE undergrad here working on a research project. Needing a ros image_raw and camera_info topic from a GigE camera (device model: The Imaging Source DFK33GX265e).
There is an existing "camera_aravis" node but it is very outdated and does not work with ROS Kinetic.
http://wiki.ros.org/camera_aravis
I have been attempting to update this node using Aravis v0.4.1.
I am able to run the camera outside of ROS using arv-viewer but I first have to reduce packet size to 1484. No real issues there.
For the ROS node: My updated program will compile, run and read the camera's properties but will not output a video stream. It hangs right after it displays the camera's properties without error. I've very new at dealing with camera sensor data and any help (or an updated ROS node) would be greatly appreciated. Here is my current cpp fiile "camera_node.cpp".
https://github.com/Kennyal3/ROS-TIS-GigE-Camera-Node/blob/master/camera_aravis_0.4.1/src/camera_node.cpp
~Kenny L.

ccannon94, 2017-10-04:
👍 this would be very helpful for my team

EmmanuelP, 2017-10-25:
To begin, enable aravis debug output before running your software, setting the ARV_DEBUG environment variable:
`export ARV_DEBUG=all:3`
And attach the console output here.

EmmanuelP, 2017-10-25:
By the way, I would suggest you to work with aravis 0.5.x releases. There could be some API instabilities until the next stable release, but hopefully this stable release should happen in the next months.

boilerbots, 2017-11-07:
If you take a look at all the Forks from the original packages you can see that this one seems to be active, you might give it a try it seems to work fine under Kinetic. https://github.com/linknum23/camera_aravis

EmmanuelP, 2017-11-22:
No update, old report. I'm closing this issue. Please reopen if you have the requested informations.

Thanks.

closed 2017-11-22 by EmmanuelP